vvip-68/GyverPanelWiFi #70: проблема компиляции версии 1.07 и выше

KAKTO, 2020-12-21:
Можно проверить код в файле \libraries\EspSoftwareSerial\src\SoftwareSerial.h ? 
Начиная с версии 1.07 изменены строки 130-137, в строке 132 открыта (и не закрыта, на сколько я понимаю) фигурная скобка.  
Заменил этот файл файлом версии 1.06 компиляция проходит без ошибок

RadioLed2000, 2020-12-21:
Ради интереса скачал старую версию и сравнил.
В SoftwareSerial.h в новой версии в этих строчках находится код:
```
#if defined(ESP8266)
    int availableForWrite() override {
#else
    int availableForWrite() {
#endif
        if (!m_txValid) return 0;
        return 1;
    }
```
В старой версии:
```
    int availableForWrite() {

        if (!m_txValid) return 0;
        return 1;
    }
```

vvip-68, 2020-12-21:
На гитхабе этой библиотеки идут обсуждения по поводу этой функции **int availableForWrite() override**

vvip-68, 2020-12-21:
Так а к чему вообще весь этот абстрактный спор - есть у этой функции override или нет?

RadioLed2000, 2020-12-21:
Так а почему у всех работает, а у человека нет?
 @KAKTO, какую плату прошиваете?

vvip-68, 2020-12-21:
И вообще - ответ от разработчиков

> Don't use the EspSoftwareSerial from the Arduino library repository with the ESP8266, it's already versioned and part of the ESP8266 Arduino core. For ESP32, do use the Arduino library repository, though. Even with the ESP32, be sure to match new EspSoftwareSerial releases to current, often "unreleased", Git master branch commits.

ТО есть если вы компилите под ESP8266 - эта библиотека вообще не участвует в сборке, ибо ее версия уже находится внутри ядра ESP8266 и используется именно "ядерная" версия библиотеки, а не это вот всё...

KAKTO, 2020-12-21:
Не было бы вопроса, если бы скобка на 132 строке не сворачивала код до конца. Я понимаю так: на 138 строке скобка закрывает скобку на 134 строке ( файл смотрю здесь на github). Если можно так:
132 {
134 {
138 }
иду учить азы.
В библиотеке EspSoftwareSerial нашёл изменения в 2 файлах, чего небыло до 1.07 версии: SoftwareSerial.h и SoftwareSerial.cpp
Но компиляция не проходит с изменённым SoftwareSerial.h
Согласен, что нужно менять всю библиотеку (компиляция проходит без проблем). Всего лишь вижу файл из-за которого не компилится.

vvip-68, 2020-12-21:
@KAKTO? да, прочитайте про директивы условной компиляции, как работают и всё такое

vvip-68, 2020-12-21:
Почему у меня компилится? И под ESP8266 и под ESP32? Что я делаю не так?
Давайте полный лог ошибок компилятора...

vvip-68, 2020-12-21:
Возможен вариант, что поскольку EspSoftwareSerial берется из ядра ESP8266 - вы просто используете старую версию ядра, в котором что-то не срастается. Какая версия ядра у вас? Обновите ядро до последней версии.

KAKTO, 2020-12-21:
Вообще-то компиляция не проходит на NodeMCU 1.0 (2.6.1) Как писал: решил заменой на старую библиотеку

vvip-68, 2020-12-21:
Ну, решили - и хорошо.
Вообще-то последняя версия ядра - 2.7.4.    2.6.1 - устаревшая и вполне может быть, что не совместима с актуальными версиями библиотек.

closed 2020-12-21 by vvip-68

KAKTO, 2020-12-21:
Вот я пристал со скобками. Я даже здесь не соображаю. Пошёл азы учить и ядро поновее копать.
Arduino: 1.8.12 (Windows 10), Плата:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:1MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

In file included from sketch\a_def_hard.h:420:0,

                 from C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\GyverPanelWiFi_v1.08.ino:20:

C:\Program Files (x86)\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003

 #    pragma message "FastLED version 3.003.003"

                     ^

In file included from C:\Program Files (x86)\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from sketch\a_def_hard.h:420,

                 from C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\GyverPanelWiFi_v1.08.ino:20:

C:\Program Files (x86)\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

In file included from sketch\a_def_hard.h:432:0,

                 from C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\GyverPanelWiFi_v1.08.ino:20:

C:\Program Files (x86)\Arduino\libraries\EspSoftwareSerial\src/SoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override

     int availableForWrite() override {

         ^

In file included from sketch\a_def_hard.h:418:0,

                 from C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\GyverPanelWiFi_v1.08.ino:20:

C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\runningText.ino: In function 'String processDateMacrosInText(String)':

C:\Program Files (x86)\Arduino\libraries\Time/TimeLib.h:57:35: warning: narrowing conversion of '(((int)iYear) + -1970)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

 #define  CalendarYrToTm(Y)   ((Y) - 1970)

                                   ^

C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\runningText.ino:1161:69: note: in expansion of macro 'CalendarYrToTm'

            tmElements_t tm = {0, iMinutes, iHours, 0, iDay, iMonth, CalendarYrToTm(iYear)};

                                                                     ^

C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\runningText.ino: In function 'void rescanTextEvents()':

C:\Program Files (x86)\Arduino\libraries\Time/TimeLib.h:57:35: warning: narrowing conversion of '(((int)iYear) + -1970)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

 #define  CalendarYrToTm(Y)   ((Y) - 1970)

                                   ^

C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\runningText.ino:1665:62: note: in expansion of macro 'CalendarYrToTm'

       tmElements_t tm = {0, iMinute, iHour, 0, iDay, iMonth, CalendarYrToTm(iYear)}; 

                                                              ^

C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\runningText.ino: In function 'void extractMacroSDates(String)':

C:\Program Files (x86)\Arduino\libraries\Time/TimeLib.h:57:35: warning: narrowing conversion of '(((int)iYear1) + -1970)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

 #define  CalendarYrToTm(Y)   ((Y) - 1970)

                                   ^

C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\runningText.ino:1966:65: note: in expansion of macro 'CalendarYrToTm'

     tmElements_t tm1 = {0, iMinute1, iHour1, 0, iDay1, iMonth1, CalendarYrToTm(iYear1)}; 

                                                                 ^

C:\Program Files (x86)\Arduino\libraries\Time/TimeLib.h:57:35: warning: narrowing conversion of '(((int)iYear2) + -1970)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

 #define  CalendarYrToTm(Y)   ((Y) - 1970)

                                   ^

C:\Users\XARA\Desktop\GyverPanelWiFi_v1.08\runningText.ino:1967:66: note: in expansion of macro 'CalendarYrToTm'

     tmElements_t tm2 = {59, iMinute2, iHour2, 0, iDay2, iMonth2, CalendarYrToTm(iYear2)}; 

                                                                  ^

exit status 1
Ошибка компиляции для платы NodeMCU 1.0 (ESP-12E Module).

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

RadioLed2000, 2020-12-21:
@KAKTO, попробуйте удалить всю папку c:\Program Files (x86)\Arduino\libraries\EspSoftwareSerial\
Будет ли компилироваться?

vvip-68, 2020-12-21:
Ю Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

Из этого отчета мало что видно. Вот, включите более полный.
Но это только если вы обновили версию ядра до актуальной. Если нет - дальше не в чем разбираться - пустоею
Кстати, версия Arduino IDE 1.8.12 - тоже устаревшая.
Были случаи , когда некоторые проблемы решались обновление на актуальную версию. Она 1.8.13
Потому что от версии ArduinoIDE могут меняться настройки и/или версия компилятора.

KAKTO, 2020-12-21:
Поставил последнюю версию ядра ESP - нет проблем (сам мучился и людей мучил - дилетант блин) и с удалённой папкой компилируется. 
Извиняюсь за отнятое время. Я ф школу азы учить.
Вопрос закрыт.