vvip-68/GyverPanelWiFi #72: Бегущая строка

Mariner21V, 2020-12-22:
Макрос

vvip-68, 2020-12-22:
Ничего не понял.

Mariner21V, 2020-12-22:
Ну хорошо, попытаюсь ещё раз.
Итак, пишу надпись. Затем прописываю макрос "период отображения строки" {S} . Сохраняю. Но строка пробегает независимо от указанного периода времени. Макросы других вариантов бегущих строк срабатывают как надо.

vvip-68, 2020-12-22:
Что-то я у себя в ни в коде, ни в wiki вообще макроса {S} не нахожу...
Это точно из этого проекта?
Для отображения текста указанное количество секунда используется макрос {T30}
Здесь 30 - число секунд, которое будет отображаться бегущая строка. Можно ставить другое нужное число секунд...

vvip-68, 2020-12-22:
А, вспомнил... Это т недавно добавлялся... Сейчас гляну что с н им не так...

vvip-68, 2020-12-22:
И пришли текст именно как ты его у себя пишешь, чтоб мне было с чем работать

Mariner21V, 2020-12-22:
Хорошо. Но я писал его в разных вариантах. Даже точно как примеры в wiki были изложены.

vvip-68, 2020-12-22:
Завтра посмотрю...

Mariner21V, 2020-12-22:
Пойду отключать "Добрый вечер!"  Спасибо за ответ !

vvip-68, 2020-12-22:
И лучше бы прислал текст. Я мог проверять это на периоде дат и там, к примеру, это работает.
Хотя я вроде проверял и на интервале в одних сутках.
Но, все же вернее именно тот вариант, который неправильно работает. А то вдруг неправильно вычисляется если какие-то определенны часы указать...

Mariner21V, 2020-12-22:
Хорошо. Пишу....

Mariner21V, 2020-12-22:
Доброе утро! {S06:05#08:35} - вариант 1, Добрый вечер! {S**.**.2020 16:15#**.**.2020 23:15}{S**.**.2021 16:15#**.**.2021 23:15} - вариант 2. 
Хотел поинтересоваться по поводу такого варианта (тоже прописывал в приложении):
До Рождества осталось {R07.01.2021#2}{S02.01.2021#06.01.2021}
строка 2 - С Рождеством! {S07.01.2021}{-}
Вот.

vvip-68, 2020-12-23:
В общем - потестировал. Сидел как дурак час пялился в эту матрицу, читал строчки.
Набор был такой:
```
for (byte i = 0; i<36; i++) textLines[i] = "";
  
  textLines[0]  = "##";
  textLines[1]  = "Всё будет хорошо!";
  textLines[2]  = "22:25-22:30{S22:25#22:29}";
  textLines[3]  = "22:30-22:34{S22:30#22:34}";
  textLines[4]  = "22:35-22:40{S22:35#22:39}";

  textLines[5]  = "Строка 1{S**.**.2020 22:50#**.**.2020 22:54}";
  textLines[6]  = "Строка 2{S**.**.2020 23:00#**.**.2020 23:04}{S**.**.2020 23:10#**.**.2020 23:14}";
  
  textLines[7]  = "Ещё строка";

  saveTexts();
```
Интервал показа строки - каждые 10 секунд.
Результат: все строки с макросом {S} показываются именно в те периоды времени, которые прописаны в макросе.
Всё работает. Что я делаю не так?

closed 2020-12-23 by vvip-68

LisovR, 2021-12-05:
Я не отрицаю что я..., но нифига не понимаю что не работает.
Прошивка 1.12, плата wemos d1 mini.
Прописываю: Тест!!!{S05.12.2021 19:00#05.12.2021 19:05} работает всегда!