vvip-68/GyverPanelWiFi #73: Бесконечное переподключение к mqtt

sysardex, 2020-12-23:
Собственно сабж

```
04:35:30.227 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:30.227 -> Подключение к MQTT-серверу выполнено.
04:35:30.227 -> Подписка на topic='cmd' >> OK
04:35:30.227 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:32.731 -> 
04:35:32.731 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:32.731 -> Подключение к MQTT-серверу выполнено.
04:35:32.731 -> Подписка на topic='cmd' >> OK
04:35:32.731 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:35.230 -> 
04:35:35.230 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:35.267 -> Подключение к MQTT-серверу выполнено.
04:35:35.267 -> Подписка на topic='cmd' >> OK
04:35:35.267 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:37.697 -> 
04:35:37.697 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:37.734 -> Подключение к MQTT-серверу выполнено.
04:35:37.734 -> Подписка на topic='cmd' >> OK
04:35:37.734 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:40.205 -> 
04:35:40.205 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:40.240 -> Подключение к MQTT-серверу выполнено.
04:35:40.240 -> Подписка на topic='cmd' >> OK
04:35:40.240 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:42.725 -> 
04:35:42.725 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:42.725 -> Подключение к MQTT-серверу выполнено.
04:35:42.725 -> Подписка на topic='cmd' >> OK
04:35:42.725 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:45.205 -> 
04:35:45.205 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:45.242 -> Подключение к MQTT-серверу выполнено.
04:35:45.242 -> Подписка на topic='cmd' >> OK
04:35:45.242 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:47.710 -> 
04:35:47.710 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:47.747 -> Подключение к MQTT-серверу выполнено.
04:35:47.747 -> Подписка на topic='cmd' >> OK
04:35:47.747 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:49.628 -> MQTT >> OK >> test_window/WiFiPanel-4/stt/SQ	 >> {"SQ":"X"}
04:35:49.666 -> MQTT >> OK >> test_window/WiFiPanel-4/stt	 >> {"W":21,"H":12,"DM":true,"PS":true,"PD":60,"IT":0,"AL":false,"RM":true,"PW":0,"BR":255,"WU":1,"WT":15,"WR":213,"WS":1502026,"WC":true,"WN":false,"WZ":true,"EF":25,"EN":"Северное сияние","UE":true,"UT":true,"UC":true,"SE":"X","SS":"X","BE":"X","CE":true,"CC":0,"CO":0,"CK":0,"NC":0,"SC":105,"C1":"255,255,255","DC":true,"DD":2,"DI":240,"NP":true,"NT":60,"NZ":3,"NS":"ru.pool.ntp.org","DW":true,"OF":false,"TM":false,"AW":0,"AT":"1 0 0|2 0 0|3 0 0|4 0 0|5 0 0|6 0 0|7 0 0","AD":10,"AE":43,"MX":false,"AU":false,"AN":"PanelAP","NW":"ThinkNetwork","IP":"192.168.100.62","QZ":true,"QA":true,"QP":1883,"QS":"192.168.100.24","QU":"mqtt","QD":1,"QR":"test_window","TE":true,"TI":90,"TS":"500000000000000000000000000000000000","CT":0,"C2":"255,255,255","OM":1629,"ST":69,"AM1T":"00 00","AM1A":-3,"AM2T":"00 00","AM2A":-3,"AM3T":"00 00","AM3A":-3,"AM4T":"00 00","AM4A":-3}
04:35:50.238 -> 
04:35:50.238 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:50.238 -> Подключение к MQTT-серверу выполнено.
04:35:50.238 -> Подписка на topic='cmd' >> OK
04:35:50.276 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:52.726 -> 
04:35:52.726 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:52.762 -> Подключение к MQTT-серверу выполнено.
04:35:52.762 -> Подписка на topic='cmd' >> OK
04:35:52.800 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
04:35:55.253 -> 
04:35:55.253 -> Подключаемся к MQTT-серверу '192.168.100.24:1883'; ClientID -> 'WiFiPanel-4' ...
04:35:55.253 -> Подключение к MQTT-серверу выполнено.
04:35:55.289 -> Подписка на topic='cmd' >> OK
04:35:55.289 -> MQTT >> OK >> test_window/WiFiPanel-4/pwr	 >> online
```

AndreyLyamkin, 2020-12-23:
Странно, но у меня тоже локальный MQTT сервер на Mosquitto broker, и такого ни разу не было. Было, что не подключалось. Но потери. Wi-Fi не падает? Еще есть такой параметр библиотеке FastLED - "WAIT_TIME", вот можно поиграться с его значением. 
Подробнее тут: https://github.com/vvip-68/GyverPanelWiFi/wiki/Три-шага-к-успеху.-Некоторые-особенности-настроек...

vvip-68, 2020-12-23:
Опасный параметр. Если на гирлянде нет артефактов, неустранимых питанием - его лучше не трогать. При уменьшении его начинает отваливаться WiFi, котроллер впадает в кому и через несколько секунд/минут перегружается. У меня на гирлянде большой на балконе - если до 15 не уменьшить - текст прыгает и рассыпается. Если уменьшить - начинаются периодические перезагрузки и отваливания WiFi...

Вот такие постоянные переподключения у меня только с одним из публичных бесплатных серверов были.
Может в настройках сервера что-то не так? Например разрывать соединение после.... ответа? после маленького таймаута? Если поймете что не так - расскажите. Будет еще один кирпичик в базу знаний.

sysardex, 2020-12-23:
Потерь нет, пинги идеальные. Wifi не переподключается при этом. mqtt сервер внутри сети, это тот что ставится на hassio (Home assistant) в докере. У него куча других mqtt клиентов при этом и никто не отваливается. Тестовые консольные mqtt клиенты коннектятся/подписываются на топики нормально, их не рубит. Попробую поиграться с параметром так как заметил что через 10-15 минут работы еспшка ребутится.

sysardex, 2020-12-23:
Указал WAIT_TIME = 25, картинку еще не проверял но переподключения пропали. Но ESP.restart() примернно раз в 10-15 минут остался.

vvip-68, 2020-12-23:
Ну убери этот ESP.restart() в a_main.ino строки 24-29
Это был эксперимент. У мня удаленная гирлянда - хотел наблюдать за ней по MQTT что там происходит.
У нее этот WAIT_TIME вообще до 15 снижен, рассыпания текста уже при 20 начинаются.
В таком режиме у нее WiFi Начинает часто отваливаться. И все рывками начинает идти - движение, секунды 2-3-4 все замирает. Снова движение и опять все замирает. Это она в этот момент пытается в инет вылезть либо время/погоду взять, либо MQTT соединиться. Время цикла которое там проверяется  - увеличивается до 2-3-4 секунд при норме 25 мс...
В этот момент пробовал ребутить, чтобы хоть как-то работало. На нормальных домашних часах в сети это ника не сказывается - работают да работают себе сутками напролет. По факту все равно не помогло. Съездил-посмотрел.
Соединяется, первый кадр эффекта, пауза, ребут. И так по кругу.... Что с ней делать - не знаю уже. Приношу контроллер домой тот - всё работает стабильно. Разница в провайдерах. Там - Дом.ру, дома местный городской Telecoma.
То ли провайдер дом he такой хреновый...

AndreyLyamkin, 2020-12-23:
я вчера запустил панель на балконе. Были самопроизвольные всплески. Пробовал напряжение уменьшать - не помогло. Уменьшил "WAIT_TIME" до 10 и все нормализовалось. И отваливаний нет

closed 2020-12-24 by vvip-68

KirillAstavin, 2020-12-24:
Здравствуйте! Как тут связаться с автором? Нужна платная консультация по настройке скетча.

vvip-68, 2020-12-24:
Ну так спрашивайте. Или если форум Гайвера читаете или комментарии к его проектам GyverMatrixBT или Рюкзак,  - там Иван Иванов отвечает - напишите ему в личку в ВКонтакте, если тут писать не хотите. Чего-нибудь посоветует, может...

KirillAstavin, 2020-12-24:
Хотелось бы связаться на прямую с автором, думаю так будет на много быстрее. Мне нужно убрать не нужные эффекты , настроить mqtt, так как по инструкции есть не состыковочки и не получается. Да и нет времени самому разбираться. Ну и отблагодарить естественно!  И есть ещё пару вопросов. Напишите мне в ВК если есть возможность. Благодарю! Мой вк https://vk.com/k.astavin