vvip-68/GyverPanelWiFi #74: Не подключается после перезагрузки

ledskl, 2020-12-24:
Доброго времени суток. Проблема в следующем: полностью настраиваю плату, соединение с роутером и выделенный IP настроены и работают, до того момента пока не отключишь матрицу от сети или перезагрузишь NodeMCU. Потом приложение не хочет подключатся никак: ни сразу, ни через три минуты (таймаут в скетче видел) и позже. Что можете посоветовать? Платы NodeMCU V.3, есть разные, но все так ведут себя. Со скетчем GyverMatrixWiFi с Вашей страницы, платы работает без проблем, после ребута все коннектится.

vvip-68, 2020-12-24:
В этот проект код соединения с сетью перенесен из GyverMatrixWiFi - без изменений, насколько я помню. Даже не знаю что посоветовать - смотрите что пишет в мониторе порта. 
Я сталкивался пару раз с роутерами - пишешь при настройке соединения что адрес устройства будет, к примеру, 192.168.0.100 - после перезагрузки платы она получает адрес 192.168.0.114.  Вот не хочет почему-то, чтобы адрес 100 был - упорно на 114 перебрасывает. Может у вас тот же случай - смотрите в мониторе порта какой реально адрес получает ваша панель. Или можно 4-раза на кнопку нажать - она адрес на матрицу выведет...

closed 2020-12-24 by vvip-68

ledskl, 2020-12-25:
......................................................................
................................................................................
.............................................................22:19:58.786 -> ...................
22:20:46.371 -> ................................................................................
22:24:06.720 -> ......................................
22:25:40.254 -> Запрос текущей погоды
22:25:40.359 -> Погода получена: Yandex
22:25:40.359 -> Сейчас: пасмурно, 0ºC
22:25:40.359 -> Код иконки: 'ovc'
22:25:40.359 -> Цвет неба: '#162747'
22:25:40.359 -> 
22:25:41.821 -> ..........................................
22:27:26.916 -> ......................................

Вот что в мониторе порта за 10 минут после включения.  Нет IP вообще, но все настроенные режимы в демо крутятся, погода подгрузилась. 
И попутный вопросик: как работает дата и прогноз вместе с часами, галки ставлю, сохраняю и ничего.

vvip-68, 2020-12-25:
Из скриншота ничего не видно. Это какой-то кусок из середины.
Вся важная информация в данном случае - с начала вывода лога.
Раз получена погода - значит соединение с интернетом есть.
Точки - скорее всего пытается подключиться к MQTT-серверу, а настройки соединения для него указаны неверные.
Больше по этому куску лога ничего сказать не могу