vvip-68/GyverPanelWiFi #75: Проблема с прошивкой.

araneon, 2020-12-24:
Добрый день.
Уважаемый @vvip-68, не подскажите в чём моя ошибка (сразу скажу, что опыта с Arduino в принципе нет).
Пытаюсь загрузить в ESP-WROOM-32 ваш проект, изменил в **a_def_hard.h** параметры для своего случая на такие:

// Профиль устройства, под которое выполняется компиляция и сборка проекта
#define DEVICE_ID   **7**

// ================== Для ESP32 ====================

#if (DEVICE_ID == 7)   **// добавил сам, поправьте если не правильно сделал.**
/*
 * NodeMCU ESP32
 * Физическое подключение:
 * Матрица зигзаг, левый нижний угол, направление в право
 * Пин ленты    - 13
 * Пин кнопки   - 15
 * MP3Player    - 17 к RX, 16 к TX плеера 
 * TM1637       - 33 к DIO, 32 к CLK
 * SD-карта     - 5 - CS, 18 (CLK), 19 (MISO), 23 (MOSI)  - у SD-card wemos d1 shield - D8 (CS), D5 (CLK), D6 (MISO), D7 (MOSI)
 * В менеджере плат выбрано "ESP32 Dev Module"
 */ 
#if defined(ESP32)
#define WIDTH 18              // ширина матрицы
#define HEIGHT 13             // высота матрицы
#define DEVICE_TYPE 1         // Использование матрицы: 0 - свернута в трубу для лампы; 1 - плоская матрица в рамке   
#define MATRIX_TYPE 0         // тип матрицы: 0 - зигзаг, 1 - параллельная
#define CONNECTION_ANGLE 0    // угол подключения: 0 - левый нижний, 1 - левый верхний, 2 - правый верхний, 3 - правый нижний
#define STRIP_DIRECTION 1     // направление ленты из угла: 0 - вправо, 1 - вверх, 2 - влево, 3 - вниз
#define USE_SD 0              // поставьте 0, если у вас нет SD-card модуля
#define USE_MP3 0             // поставьте 0, если у вас нет звуковой карты MP3 плеера
#define USE_TM1637 0          // поставьте 1, если используется дополнительный индикатор TM1637, 0 если индикатора нет
#define USE_POWER 0           // 1 - использовать отключение питания матрицы через MOSFET; 0 - не использовать 
#define USE_MQTT 1            // 1 - использовать управление по MQTT-каналу; 0 - не использовать 
#define USE_WEATHER 0         // 1 - использовать получение информации о текущей погоде; 0 - не использовать 

#define A_DEF_PASS 0          // 1 - Настройки MQTT и API KEY OpenWeatherMap в отдельном файле a_def_pass.h
                              // 0 - Настройки MQTT и API KEY OpenWeatherMap в скетче в a_def_soft.h в строках:
                              // Смотри комментарий к блоку DEVICE_ID == 0

// Схема подключения для ESP32 - в папке Schemes, файл 'ESP32+ALL.JPG'
 
#define LED_PIN (13U) // (13U)м // пин ленты, физически подключена к пину D13 на плате
#define PIN_BTN (15U)         // кнопка подключена сюда (PIN --- КНОПКА --- GND)
#define POWER_PIN (26U)       // G26 управляющий пин вкл/выкл матрицы через MOSFET или реле
#define SD_CS_PIN (5U)        // G5 пин выбора (ChipSelect) SD карты. Также SD карта использует G18 (CLK), G19 (MISO), G23 (MOSI) // на SD-shild: D8 (CS), D5 (CLK), D6 (MISO), D7 (MOSI)

#define SRX (16U)             // G16 RX пин ESP32, подключен в TX пин (3) модуля DFPlayer
#define STX (17U)             // G17 TX пин ESP32, подключен в RX пин (2) модуля DFPlayer
#define DIO (33U)             // TM1637 display DIO pin
#define CLK (32U)             // TM1637 display CLK pin

#endif
#endif     // добавил сам

// =======================================================

У меня пока кроме самой платы ESP-WROOM-32, блока питания и ленты WS2812, больше ничего нет, по этому в настройках выше вроде всё лишнее (что смог понять :)) отключил.
При попытке загрузить проект в плату, получаю ошибку **'WIDTH' was not declared in this scope** и открывается страница g_life, где подсвечивается 1-ая строка (#define  BYTE_WIDTH  (WIDTH / 8)             // Массив должен быть кратен 8 битам (сколько байт по ширине))
Не подскажите что нужно ещё поправить?

Весь лог из Arduino IDE выложил на [Google Диск](https://drive.google.com/file/d/1UlQx-7fva4hRzptuRxy_ceTw2WepmVU8), так как сюда не помещается.

vvip-68, 2020-12-24:
Для ESP32 там отдельный блок. В нем свои параметры поправь. И не надо для него никаких #If (DEVICE_ID) писать
Ты или написал и сам DEVICE_ID 7 d в але не поставил или... Или лишний #endif в конце поставил.
В общем структура кода у тебя нарушена после твоих правок

araneon, 2020-12-24:
Спасибо большое, поправил, теперь скетч грузится, но есть какие то другие ошибки.

In file included from sketch\a_def_hard.h:423:0,
                 from C:\Christmas_WINDOW\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.08\GyverPanelWiFi_v1.08.ino:20:
C:\Soft\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003
 #    pragma message "FastLED version 3.003.003"
                     ^
In file included from C:\Soft\Arduino\libraries\FastLED/FastLED.h:65:0,
                 from sketch\a_def_hard.h:423,
                 from C:\Christmas_WINDOW\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.08\GyverPanelWiFi_v1.08.ino:20:
C:\Soft\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
И в итоге работает только первые 6 полос по 13 диодов, а остальные 12 полос молчат.
Пробовал менять угол подключения, не помогло. (подключил, как "правый нижний"), если я смотрю на диоды, то точка подключения получается с права внизу и дальше змейкой (в верх - в низ) уходит справа налево.

vvip-68, 2020-12-25:
1. Это не ошибки, это информационные сообщения.
2. Ищи там где горит последний / не горит первый непропай, соплю или один из этих  диодов сгоревший.
При условии что размеры матрицы указаны правильно.

sanyafifa, 2020-12-25:
> P.S. Походу косяк с подключением на ленте, завтра буду перепаивать.

Прозвоните диоды последний который работает и тот который не работает. "Короткого" между любыми контактами быть не должно.

closed 2020-12-25 by vvip-68

araneon, 2020-12-25:
Лента в порядке, подключил контроллер T-1000S, все диоды горят, режимы меняются, а вот с ESP не хочет, как в прочем и с китайской Arduino NANO, на которой ранее эта лента полноценно работала.

vvip-68, 2020-12-25:
Вывод - лента впорядке, не работает микроконтроллер ESP или неправильно собрана схема.
Для Arduino Nano этот проект вообще не предназначен.
Проверяйте правильность сборки схемы.