vvip-68/GyverPanelWiFi #76: Бегущая строка

Des-75, 2020-12-25:
Здравствуйте.
Версия скетча "WiFiPanel-v.1.08.2020.1223-beta7".
Настройки бегущей строки:
0 #123
1 {B#000000} {C#00FF00} {D:dddd} {C#0000FF} {D:dd} {D:MMM} {C#FF0000} {D:HH:mm} {#3}
2 {B#000000} До Нового Года {R01.01.2021 0:00#Z}{#3}
3 {B#000000} На улице {WT} °С
4 {B#000000} С Днём Энергетика! {S22.12.****}{-}
5 {B#000000} С Рождеством! {S07.01.****}{-}
..
Z {B#000000} С Новым Годом!{S01.01.**** 0:00#14.01.**** 23:59}{-}

1. Столкнулся с тем, что время в бегущей строке #1 неверно отображает формат часа в виде HH (т.е. от 00 до 23). Сначала было нормально, потом обратил внимание, что время 9 вечера, а час показывает как 09, а не 21. Сменил на hh, ничего не поменялось (естественно), вернул обратно на HH, опять ничего не изменилось. Удалил все строки, ввёл заново - стало нормально. На следующий день, опять та же песня! Ребут тоже не помог.
2. Строки #4 и #5 не работают как заявлено. Они или не отображаются вообще (в текущей конфигурации, и в т.ч. когда нотификатора {-} не было), или отображаются всегда (когда строка #0 имела вид #12345 (нотификатор {-} отсутствовал). ЧЯДНТ?

vvip-68, 2020-12-25:
1. Не смог воспроизвести - всегда показывается в правильном формате - если РР - в 24-х часовом, если hh - в 12-часовом. По коду тоже не вижу каких-то проблем или как оно может работать неправильно.
2.  Посмотрю. Вероятно для указанной последовательности - есть косяк. Режим последовательного воспроизведения строк был написан давно, после него многое что добавлялось, что, вероятно не учитывается. И вообще этот режим крайне редко используется. Посмотрю, исправлю что не та.
3.  Макрос R не работает со звездочками в дате - только с конкретной датой. Иначе сразу после нового года (если год указан звездочками) стало бы писаться - "До нового года 365 дней", а это не то, что мне нужно. Мне нужно, чтобы писалась строка-заместитель. А если звездочка в числе или месяце - вообще непонятно что с эти делать и как считать остаток дней.

vvip-68, 2020-12-25:
Я исправил косяки по п.2. Прошу заметить, если даже с текущая дата попадает в период, определяемый макросом {S}, но строка отключена макросом {-} или '-' в начале строки - она показываться не будет.
Также если строка 3 - НЕ отключена макросом {-} - она будет показываться как и все остальные строки.
Если отключена, но указана как дополнительная строка макросом {#3} - она будет показана, даже если отключена

Ну и по п.1 - ни разу не проявилось и не вижу причин, чтобы проявлялось, судя по коду...

Des-75, 2020-12-26:
п.1 попробую ещё поковырять, возможно найду закономерность.
п.2 проверю.
п.3 Да, я понимаю как это работает. Вопрос был не в этом, но ладно.

Строка 3 сделана без {-} сознательно.

closed 2020-12-26 by vvip-68