vvip-68/GyverPanelWiFi #77: не подключается после перезагрузки 2

ledskl, 2020-12-25:
Я очень извиняюсь за назойливость, но стало еще страннее:

......................................................................
................................................................................
.............................................................22:19:58.786 -> ...................
22:20:46.371 -> ................................................................................
22:24:06.720 -> ......................................
22:25:40.254 -> Запрос текущей погоды
22:25:40.359 -> Погода получена: Yandex
22:25:40.359 -> Сейчас: пасмурно, 0ºC
22:25:40.359 -> Код иконки: 'ovc'
22:25:40.359 -> Цвет неба: '#162747'
22:25:40.359 ->
22:25:41.821 -> ..........................................
22:27:26.916 -> ......................................

Вот что в мониторе порта за 10 минут после включения. Нет IP вообще, но все настроенные режимы в демо крутятся, погода подгрузилась.
И попутный вопросик: как работает дата и прогноз вместе с часами, галки ставлю, сохраняю и ничего.

ledskl, 2020-12-25:
Кстати вот скрин с приложения на вкладке Подключения 
[image]
Потом могу смахнуть окно и заполнить данные.

ledskl, 2020-12-25:
Андроид 10 чистый(ONE который)

vvip-68, 2020-12-25:
Да, есть ошибка в приложении, когда передаваемый в приложение IP it задан - то есть программа не подключилась к роутеру. Забыл поправить - поправлю.

ledskl, 2020-12-25:
А по данным монитора порта есть мысли, подскажете?

vvip-68, 2020-12-25:
Я же написал - по этому куску вырванному из середины видно только что
а) соединение с интернетом есть - погода получается
б) к MQTT серверу подключиться не может - видимо неправильно настроены учетные данные для входа в MQTT

Больше из куска вырванного из середины ничего не видно. Вся важная информация - в начале лога, который не приведен

vvip-68, 2020-12-25:
Дата и прогноз работают только с малыми часами. У вас я не знаю какие часы включены - не видно.
Дату в больших часах думаю сделаю чуть позже, когда будет время...

ledskl, 2020-12-25:
Нет, это весь лог с самого начала.
Часы маленькие конечно

ledskl, 2020-12-25:
Точки  в логе идут с момента включения в сеть

vvip-68, 2020-12-25:
Вот начало лога:
```
03:08:18.537 -> WiFiPanel-v.1.08.2020.1223-beta7
03:08:18.571 -> Host: 'WiFiPanel-0' >> 48x16
03:08:18.571 -> 
03:08:18.571 -> Загрузка строк выполнена.
03:08:18.571 -> Ипользованы адреса EEPROM 800 - 2227
03:08:18.571 -> Свободно ячеек 1868
03:08:18.571 -> Сохранение строк выполнено.
03:08:18.571 -> Ипользованы адреса EEPROM 800 - 2227
03:08:18.571 -> Свободно ячеек 1868
03:08:19.427 -> 
03:08:19.427 -> Подключение к AlvaNet
03:08:19.462 -> ..........................................
03:08:30.016 -> WiFi подключен. IP адрес: 192.168.0.98
03:08:30.016 -> 
03:08:30.016 -> UDP-сервер на порту 2390
03:08:30.016 -> NTP-сервер ru.pool.ntp.org -> 78.36.18.184
03:08:30.016 -> Отправка NTP пакета на сервер ru.pool.ntp.org
03:08:30.016 -> 
03:08:30.016 -> Запрос текущей погоды
03:08:30.223 -> Погода получена: Yandex
03:08:30.223 -> Сейчас: ясно, -25ºC
03:08:30.223 -> Код иконки: 'skc-n'
03:08:30.223 -> Цвет неба: '#000917'
03:08:30.223 -> Темное время суток
03:08:30.258 -> UDP << ip='78.36.18.184:123'; ntp sync
03:08:30.258 -> UDP пакeт размером 48
03:08:30.292 -> Разбор пакета NTP
03:08:30.292 -> Секунд с 1970: 1608952110
03:08:30.292 -> Текущее время: 26.12.2020 03:08:30
```
У вас этого нет в приведенном выгрызке. То есть - это не весь лог

ledskl, 2020-12-25:
нету этого увы

vvip-68, 2020-12-25:
Значит у вас не моя прошивка или сильно исправленная. По другим я не могу помочь...

ledskl, 2020-12-25:
......................................................................
................................................................................
.............................................................22:19:58.786 -> ...................
22:20:46.371 -> ................................................................................
22:24:06.720 -> ......................................
22:25:40.254 -> Запрос текущей погоды
22:25:40.359 -> Погода получена: Yandex
22:25:40.359 -> Сейчас: пасмурно, 0ºC
22:25:40.359 -> Код иконки: 'ovc'
22:25:40.359 -> Цвет неба: '#162747'
22:25:40.359 -> 
22:25:41.821 -> ..........................................
22:27:26.916 -> ................................................................................
22:30:47.244 -> ................................................................................
22:34:07.604 -> ................................................................................
22:37:27.809 -> .............................................................................NTP-сервер ru.pool.ntp.org -> 188.225.9.167
22:40:40.293 -> Отправка NTP пакета на сервер ru.pool.ntp.org
22:40:40.293 -> 
22:40:40.293 -> Запрос текущей погоды
22:40:40.363 -> Погода получена: Yandex
22:40:40.398 -> Сейчас: пасмурно, 0ºC
22:40:40.398 -> Код иконки: 'ovc'
22:40:40.398 -> Цвет неба: '#132340'
22:40:40.398 -> 
22:40:40.398 -> UDP << ip='188.225.9.167:123'; ntp sync
22:40:40.398 -> UDP пакeт размером 48
22:40:40.398 -> Разбор пакета NTP
22:40:40.398 -> Секунд с 1970: 1608936038
22:40:40.398 -> Текущее время: 25.12.2020 22:40:38
22:40:40.398 -> --------------------
22:40:40.640 -> ...
22:40:48.143 -> ................................................................................
22:44:08.507 -> ................................................................................
22:47:28.718 -> ................................................................................
22:50:49.053 -> ................................................................................
22:54:09.377 -> .....................................
22:55:40.303 -> Запрос текущей погоды
22:55:40.490 -> Погода получена: Yandex
22:55:40.490 -> Сейчас: пасмурно, 0ºC
22:55:40.490 -> Код иконки: 'ovc'
22:55:40.490 -> Цвет неба: '#111f3a'
22:55:40.490 -> 
22:55:41.965 -> ...........................................
22:57:29.586 -> ................................................................................
23:00:49.899 -> ................................................................................
23:04:11.665 -> ................................................................................
23:07:32.581 -> ...........................................................................NTP-сервер ru.pool.ntp.org -> 94.247.111.10
23:10:40.334 -> Отправка NTP пакета на сервер ru.pool.ntp.org
23:10:40.334 -> 
23:10:40.334 -> Запрос текущей погоды
23:10:40.437 -> Погода получена: Yandex
23:10:40.437 -> Сейчас: пасмурно, 0ºC
23:10:40.437 -> Код иконки: 'ovc'
23:10:40.437 -> Цвет неба: '#0e1b33'
23:10:40.472 -> 
23:10:40.472 -> .UDP << ip='94.247.111.10:123'; ntp sync
23:10:40.472 -> UDP пакeт размером 48
23:10:40.507 -> Разбор пакета NTP
23:10:40.507 -> Секунд с 1970: 1608937838
23:10:40.507 -> Текущее время: 25.12.2020 23:10:38
23:10:40.507 -> --------------------
23:10:43.001 -> ....
23:10:53.002 -> ............

вот с начала по настоящее время
Прошивка ваша 1,08, ничего кроме параметров матрицы не правил

vvip-68, 2020-12-25:
Нет самого первого вывода даже - в котором выводится версия прошивки

ledskl, 2020-12-25:
и я про тоже, все что есть, это и странно

vvip-68, 2020-12-25:
Нажмите прямо сейчас кнопку Reset на контроллере

ledskl, 2020-12-25:
Со второго ребута лог пошел без всяких точек, Вы были правы роутер сам раздает IP, не тот что в приложении.
Спасибо огромное за терпение и помощь. А с малыми часами и календарем?

ledskl, 2020-12-25:
Может USB кабель кривой.

vvip-68, 2020-12-25:
При чем тут кабель?
А что с малыми часами и календарем? Для вывода календаря и температуры нужно как минимум 11 строку в высоту. У вас есть место для вывода на матрице?

ledskl, 2020-12-25:
Понятно, у меня десять. Всё, еще раз БОЛЬШОЕ СПАСИБО и с Наступающим Вас

ledskl, 2020-12-25:
Про кабель это я про кривой лог.

vvip-68, 2020-12-25:
Если б кабель был нерабочим - вообще бы ничего не призодило в монитор порта. И прошить бы вы не смогли контроллер

closed 2020-12-25 by vvip-68

vvip-68, 2020-12-25:
Я обновил приложение - теперь не должно вылетать как на скриншоте выше

ledskl, 2020-12-25:
Ок, попробую