vvip-68/GyverPanelWiFi #78: Возможность отключить один или несколько светодиодов

shostko, 2020-12-26:
@vvip-68 не могли бы вы добавить в прошивку возможность отключить один или несколько светодиодов? (не для постоянной регулировки, а вообще на уровне кода)
 Т.е. чтобы в коде была возможность прописать индексы светодиодов, и они просто никогда не включались. При этом никак не затрагиваю логику эффектов и вкл/выкл соседних диодов.

  Или может быть вы могли бы направить меня в правильном направлении? 
  пытался выставить `leds[1] = CRGB::Black;` перед каждым вызовом `FastLED.show();` - не помогает, всё равно все светодиоды горят. пробовал так же вариацию с функцией для определения позиции по x,y

vvip-68, 2020-12-26:
Ну, наверное что-то не так делаете.

>leds[1] = CRGB::Black; перед каждым вызовом FastLED.show();

или в функции где вывод в позицию X,Y - верный путь и не может не работать. Именно так - заданием цвета светодиода с определённым индексом (с определенными координатами) и формируется отображение.

А от чего такие странные желания?

shostko, 2020-12-27:
у родственников вчера вешал такую же гирлянду на балкон. толи что-то я накосячил при пайке/подключении, толи брак - если один светодиод горит, то все остальные сходят с ума и моргают. если его не включать - с остальными все норм. (проверял через рисовалку). решил не прыгать с пояльником под потолок на балконе (успел все приклеить), а программно выключить один диод. (поэтому и вариант с закрашиванием не катит))

через FindAndReplace заменил все `FastLED.show();` на `leds[getPixelNumber(0,1)] = CRGB::Black;FastLED.show();`, но что-то не увидел ни одного путухшего диода.
это верный подход @vvip-68? пойду сейчас на тестовом стенде на столе пробовать.

shostko, 2020-12-27:
вариант в принципе нужен временный. после праздников всё-равно все сниму и аккуратно на столе проверю/перепаяю или заменю диод

vvip-68, 2020-12-27:
Вариант верный. Не вижу причин почему бы он не работал.
По твоему коду - это левый нижний угол 0-я колонка (X), 1-я строка (Y).
Вне зависимости от указанного угла подключения / направления.
Тут же всё верно?

vvip-68, 2020-12-27:
@shostko - проверил на стенде - гаснет диод. То есть подход рабочий.

shostko, 2020-12-27:
спасибо, @vvip-68 
проверил сейчас на контролере и матрице у себя - всё работает и с x,y (надо было правда другие координаты указывать. я предполагал, что они относительно начала матрицы по настройкам. а они с левого нижнего угла), и просто 1 (у меня левый верхний угол второй диод в первой колонке)

либо я вчера не доглядел, либо там хардварно нетолько в этом проблема.

closed 2020-12-27 by shostko