vvip-68/GyverPanelWiFi #79: Oтображение текста и часов

gribo4ik91, 2020-12-26:
Отображение текста и часов для режиме  "Гирлянда на балконе V2" осуществляется в не зеркальном режиме то есть текст и часы читаемы из дома о не с улицы.   

Подключение:
#define DEVICE_TYPE 1         // Использование матрицы: 0 - свернута в трубу для лампы; 1 - плоская матрица в рамке   
#define MATRIX_TYPE 0         // тип матрицы: 0 - зигзаг, 1 - параллельная
#define CONNECTION_ANGLE 0    // угол подключения: 0 - левый нижний, 1 - левый верхний, 2 - правый верхний, 3 - правый нижний
#define STRIP_DIRECTION 1     // направление ленты из угла: 0 - вправо, 1 - вверх, 2 - влево, 3 - вниз

fastdron, 2020-12-26:
Задайте угол подключения 3 (правый нижний) и все будет как надо. Угол следует определять смотря на матрицу с улицы а не из квартиры...

gribo4ik91, 2020-12-26:
Спасибо все работает. Моя вина.

closed 2020-12-26 by gribo4ik91