vvip-68/GyverPanelWiFi #81: Гирлянда на окно

Junior220584, 2020-12-27:
Всем привет!
Заранее благодарен, если сможете подсказать по вопросам ниже.
Имеется гирлянда 24х11, к ней подключена ESP-12F. На плату заливается прошивка gyverpanelwifi v1.08 от vvip. Все почти замечательно работает)), но есть ряд вопросов:
1. Не получается сконнектить с SD card адаптером. Подключаю к D5-D8. Монитор порта не инициализирует карту, хотя отдельно все работает.
2. Не разобрался как запустить часы и температуру. Оверлеем поверх эффектов запускались, а в виде отдельного эффекта нет. Ткните носом, пожалуйста.
3. И при подключении сенсорной кнопки, ESP вообще не стартует. Тоже не понятно почему.

vvip-68, 2020-12-27:
1. С картой не подскажу. Приведенная распиновка работает на Wemos d1 mini с выбранным в настройках компилятора NodeMCU v1.0.  Если другая плата (не Wemos d1 mini, а тем более чистый ESP-12F) или в настройках  компиляции выбрано не NodeMCU - надо искать / смотреть соответствие пинов этих плат пинам в указанной комбинации. Ориентир - смотрите куда на вашей плате аппаратно выведены пины MOSI, MOSI, CLK и их соответствие D5-D7 в подключении выше. Вместо D8 (Chip Select) можно использовать любой другой.
2. Для запуска часов отдельным эффектом на вкладке Эффекты в приложении нужно выбрать эффект "Часы". Температуры отдельным эффектом не предусмотрено - но можно отображать их в бегущей строке.
В данной версии прошивки температура отображается только в режиме "малые часы". Для больших - не сделал. Может быть сделаю.
3. Для подключения сенсорной кнопки нужно выбирать пины осторожно. Вы ее скорее всего повесили (в нотации комбинации Wemos/NodeMCU) на D4, который является отражением пина GPIO2 микроконтроллера ESP8266.
GPIO2 совместно с GPIO0 и GPIO15 при старте микроконтроллера определяют режим запуска микроконтроллера. Если на GPIO2 при старте подается высокий уровень - микроконтроллер переключается в режим ожидания загрузки прошивки. Вполне понятно что в этом режиме выполнения программы не происходит.

closed 2020-12-27 by vvip-68

Junior220584, 2020-12-28:
vvip-68, cпасибо за ответ!
А может подскажете еще?)
1. ПИНы (MOSI, MISO, CLK) под SD где-то прописываются в прошивке или они аппаратно зарезервированы?
2. У меня к сожалению нет андроида, поэтому не смогу выбрать часы с телефона. Хотел все в прошивке скорректировать, но не разобрался как. Есть какие-то варианты?
3. По кнопке понял, буду пробовать другие пины.
4. Можете еще подсказать как добавить текстовую бегущую строку, с указанием вкладок, в которых нужно провести корректировки? С описанием, которое выложено, к сожалению, не разобрался( 

Заранее спасибо!!!

vvip-68, 2020-12-28:
1. Аппаратно зарезервированы
2. Глобальная переменная thisMode - текущий режим. Константа режима часов - MC_CLOCK
4. Вообще не понял вопроса. Бегущую строку отдельным эффектом? - Все вкладки, переписывая весь скетч.

Junior220584, 2020-12-28:
Ок, спасибо!
По третьему вопросу: да, думал, что между эффектами можно пустить бегущую строку, типа: С Новым Годом! Это было в прошивке Гайвера. Подумал, что у вас это как-то аналогично реализовано.

vvip-68, 2020-12-28:
Нет, здесь бегущая строка пробегает поверх эффекта через заданные промежутки времени. Какая именно строка - выбирается или по порядку из набора строк, или в указанной последовательности или одна из случайно выбранных. 
В тексте строки можно указать поверх какого эффекта показывать (фон строки в виде эффекта) или указать фон нужного цвета. В отдельном эффекте "Бегущая строка" не вижу абсолютно никакого смысла.

Junior220584, 2020-12-28:
Понял) Спасибо за ответы!!!