vvip-68/GyverPanelWiFi #82: Текст, бегущая строка по запросу из вне (mqtt)

sanyafifa, 2020-12-27:
Хочу поблагодарить за MQTT реализацию. все работает.
Мысли по расширению функционала проекта. 

_Можно ли сделать бегущую строку, которая будет показываться только когда приходит в mqtt топик в определенный период времени?_ 

Объясню идею, есть сервер, в нем есть куча данных для уведомлений:
- сработала сигнализация
- поднялся уровень CO2
- позвонили в звонок
- упала температура обогревателя
- пришло письмо на почту

И вот это хотелось бы вывести на панель только когда это актуально (не ждать когда эффект с бегущей строкой наступит).
Поправьте если было уже обсуждение

vvip-68, 2020-12-27:
Обсуждения не было. С ходу - вариант реализации на том что есть - 
при наступлении такого события имитируются действия как в приложении (командами, посылаемыми по MQTT-каналу):
1. Записать текст строки в какую-нибудь ячейку. С макросом {-} или э-э в начале строки, чтобы не показывалась просто по перебору.
2. Имитировать нажатие кнопки "Просмотр" - при этом строка игнорирует наличие макроса "отключено" и показывается на экране сразу же при поступлении команды. В строке могут быть макросы "показывать N раз"  или "Показывать N-секунд"

А вообще - кто ж смотрит на строку постоянно? Тогда надо еще какой-то звуковой сигнал прикручивать и запуск сигнала.
Опять же в варианте реализации c USE_MP3 == 1 - есть команды - начать воспроизведение звука / закончить воспроизведение.

То есть уже на имеющемся наборе это всё реализуемо. Но можно, конечно, и отдельную команду придумать - получить строку и воспроизвести ее никуда не сохраняя.

vvip-68, 2020-12-27:
Или заранее забить в набор ячеек набор сообщений, а потом просто отправлять команду "Просмотр" для определенной ячейки из набора.

sanyafifa, 2020-12-27:
> 2\. Имитировать нажатие кнопки "Просмотр" - при этом строка игнорирует наличие макроса "отключено" и показывается на экране сразу же при поступлении команды.

вот за это я и забыл. т.е. выключи все эффекты и прокрути текст. 

> А вообще - кто ж смотрит на строку постоянно? Тогда надо еще какой-то звуковой сигнал прикручивать и запуск сигнала.

Это уже полет фантазии. У меня есть обычная блютуз колонка которая подключается к малинке (что хочешь играй). А у людей и Алиса и ТВ умные...

 Я себе так алгоритм представляю:
заливаем цветом -> ждем сек -> черный экран -> ждем -> строка -> демо режим

vvip-68, 2020-12-27:
На имеющемся - забиваем строку с непечатными символами - которым нет буквы в наборе символов шрифта. В эту строку пишем макрос цвета фона, например {B#FF0000} - красный фон. Можно забить макрос "Играть N секунд" - отправляем команду "играть". далее через N времени - даем команду "играть " уже содержательной строке. Новая строка прервет предыдущую и начнет воспроизводиться.

vvip-68, 2020-12-27:
А можно еще макрос звука придумать, который в строку вставляется - "Играть звук N из папки M" на карте dfPlayer`a. Тогда при начале показа строки будет срабатывать макрос звука и начинать его воспроизводить. По окончании строки - глушиться...

closed 2020-12-28 by vvip-68

DimUrm, 2020-12-30:
> Но можно, конечно, и отдельную команду придумать - получить строку и воспроизвести ее никуда не сохраняя.

Вот это было бы супер