vvip-68/GyverPanelWiFi #84: Опечатка

AlexSZbrn, 2020-12-27:
https://github.com/vvip-68/GyverPanelWiFi#%D1%81-%D1%83%D0%BF%D1%80%D0%B0%D0%B2%D0%BB%D0%B5%D0%BD%D0%B8%D0%B5%D0%BC-%D0%BF%D0%B8%D1%82%D0%B0%D0%BD%D0%B8%D0%B5%D0%BC-%D0%BC%D0%B0%D1%82%D1%80%D0%B8%D1%86%D1%8B-%D1%87%D0%B5%D1%80%D0%B5%D0%B7-%D1%80%D0%B5%D0%BB%D0%B5
В предложении "Схемы управления питанием для P-канального MOSFET IRF4905 и N-анального MOSFET IRF3205 приведены ниже" опечатка по Фрейду? Или эти полевики и правда так через задн цу работают? ;-)

AlexSZbrn, 2020-12-27:
А проект неплох, к следующему НГ буду копать чуть глубже)))

vvip-68, 2020-12-28:
> Или эти полевики и правда так через задн цу работают? ;-)

Китайские - да. Сопротивление открытого не соответствует заявленным параметрам обычно. Уже при 10 амперах дико греются - нужен массивный радиатор. Ну или не подделка. Или вообще лучше обыкновенное реле ставить.

closed 2020-12-28 by vvip-68