vvip-68/GyverPanelWiFi #85: WiFiPanel-v.1.08 бегущая строка вверх ногами

Elitesnew, 2020-12-28:
Поставил на тел, настроил все кроме бегущей строки(

Подскажите почему весь текст в бегущей строке отображается кверх ногами???
время показывает правильно, только настраиваемая строка перевернуты буквы на голову(
все что вводишь для отображения вручную текстом в приложении, все вверх ногами, даже и не могу понять куда копать(

Например Время и погода отображаются правильно, время движется тоже правильно, справа на лево.
Весь набранный текст отображается вверх ногами, но едет правильно, справа на лево.
Я уже чего только не пробовал, все типы подключения даже перепробовал, нашел даже наоборот, тоесть текст отображается правильно, а время и погода вверх ногами.
Все и везде перерыл, весь инет облазил, не могу найти как исправить
помогите(((  неделю уже мучаюсь

Elitesnew, 2020-12-28:
[image]
[image]

limittouch, 2020-12-28:
попробуй прошить не последнюю версию
там же чего-то меняли
https://github.com/vvip-68/GyverPanelWiFi/commit/cecf424c08b7ed2c8fba61046f37bbfeb2de942c

limittouch, 2020-12-28:
ИМХО
угол начала определяется с лицевой стороны
время и температуру недавно меняли https://github.com/vvip-68/GyverPanelWiFi/commit/cecf424c08b7ed2c8fba61046f37bbfeb2de942c

попробуй прошить не последнюю версию и посмотреть что будет
у меня на этой работает нормально
`#define FIRMWARE_VER F("LED-Panel-WiFi v.1.07.2020.1206")`

vvip-68, 2020-12-29:
Вот, взятая с гитхаба прошивка, в которой подставлены только размеры матрицы и угол подключения так себя ведет?
Думаю, что нет. Значит откатывайте назад что еще правили в прошивке и потом возвращайье по одному, смотрите какое исправление приводит к такому результату.

closed 2020-12-29 by vvip-68

Elitesnew, 2020-12-29:
Блин, ниче там не менял, клянусь чем угодно, я хз как так вышло, я же даже удалял и ставил заново, не помогало, не могу реально понять как так могло случиться(
было
#define MIRR_V 0                            // отразить текст по вертикали (0 / 1)
#define MIRR_H 1                            // отразить текст по горизонтали (0 / 1)

поправил и все заработало, СПАСИБО ОГРОМНОЕ!!!  неделю мучался(

vvip-68, 2020-12-29:
Тут читали?
https://github.com/vvip-68/GyverPanelWiFi/wiki/%D0%AD%D0%BA%D1%81%D0%BF%D0%BE%D1%80%D1%82-%D1%8D%D1%84%D1%84%D0%B5%D0%BA%D1%82%D0%BE%D0%B2-%D0%B0%D0%BD%D0%B8%D0%BC%D0%B0%D1%86%D0%B8%D0%B8-%D0%B8%D0%B7-%D0%9F%D0%9E-%C2%ABJinx!%C2%BB

Вот этот фрагмент видели?
[image]

Elitesnew, 2020-12-29:
нее, эту инструкцию я не видел, другие читал
Спасибо еще раз, что не бросили и не послали!
С наступающим и всего наилучшего!

Elitesnew, 2020-12-29:
там же шилдик одевается на этот вемос, не перепутать
карта на 4гб фат32
может в настройках скетча тоже надо что то настраивать?
помогите последний раз пожалуйста

vvip-68, 2020-12-29:
Нет. Только подключение к правильным соответствующим пинам

vvip-68, 2020-12-29:
Такая конструкция?
[image]

Elitesnew, 2020-12-29:
да, один в один, все пины там есть эти

vvip-68, 2020-12-29:
А в скетче эти пины не менял? Прошивка собирается с выбранной в настройках платой Node MCU?

vvip-68, 2020-12-29:
Что конкретно в логах пишет при старте прошивки? В мониторе порта

Elitesnew, 2020-12-29:
да, иоде мсу 1.0 12е, все по инструкции делал, менять что либо не меняю, если ни где не прописано
пропаял разъемы, думал может неконтакт где, не помогло(

vvip-68, 2020-12-29:
Может шилд бракованный? Или карта пустая? Других причин не вижу...

vvip-68, 2020-12-29:
Ты шилд правильно воткнул, не в перевернутом состоянии? Пины на вемосе и на шилде совпадают? D5 - в D5 и так далее?

Elitesnew, 2020-12-29:
щас попробую другую карту и попробую форматнуть в эксфат, если не поможет то это все(((
на карте есть папка с эффектом одним

vvip-68, 2020-12-29:
Нет, кондер не может мешать.

vvip-68, 2020-12-29:
У тебя в "гребенке" все пины пропаяны? А то я паял только те, что думал что нужны - земля, +5 вольт, D5,D6,D7,D8/ Остальные оставлял незапаянными - чего на них припой тратить. Не работало. Оказалось ему еще с вемосам пин +3.3 вольта нужен

Elitesnew, 2020-12-29:
прозвонил и пропаял все еще раз
кандер этот срезал
не помогло
даже не верится что шилдик бракованный
скорее всего накосячил где то(

vvip-68, 2020-12-29:
Где там можно накосячить? 6 проводков...
Залей пример работы с SD-картой, которые идут вместе с ядром...
Он тоже не работает?

Elitesnew, 2020-12-29:
взял другой вемос
прошил и он увидел карту и даже в приложении при выборе эффекта сд карта, оно видит файлы .out
но они почему то не воспроизводятся, пустой экран и все, только время бежит
блин, писец как все сложно((( куда я полез, мне уже самого себя жалко

попробую новые анимации сделать, может получится

vvip-68, 2020-12-29:
А у тебя карта надежная, не китайская?
Есть у меня китайская карточка на 8гиг. Именно так она и видится компом.
А реально на ней - 128 МБ. ТО есть, файлы пишутся, в папке их видно, список считывается. А вот открываешь на просмотр любым редактором двоичным (в том же Far`e) - там одни нули.
Соответственно такие файлы на матрице - черный экран.

vvip-68, 2020-12-29:
В комплекте проекта в папке Tools есть утилита - JinxFramer.
Открой ей файл прямо с карты и полистай фреймы. Если там черный экран - значит у тебя примерно такая карта, как я описал выше. На флэшке только оглавления файлов, а содержимого нет.

Elitesnew, 2020-12-29:
Блин, разобрался наконец с вашей помощью, уррра!!!! Даже и не знал что так можно быстро все проверить.
Спасибо! Был очень приятно удивлен, что еще есть на этой планете люди, которым реально не лень помочь, а то я уже привык, что всем пофиг на других, главное только мои проблемы.

СПАСИБО, всех благ Вам, напишите номер карты, я как получу денежку, то обязательно Вас отблагодарю! 
Очень тяжело на этом сайте что-либо найти самостоятельно!

vvip-68, 2020-12-29:
На сайте проекта есть кнопка "Спонсор" в правом верхнем углу. Вторая ссылка - карта. Первая - PayPal. Если есть желание помочь проекту - welcome и спасибо.

Elitesnew, 2020-12-29:
Очень много нюансов в сборке и не подготовленному человеку очень тяжело все найти в разных местах, я наверно войну и мир прочел уже, где только не лазил и оказывается многое не нашел, только что искал целый час где в прошивке сменить часовой пояс, для правильного времени, но нашел.
Очень тяжело, чес слово, без вашей помощи я бы точно не справился, а на том форуме гайвера мне так никто и не помог.

vvip-68, 2020-12-29:
Потому что все настройки (и часовой пояс в том числе)  проще поменять в приложении, чем искать по всей прошивке.
И смена в прошивке (для гурманов и/или гиков) требует сброса постоянной памяти и всех настроек, чтобы изменения применились, а изменения настроек через приложение - изменения применяются сразу и без проблем и вполне очевидны в отличии от поиска в скетче.

Elitesnew, 2020-12-29:
Вот к примеру меня просит собрать такое друг и он вообще не понимает в компах, но хочет просто воткнуть в розетку и чтобы все работало, и в таком случае придется изменять и все настраивать именно в прошивке.

vvip-68, 2020-12-29:
Зачем? В прошивке меняется только размеры матрицы и угол подключения, все остальное - в приложении. Сначала подключается к точке доступа PanelAP (а это всё написано на первой странице проекта), а дальше настраивается всё что нужно из приложения, включая имя/пароль/сетевой адрес и прочие часовые пояса.

Один нюанс - сталкивался пару раз - роутер не всегда выдает тот адрес, который просишь. И тут после применения сетевых настроек нужно смотреть на матрицу какой IP она получила (можно также вывести на матрицу по 4-х кратному нажатию кнопки) или в мониторе порта.

Кроме этого - в прошивке ВООБЩЕ НИЧЕГО не нужно править.

vvip-68, 2020-12-29:
А с кнопочного телефона и Айфона - увы не настроить. Можно поставить приложение на планшет (если конечно он нормальный, а не айпад какой-нибудь). Если нет совсем телефона - зайди со своего все настрой и отдай.
Но использовать все это без возможности как-то менять настройки, включать/выключать, ставить будильники и режимы по времени, менять тексты - как-то совсем грустно.

Elitesnew, 2020-12-29:
Самый большой плюс это конечно же менять тексты в бегущей строке и сами макросы, ну очень классная вещь, за день разобрался, будильник считаю лишним, врятли кто им будет пользоваться, так же как и кнопкой и играми.
У меня под окнами целая толпа собирается поглазеть и заснять, 1этаж, 644 диода с сеткой 6.5см, красотища неописуемая, но это с эффектами через контроллер т1000с и программу лед едит, я там всю красоту сделал. Встроенные эффекты в прошивке, честно говоря не очень, но теперь я все перенесу на карту в нужном формате и вот тогда будет вообще супер, все в одном!

limittouch, 2020-12-29:
> 
> Вот этот фрагмент видели?
> [image]

@vvip-68, в последней версии в приложении на android имена пишутся?
иначе осмысленные названия - сомнительное удовольствие
на карте _Rain.out_, а в приложении - _002_ и поди угадай, Who's Who

limittouch, 2020-12-29:
> Зачем? В прошивке меняется только размеры матрицы и угол подключения, все остальное - в приложении. Сначала подключается к точке доступа PanelAP (а это всё написано на первой странице проекта), а [...]

@vvip-68, в прошивке можно переписать дефолтовые настройки, чтобы роутер сразу цеплял
правда, у меня не сработало: внутренняя сеть _10.255.255_ :) а дефолты, судя по всему, только на _198.162_ рассчитаны

vvip-68, 2020-12-30:
Там на адрес в сети - ровно один единственный дефолт - желаемый адрес.
Все остальные (маска, gateway) - строятся исходя из этого адреса.

vvip-68, 2020-12-30:
@limittouch 
> @vvip-68, в последней версии в приложении на android имена пишутся?
иначе осмысленные названия - сомнительное удовольствие
на карте Rain.out, а в приложении - 002 и поди угадай, Who's Who

Да, в последней версии имена пишутся

Elitesnew, 2020-12-30:
Почти все сделал, карту памяти приложение видит, все мои 25 эффектов тоже видит
осталось последнее
как настроить чтобы ВСЕ эффекты на карте памяти перебирались и показывались?
а то приложение видит карту памяти как один эффект и все(

vvip-68, 2020-12-30:
Отключить часы - на вкладке часов снять галку "Часы поверх эффектов". Либо в каждом эффекте где часы не нужны снять такую галку. Чтобы они стояли, а не ездили - на странице настройки часов ползунок скорости этого сдвига сдвинуть до упора в лево - в ноль то есть.

Ну да, скетч так и написан - когда доходит до эффекта SD карты - выбирается случайный или какой указан. Хочется по-другому - нужно менять скетч чтобы работал по-другому.

Elitesnew, 2020-12-30:
с часами ладно, пусть уж как есть, отключать их не хотелось, думал просто задержку сделать, как с бегущей строкой, а вот с эффектами очень расстроился( 
сделал 25 эффектов а отобразить их все по порядку не могу(((
это сложно сделать в скетче?
сам могу справится?
ведь он берет эффекты что есть по порядку и проигрывает, может не сложно ему подсунуть чтобы он продолжил дальше список с карты?
какой смысл тогда вообще в карте если нет возможности отобразить все эффекты с нее(((
эх...

vvip-68, 2020-12-30:
В моем случае - чередуясь с программными эффектами эффекты с карты когда-нибудь да покажутся. Все за месяцы работы. По одному. Тебе же придется городить какой-то огород. вот в процедуре nextNandler() и prevHandler() - берется просто случайный или следующий эффект и начинает проигрываться. Если не случайно, а по порядку - просто увеличивается ID эффекта и при достижении последнего - возвращается к самому первому. Ntmt же надо переписать так, чтобы после "выпадения" эффекта MA_SDCARD брать снова SDCARD но со следующим файлом с карточки.

Когда эффект меняется (в данном случае при выпадении ID = MC_SDCARD - флаг loading ставится в true и в большом case где выбирается какую процедуру запускать - выбирается sdcardRourine()

В ней - если flag loading==true - брать случайный файл из массива загруженных или указанный.

Тебе нужно написать, чтобы при смене эффекта в nextHandler снова брался SDCARD, а при выборе файла - следующий после проигранного.

Вот и все переделки

Elitesnew, 2020-12-30:
ндауж((( я понял что ни чего не понял(
эх...
не тот уровень у меня чтобы такое провернуть

vvip-68, 2020-12-30:
Там три строчки кода.

Elitesnew, 2020-12-30:
строчек то может и не много, но написанная Вами инструкция очень страшно звучит(
просить конечно же не буду, чтобы написали и показали куда вставить
мне всего то под новый год хотелось показать все мои эффекты с карты(

vvip-68, 2020-12-30:
У тебя ВК есть? Тут неудобно длиннопостить. Долгою Напиши в https://vk.com/iivanovii

limittouch, 2020-12-31:
отключить "Использовать в демо-режиме" для всех эффектов, кроме SD-карты
в эффекте "SD карта" выбрать "Случайный эффект"

ИМХО: эффекты с SD карты как вишенка на торте: листаются стандартные, листаются, бабах, попался с SD, люди думают, что уже все видели, бабах, попался другой

Elitesnew, 2021-01-02:
limittouch
эти вишеньки на торте далеко не все испробуют)
да и убить не один день на 25 эффектов, где будет показываться только один эффект раз в 10-15 минут, да и еще не факт что разный, это точно садо мазо и того не стоит

Слава богу, что с помощью уважаемого vvip-68(к сожалению даже и не знаю как его зовут), праздник удался и все мои эффекты показываются в очередном порядке
красотища неописуемая, вот тестовый вариант, некоторые уже переделал\убрал\добавил\объединил
https://youtu.be/REOQVdSCG7E