vvip-68/GyverPanelWiFi #86: Не работает маска года

limittouch, 2020-12-28:
> Пример:
> Строка D: _До Нового года осталось {P01.01.****#E#60#60}_
> Строка E: _-С Новым годом!!!!_

для проверки устанавливал текущую дату и время + 2 минуты:
    `До Нового года осталось {P28.12.**** 15:30#E#60#60}`
событие не происходит (строка не отображается)

указываешь год - происходит
    `До Нового года осталось {P28.12.2020 15:30#E#60#60}`

vvip-68, 2020-12-29:
runningText.ino - если в строке 1659 еще нет `star_cnt = 0;` 0 поставьте

closed 2020-12-29 by vvip-68

limittouch, 2020-12-29:
> unningText.ino - если в строке 1659 еще нет `star_cnt = 0;` - поставьте

строка 1659 - комментарий в начале функции `forThisDate`, в которой star_cnt не используется

если речь про функцию `extractMacroSDates`, то там в начале **уже есть** `uint8_t  stage = 0, star_cnt = 0;`

vvip-68, 2020-12-29:
Ну, вы вероятно смотрите уже ПОСЛЕ того как я выложил исправленную версию. Там строки могли сдвинуться.

limittouch, 2020-12-29:
понял о чем речь
`case ' ':`

vvip-68, 2020-12-29:
Именно.