vvip-68/GyverPanelWiFi #88: Нет маски времени

limittouch, 2020-12-28:
При задании события нельзя указать его периодичность в часах или минутах

> Примеры: "С Рождеством!{S07.01.2020}" - один день только 7 января 2021 года (весь день с 0:00 до 23:59)

например, нужно:

весь день, каждый час
`С Рождеством!{S07.01.2020 **:00}`

весь день, каждые 10 минут
`С Рождеством!{S07.01.2020 **:*0}`

это работать не будет

vvip-68, 2020-12-29:
Да, это не предусмотрено. Это работать не будет. Этот макрос - не для этого.
Он задает допустимый диапазон отображения строки - с..по..., интервал определяется как и для всех строк - в настройках бегущей строки. Он даже по смыслу не подходит.  Возможно это должен быть отдельный макрос типа {P} - но другой - там тоже маска во времени не поддерживается и там тоже смысл несколько другой. И механизм реализации проверки события тоже совсем другой.

Может быть когда-нибудь сделаю. Не уверен в его нужности.

closed 2020-12-29 by vvip-68

limittouch, 2020-12-29:
простой пример:

> Примеры: "_С Рождеством!{S07.01.2020}_" - один день только 7 января 2021 года (весь день с 0:00 до 23:59)

**каждую минуту отображается текущее время и температура**, наступает Рождество и каждую минуту дополнительно начинает долбить _С Рождеством!_
как навязчивая реклама :)

для фразы _С Рождеством!_ хватило бы периодичности раз в час, но задать так не выйдет, т.к. отображение времени раз в час - это моветон

vvip-68, 2020-12-29:
Моветон использовать бегущую строку для отображения времени и температуры каждую минуту. Для этого есть часы и температура отображаемые поверх эффектов...

limittouch, 2020-12-29:
если матрица небольшая по высоте, часы и температура портят эффект
+ плохо видны на фоне эффекта
+ отображаются постоянно
однократный прогон бегущей строки на черном фоне с заданной периодичностью гораздо эффективнее,
хотя и не смотрится так эффектно

я про режим "гирлянда", когда часы - просто приятный бонус

vvip-68, 2020-12-29:
Ну, если  у вас из бегущих строк только часы с температурой и с рождеством - забейте 
В ячейку №1 "{D}" или вариант с форматом
В ячейку №2 "{WT}"
В ячейку №3 "C Рождеством {S07.01.****}"
с нужными настройками фона. Поставьте нужный интервал вывода строк- 60 секунд или сколько нужно,
В ячейку №0 что-то типа  "#121212121212123". Количество этих 12 посчитайте, чтобы 3-ка выходила раз в час

limittouch, 2020-12-29:
Изврат какой )))

так-то у меня не три строки, и миксовать их подобным образом тот еще треш
+ я не уверен, что корректно указывать в последовательности ячейки №0 строку с макросом события на конкретную дату