vvip-68/GyverPanelWiFi #89: Нет возможности задать формат времени

limittouch, 2020-12-28:
> Пример:
> Строка D: _До Нового года осталось {P01.01.****#E#60#60}_
> Строка E: _-С Новым годом!!!!_

будет ползти строка
_До Нового года осталось 7 **секунд**_
убрать слово "секунд", чтобы уменьшить объем бесполезного текста, нельзя

пока строка приползет в следующий раз, обратный отсчет благополучно закончится

vvip-68, 2020-12-29:
А что означает "До нового года осталось 7", если нет слова "секунд"? Звучит коряво, не по-русски.
Впрочем, - слово "секунд" формируется в utils.ino - процедура   WriteSeconds() строка 553. Можете убрать тут.
Или в runningText.ino - там где вызовы этой процедуры - строка 1308 и 1363 - исправьте как вам нужно.

closed 2020-12-29 by vvip-68

limittouch, 2020-12-29:
не фраза _До нового года осталось 7_ , а просто  _7_, _6_ , _5_, _4_, _3_, _2_ ,_1_

в полной фразе, пока уедет _секунд_, пока проедет повторно _До нового года осталось_, отсчет благополучно закончится

и речь шла не про **выкосить в прошивке единицы времени**, а про **добавить возможность задавать формат**
например,

00:01:05
00:01:04
...
00:00:57
00:00:56
...
00:00:00

красиво ведь, прям Пила или Квест %)