vvip-68/GyverPanelWiFi #90: Нельзя остановить бегущую строку

limittouch, 2020-12-28:
> Пример:
> Строка D: _До Нового года осталось {P01.01.****#E#60#60}_
> Строка E: _-С Новым годом!!!!_

когда обратный отсчет идет на секунды, строка может благополучно уползти
для отсчета нет возможности задать макрос "стоячей" отцентрованной строки

vvip-68, 2020-12-29:
Нет, такой возможности нет. Надо подумать - может когда-нибудь...
Но - ведь у всех размеры матриц разные. У многих 16x16, например.
Что будет показывать отцентрированная стоячая строка? Две буквы из центра слова?

closed 2020-12-29 by vvip-68

limittouch, 2020-12-29:
> Что будет показывать отцентрированная стоячая строка? Две буквы из центра слова?

этот режим подразумевает короткие строки, т.е. будет показывать только **цифры секунд**