vvip-68/GyverPanelWiFi #91: SD карта

Junior220584, 2020-12-29:
Уважаемый, vvip-68, буду благодарен за ответ)
Взял плату ESP8266 Amica, подключил все пины SD в разных вариациях, не работает. Пишет карта недоступна. В чем еще может быть причина? Почитал в инете, типа обязательно необходимо включить функцию FatFs, не знаю что это. 
Можете подсказать что еще предпринять? Может заплет с самой картой? Она отформатирована в FAT, на Ардуино МЕГА работает и инициализируется....

vvip-68, 2020-12-29:
Я не знаю что за плата ESP8266 Amica.
Если что-то вроде такого
[image]
SD карта должна быть подключена к тем зеленым пинам CLK MISO MOSI
В Wemos d1 mini это пины D5,D6,D7
[image]

Это аппаратные пины SPI, именно они используются библиотекой для работы с картой.
Если у вас не так - работать не должно. Сделайте так.

vvip-68, 2020-12-29:
Залейте тестовую прошивку из примеров - работа с SD-картой, проверьте работает она или нет

Junior220584, 2020-12-29:
Плата как на первой картинке, просто производитель другой. С тестовыми прошивками тоже не работало, Сейчас вроде разобрался, буду дома проверять. В итоге когда указываешь пин CS - D8 - карта не инициализируется. С просто "8" все читается. Надо только с лентой проверить, будет ли он out файл читать. Причем раньше я пробовал с пинами 2, 4 - тоже не работало, может мой косяк, у меня слабая матчасть по ардуинке))
Но все равно огромное спасибо, осталось разобраться с текстовыми строками и макросами)))) видимо вопросы буду писать позже))
VVIP-68 СПАСИБО!!!

vvip-68, 2020-12-29:
Соответствие всех этих номеров чистых и обозначений - 8->D8
сильно зависит от того какая плата выбрана в настройках.
Так 2 на Wemos -> D2, а на NodeMCU -> D4

Зависит от ропределения явно или неявно заданных
```
#define FASTLED_ESP8266_RAW_PIN_ORDER
#define FASTLED_ESP8266_NODEMCU_PIN_ORDER
#define FASTLED_ESP8266_D1_PIN_ORDER
```
https://github.com/FastLED/FastLED/wiki/ESP8266-notes

Junior220584, 2020-12-29:
Понял, спасибо за науку! Я этого не знал.

closed 2020-12-29 by vvip-68

Junior220584, 2020-12-30:
vvip-68, можешь еще подсказать подсказать. Судя по всему все-таки ESP не открывает SD карту. Загрузил пример Listfile, он в мониторе порта не показывает какие файлы загружены, а просто выводит данные по кругу ниже. Плюс с вашим скетчем и наличием SD светодиоды не загораются от слова совсем(( Есть мысли? на всякий случай, напоминаю, что у меня ESP 8266 как на первой картинке выше. Подключаю SD так:
MISO - D6
MOSI - D7
CSK - D5
CS - D8

Как всегда заранее спасибо за ответ)

Junior220584, 2020-12-30:
wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
Initializing SD card...
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
Initializing SD card...
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
Initializing SD card...

vvip-68, 2020-12-30:
Нет, особо ничего сказать не могу. По приведенному логу - у тебя вообще контроллер по кругу перезагружается по wdt - Watchdog timer - когда скетч перестает отвечать N-ое кол-во времени (зависит от настройки, макс - 8сек) - он аппаратно сбрасывается и переходит в перезагрузку. Причину - не знаю. Попробуй другой шилд, попробуй другой микроконтроллер. Попробуй другой пример. Попробуй WDT на максимум поставить - ESP.wdtEnable(WDTO_8S); Попробуй подключиться к пинам зеленым те, что слева с аналогичным назначением.

На моей прошивке видимо по той же причине ни один диод не загорается - что контроллер не выходит на работчий режим.

Junior220584, 2020-12-30:
Да, я так и понял, что он в перезагрузку уходит. Правда не понимаю почему( Шилд точно рабочий, с мегой работает. Контроллер тоже менял. К другим пинам подключался - ничего. Буду играться, но уже запарился))))))🤦‍♂️Спасибо!!!

vvip-68, 2020-12-30:
А что за пример ты взял? Я у себя вообще примера listfile найти не могу
[image]

Junior220584, 2020-12-30:
[image]

отсюда

vvip-68, 2020-12-30:
А ты в скетче примера пины по умолчанию и инициализацию SD-карты исправлял на используемые?

Junior220584, 2020-12-30:
правлю только эту строку #define SD_CS_PIN 8
остальное оставляю без изменений

vvip-68, 2020-12-30:
У меня вообще такой строки в примере, открытом по твоему скриншоту нет...

Junior220584, 2020-12-30:
прости, я уже другие примеры из твоей папки пробую)))
из listfile эта строка
 if (!SD.begin(8)) {

vvip-68, 2020-12-30:
Вот твой пример из listfiles
if (!SD.begin(D8)) {

Вот плата
https://drive.google.com/file/d/1Q4xFQU-FcQiUhz37khUbZThNu4wVJLu0/view?usp=sharing

Вот результат
[image]

Что я делаю не так?

Junior220584, 2020-12-30:
[image]
[image]

Да ты то все правильно делаешь. Вопрос в том, что я делаю не так)))))

vvip-68, 2020-12-30:
Ну попробуй на SD-карте MOSI и MISO провода местами поменять

Junior220584, 2020-12-30:
сразу все попробовал после твоего первого сообщения, не помогает(((

vvip-68, 2020-12-30:
Инициализация именно так в примере прописана?
if (!SD.begin(**D8**)) {

Junior220584, 2020-12-30:
нет, ставил просто 8. С D8 она вообще не инициализируется

Junior220584, 2020-12-30:
[image]

vvip-68, 2020-12-30:
Не уверен что "просто 8" проецируется на пин D8, к которому провод подключен
А что значит "вообще не инициализируется?

Junior220584, 2020-12-30:
выше скрин отправил
Initializing SD card...initialization failed!

vvip-68, 2020-12-30:
Ну, по крайней мере в вечный ребут не уходит

Junior220584, 2020-12-30:
это да))) но от этого не легче)))))

vvip-68, 2020-12-30:
И с обоими вариантами MOSI MISO так?

Junior220584, 2020-12-30:
ага

vvip-68, 2020-12-30:
А у тебя в описании твоей карты нет - может ей не 3 вольта надо, а она 5-тивольтовая и при подключении к 3 вольтам ей просто питалова не зватает? Там какой-нибудь микросхемы-стабилизатора нет? На плате самой вольтаж не прописан? 
Ты же при подключении к натуральной ардуине ей не 3 вольта давал, а 5?

Junior220584, 2020-12-30:
Об этом тоже уже думал. От 5В такая же хня

vvip-68, 2020-12-30:
Осталось только взять адаптер SD-карты и подпаяться напрямую к контактам адаптера без всяких этих шилдов. Как подпаиваться к контактам адаптера - есть на ютуб ролики

vvip-68, 2020-12-30:
Вот, например - https://www.youtube.com/watch?v=pQo8lZ4CtLA&t=4s

Junior220584, 2020-12-30:
понял, буду пробовать напрямую. Спасибо!

Junior220584, 2020-12-30:
[image]

сейчас дал нормальные 5В на шилд и сделал общую землю с ESP. Один раз прочиталась, больше не хочет. Уже успех)))

vvip-68, 2020-12-30:
А раньше типа общей земли не было что ли? На фото же была? Как же она без общего минусового провода работать будет? Для напряжения нужно два проводника - сигнал и опора относительно которого мерить.

Вот этот мусор - вообще не понимаю что у тебя за вывод на экране? Откуда он? Такое впечатление, что где-то скорости не совпадают... Или контакта в проводе нет...

Junior220584, 2020-12-30:
С мусором не знаю что делать. Мне кажется так всегда было. 
Раньше я просто подключал 5В на шилд, а земля была общая. Сейчас подключил прям от БП шилд и от шилда общую землю на ЕСП.

vvip-68, 2020-12-30:
Я думал ты без питания, от USB все запитываешь. А еслли от отдельного БП - так конденсатор может нужно поставить? Если пульсации с БП идут - могут тоже проблемы создавать

Junior220584, 2020-12-30:
Кондер на шилд или на ЕСП?

vvip-68, 2020-12-30:
Если есть - поставь на оба. Если нет - на ESP...

vvip-68, 2020-12-30:
Просто если отк омпового USB - там то питание нормальное без пульсаций...

Junior220584, 2020-12-30:
есть. Ушел паять)) вернусь с обратной связью после праздников. Надеюсь смогу победить))
Спасибо!

Junior220584, 2020-12-30:
Так до этого и пробовал просто от USB - не работало. Может действительно помехи от БП....очень на это надеюсь)

Junior220584, 2020-12-30:
какие лучше?
есть 6,3V 470мкф
есть 6,3V  1000мкф

vvip-68, 2020-12-30:
Любые. А если есть tit маленькие керамические параллельно на 20-33 ya - вообще шикарно

Junior220584, 2020-12-30:
увы, такого добра, к сожалению нет. Пока попробую просто с кондерами

vvip-68, 2020-12-30:
А когда от просто USB - питание VCC тоже с пина +5V брал?

Junior220584, 2020-12-30:
да

simonovich, 2021-01-04:
> Осталось только взять адаптер SD-карты и подпаяться напрямую к контактам адаптера без всяких этих шилдов. Как подпаиваться к контактам адаптера - есть на ютуб ролики

 Я подпаялся, теперь вопрос: а как дальше расширять? Чтобы работал плеер, доп. дисплей и т.д. SD карта занимается под себя много пинов. Вроде в DFPlayer есть слот для карт. Он может использоваться также и для анимаций?

vvip-68, 2021-01-04:
> Вроде в DFPlayer есть слот для карт. Он может использоваться также и для анимаций?

Нет. DFPlayer - сам в себе вещь. Он даже список файлов с карты не отдает. Все что умеет - вернуть кол-во файлов в папке, остальные команды - только на вход - играй файл номер N в папке M.

Выход - ставить ESP32. В ней выходов больше.  В папке schemes есть схемы для ESP32 от энтузиастов, которые все это к ней подключали. Говорили, что у них все работает.

Junior220584, 2021-01-04:
Уважаемый vvip-68, я разобрался со своим шилдом. Выпаял с него стабилизатор на 3,3В и все заработало. Флешку читает, но почему-то не запускает с нее эффекты. Все эффекты записаны в один файл new.out. Находятся в папке 24x11. Монитор порта этот файл видит, а гирлянда не воспроизводит. Как проверить? Можно как-то одну флешку запускать, без всех эффектов?

vvip-68, 2021-01-04:
В папке Tools есть утилита JinxFramer. Вот она читает файл эффекта непосредственно с флэшки? Есть там кадры эффекта? Какой размер этого файла? 

Что в логах (в мониторе порта) при открытии этого файла, т.е при выборе в приложении на странице эффектов в комбобоксе этого файла?

Junior220584, 2021-01-04:
Спасибо, как всегда за быстрый ответ! 
Framer читает файл с флешки, размер 772КБ. Приложения нет, я без андроида(((
[image]

vvip-68, 2021-01-04:
Все кадры эффекта с флэшки видит Framer? Нет в конце черных? В логе не увидел чтения файла эффекта при попытке его воспроизвести. Как управлять без приложения - я не знаю.

Junior220584, 2021-01-04:
Да, Framer всё видит. Второго вопроса не понял... Раньше монитор писал, типа воспроизведение ОК. Я думал, что в коде можно все настраивать)) это не так?

vvip-68, 2021-01-04:
Вот так в логе выглядит" проигрывание" файла эффекта с SD-карты:

[image]

Основная концепция прошивки - в коде настраивается ширина/высота матрицы, угол подключения. Больше пользователю ничего знать не нужно. При первом включении создается точка доступа к которой подсоединяется приложение со смартфона и далее все настройки и управление выполняются из приложения.

Естественно в коде можно задавать значения параметров и все такое. Но это должен быть продвинутый пользователь, с навыками программирования и знающий набор переменных в программе, структуру программы,  какая настройка (переменная) за что отвечает, в какой момент что должно происходить, чтобы отработало то или иное действие на матрице.

Junior220584, 2021-01-04:
Да, я это понимаю, но в силу отсутствия андроида, пытаюсь разобраться в коде. Т.е. лучше обзавестись андроидом и не ипать себе мозг, верно?)

vvip-68, 2021-01-04:
Ну, если это развивает мозг, не даёт ему атрофироваться - почему бы и нет?
Ваш выбор по какому пути идти.

Junior220584, 2021-01-04:
да, спасибо, действительно развивает)))
Еще чуть-чуть помучаю... Записал на флешке бегущую строку "С НОВЫМ ГОДОМ!!!". Framer ее воспроизводит нормально, а гирлянда только "С НОВЫМ ГО". Не понимаю почему. Есть мысли? 
При этом на флешке 3 файла .out. А гирлянда воспроизводит только эту неполную бегущую строку и все.. Почему так?

Junior220584, 2021-01-04:
Видимо разобрался. У меня каждый эффект в прошивке длится 10 секунд. Судя по всему флешка также воспроизводится это время. Это как-то исправляется?

vvip-68, 2021-01-04:
Увеличить время эффекта до 20 секунд. Или сколько там нужно.

[11 more replies]