vvip-68/GyverPanelWiFi #92: Можно ли увеличить место для ввода WIFI пароля?

buba-911, 2020-12-29:
В начале хочу поблагодарить автора за проделанную работу.
На у меня встретилось пару проблем:
В программе ограниченно количество символов для пароля WIFI -16. Можно ли увеличить их до 20? Если да то что надо изменить в коде?

Ещё , в приложении на телефон , отсутствует вкладка игры и при выборе эффектов на выбор предлагается только тот который работает в настоящее время в демо режиме.

AndreyLyamkin, 2020-12-29:
Поверь, тут будет проще изменить пароль сети. Я уже задавал такой вопрос

vvip-68, 2020-12-29:
Ну, можно было перенести пароль сразу в самую последнюю часть фрагмента свободной области. Пока до нее доберется... Теоретически с ячейки 800 и до 4000 - область расположения строк текста. 99%, что эта область никогда не заполнится полностью. Перенеси пароль в конец этого фрагмента. 
Вот убей не понимаю зачем такие длинные пароли. 16 символов - за глаза. Даже эти 16 символов брутфорсить столетия...

closed 2020-12-29 by vvip-68