vvip-68/GyverPanelWiFi #93: Режим "Ночные часы"

sanyafifa, 2020-12-29:
Режим "Ночные часы" с последней версией
`#define FIRMWARE_VER F("WiFiPanel-v.1.08.2020.1229")`
 у меня некорректно  отображается (в самом низу матрицы и моргает). 
Может я начудил, у кого такая же проблема?

vvip-68, 2020-12-29:
Не, вроде всё нормально работает. Какой размер матрицы?

vvip-68, 2020-12-29:
Режимы часов попереключать - малые / большие / вертикальные / горизонтальные?
И что - в ночных проблема, а в дневных? Они только цветом/яркостью отличаются....

sanyafifa, 2020-12-29:
> Какой размер матрицы?

10*10

> Режимы часов попереключать - малые / большие / вертикальные / горизонтальные?

с большими все ОК. 
ночные/дневные - не работает.

vvip-68, 2020-12-29:
Че-т на 10x10 они вообще влезть не должны. Для малых горизонтальных нужно 15x5 минимум, для вертикальных - 11x7. Для больших часов - еще бОльшая матрица нужна. Они у тебя вообще не должны отображаться ни в каком виде. Удивлен.

sanyafifa, 2020-12-29:
https://user-images.githubusercontent.com/49896945/103309649-af962f00-4a26-11eb-9657-b1c25fa913d7.mp4

как-то так))

vvip-68, 2020-12-29:
Ну не должны же они на малой матрице работать - там проверка на размеры стоит и если размеры не подходят - запрещены часы. Я накосячил или ты правил? Выглядит, как будто оверлей не восстанавливается (или поздно восстанавливается) - рисует сначала поверх новые цифры.... Что-то даже не представляю как такой эффект достигается. У меня нормально показываются...

closed 2020-12-31 by vvip-68

sanyafifa, 2021-01-08:
Прошился этой версией `"WiFiPanel-v.1.08.2020.1226-beta8"` - все ок.
странно.