vvip-68/GyverPanelWiFi #98: MQTT забивает устройство до полного отказа

simonovich, 2021-01-01:
У меня гирлянда 21х12 и официальный скетч WiFiPanel-v.1.08.2020.1229
О железе: https://yadi.sk/i/qBAz6cjC2PNF9Q
В качестве брокера mqtt - srv2.clusterfly.ru

Работает замечательно. Но если активировать MQTT, то через минут 15 начинают появляться фризы, подвисания эффектов, потом хуже: устройство начинает терять wifi, отключается. При повторном включении гирлянда либо не зажигается, либо вспыхивает на пару сек и тут же окончательно гаснет.
По кнопке эффекты не меняет, не включает - максимум - "вспышка" в рандомном месте.

На новую команду по mqtt в таком "убитом" состоянии либо не отреагирует, либо "прогонит" максимально быстро один за другим целую цепочку эффектов и в конце выполнит нужную команду. Очень похоже, что таким образом он получает и обрабатывает некую скопившуюся на входе "очередь" команд.

В приложении при этом могут неправильно или вообще не подтягиваться опции. Например на странице эффектов наблюдал полную потерю списка эффектов, кроме пункта "Лампа".

А по mqtt-каналу можно наблюдать бомбежку сообщениями pwr о вкл/выкл питания:
https://yadi.sk/i/D56caPkXBe-BAg

Теперь о том, как вернуть зомби к жизни.
Нашел 2 способа:
1. Временный. Полностью отключить устройство. Девайсы с которых посылались команды mqtt - тоже выключить.
Минут через 5 - включить гирлянду. Если пробьетесь на wifi интерфейс, то считай успех.
Можно увеличить MQTT_SEND_DELAY - будет положительный эффект. Но не надолго - лучше способ 2.
2. Пробиваемся на интерфейс управления, либо командой mqtt долбим, либо прошиваем - короче любым способом отключаем подключение mqtt. После этого никаких глюков не будет.
Но и mqtt тоже не будет ))

Такая вот история. Если кто-то представляет в чем тут источник бед - хорошо бы пофиксить!

vvip-68, 2021-01-01:
По моим наблюдениям это из за поведения MQTT-сервера. В свое время я с srv2 ушел на srv1 - там тоже такое случается, но значительно реже. Надежнее всего ставить свой MQTT-брокер в локальной сети, как говорят люди, использующие MQTT-канал в своей сети со своим брокером. 

На этих srv1 и srv2 читал где-то в описании, что отправка сообщения не чаще 1 сообщения в секунду и месяц-два назад, все посылаемое чаще просто игнорировалось и уходило "в пустоту". После я уже не нашел такого ограничения и даже в настройках совсем убрал задержки. 

srv2 рвет соединения после отправки каждого запроса и контроллеру приходится каждый раз восстанавливать соединение. srv1 принимает чаще всего весь пакет сообщений без задержки но время от времени также начинает разрывать соединения.

Могу посоветовать найти какие-нибудь другие бесплатные MQTT сервера или настроить в сети свой собственный.

Однако, не исключаю варианта, что прошивка (вероятнее всего именно библиотека mqtt) при каких-то условиях начинает "истощать" память (или фрагментированный heap) и тогда на очередной запрос не может выделить память и соединение рвется. Попробуйте или поотключать всё лишнее, убрать часть эффектов или перейти на ESP32 у которого вроде как оперативной памяти немного больше.

closed 2021-01-01 by vvip-68

simonovich, 2021-01-01:
А что можно сделать со списком эффектов? В нем по прежнему один пункт.
Переустановка приложения на новую версию с чисткой кеша не помогла.
https://yadi.sk/i/mCEVjcxnZ9_zuw

vvip-68, 2021-01-02:
Улучшить связь в сети. В приложении один пункт в эффектах, когда она не может получить список эффектов, который ей отправляет устройство. Сетевой пакет со списком просто не доходит до приложения, где-то теряется.  Все остальное в приложении тоже должно работать "через раз" - параметры на страничках не актуализируются, поля просто могут быть пустыми.

vvip-68, 2021-01-02:
Еще вариант - взять другой микроконтроллер. Мне попадались экземпляры, которые очень плохо держали связь. Вот такое неполучение информации с контроллера, частое "отваливание" соединения с приложением.

vvip-68, 2021-01-02:
Вот конкретно такая модель микроконтроллера у меня совершенно не держит соединение со всеми симптомами, описанными выше
[image]

simonovich, 2021-01-02:
У меня классика - ESP 8266 , ничего такого...
https://yadi.sk/i/qlFPGnxe1TWxlQ