vvip-68/GyverPanelWiFi #100: Статический IP адрес

Vovaka1, 2021-01-02:
Приветствую. В крайней версии перестал сохраняться статический IP адрес. При нажатии на "Сохранить" на матрице показывается новый IP, но после перезагрузки приходиться исправлять в приложении на IP выданный роутером. Пробовал заливать (в ESP32) версию 1.07, там сохраняется.

vvip-68, 2021-01-02:
Наблюдаю, да. Самое смешное - что код абсолютно одинаковый для ESP32 и для ESP8266
Причин понять не могу - в этом куске кода, связанном с адресами давно ничего не менялось, вроде.
По наблюдениями - он сохраняется и далее используется ДО обновления / загрузки прошивки в контроллер.
То есть - залил, настроил - можно пользоваться. Но после обновления - слетает. В чем траблы - понять не могу. Когда / если пойму - починю. Пока - никак. Если сможете понять в чем дело - напишите.

vvip-68, 2021-01-02:
Во, вроде понял в чем дело. Обновил файл eeprom.ino
Должно поправиться...

closed 2021-01-02 by vvip-68

vvip-68, 2021-01-02:
И еще там файлы затронуты были - снова обновил.

Vovaka1, 2021-01-02:
Дважды качал архив. Не проходит компиляция.
Ошибка  'runTextSound' was not declared in this scope 
Подсвечена строка 335  runTextSound = -1;   // Нет звука, сопровождающего строку

vvip-68, 2021-01-03:
Исправил