vvip-68/GyverPanelWiFi #101: Бегущая строка на фоне

Des-75, 2021-01-03:
Здравствуйте.
Версия скетча "WiFiPanel-v.1.08.2020.1226-beta8", плата WemOS.
Настройки бегущей строки:
0 #12345
1 {B#000000} {C#00FF00} {D:dddd} {C#0000FF} {D:dd} {D:MMM} {C#FF0000} {D:HH:mm} {#3}
2 {B#000000} До Нового Года {R01.01.2021 0:00#Z}{#3}
3 {B#000000} На улице {WT} °С
4 {B#000000} С Днём Энергетика! {S22.12.**** }
5 {B#000000} С Рождеством! {S07.01.**** }
..
Z {B#000000} С Новым Годом!{S01.01.**** 0:00#14.01.**** 23:59}{-}{E2} {#3}

Итак, сейчас вместо макроса в строке 2 отображается макрос из строки Z. Тут всё правильно, но есть особенности:
Закономерности не нашёл, но иногда (примерно каждый 4-й раз, соблюдается не всегда), эффект Е2 из строки Z отображается с максимальной скоростью (скорость текста нормальная). А бывает, что не только скорость максимальная, но и яркость очень маленькая (причём в этом случае и текст тоже максимально блеклый).
При этом вызов строки 3 проходит в нормальном режиме.

vvip-68, 2021-01-03:
Ну, да. Таймер (скорость) эффекта - один и берется от текущего (настоящего) эффекта. Просто в процессе отрисовки - если в строке указан другой эффект - отрисовывается он вместо штатного, с скоростью текущего штатного. Это довольно специфический случай и городить огород с запоминанием перенастройкой таймеров, я видимо не буду. Будем считать это не багом, а фичей. Тем более - вы единственный у кого так не совпали скорости. У 99% повторяющих скорости эффектов практически одинаковые и никто этого не замечает, никого это не напрягает. Если вы готовы поработать над исправлением этого - я скажу в каком месте программы происходит отрисовка одного эффекта вместо другого - а там вы думайте, исправляйте, если сильно напрягает...

Про яркость - даже не знаю причину и в какую сторону копать. Яркость текста и часов - одна, используется системная. И настраивается только либо с ползунка яркости в приложении, либо при включении режима ночных часов (но в режиме ночных часов строка отключена). Поэтому описанный вами эффект - на одной строке яркость текста минимальная, на другой восстанавливается - вот что-то даже не могу предположить как такое может быть... В программе нет механизма для этого...

closed 2021-01-03 by vvip-68

Des-75, 2021-01-03:
Понял. Спасибо. Соглашусь считать, что это фичей.