vvip-68/GyverPanelWiFi #102: Варианты текстовых сообщений на новый год

barakumis, 2021-01-05:
Предлагаю в этой теме делиться текстовыми заготовками к празднику Новый год.
1. Новый год через {R01.01.2022#2}{B#000000} - Будет выводить фразу Новый год через (а дальше писать остаток дней до события (01.01.2022), за несколько дней добавится отображение помимо дней ещё и часов). Текст "бежит" на черном фоне ({B#000000}-это отвечает за черный фон). После наступления события (наступило 01,01,2022) будет выводиться текст из строки №2  (в строке это #2).
2. -C Новым Годом!!!{B#000000} - выводится текст только после наступления события из строки 1
3. С наступающим Новым Годом!!!{B#000000}
4. Тепла и уюта Вашему дому!{B#000000}
5. Здоровья и отличного настроения!{B#000000}
6. Пусть мечты сбываются!!!{B#000000}
7. С Рождеством!!!{B#000000}
8. 2022 привет

У меня получились такие варианты, Накидайте кто какой использует текст.

vvip-68, 2021-01-05:
Отфонарные примеры текстов есть в файле eeprom.ino в строках 412-447

closed 2021-01-13 by vvip-68

EDUARDT68, 2021-10-10:
У меня почему то получилось чёрный фон записать так: {E#0}
Как в примере автора получался белый цвет.

Pravda86, 2021-11-02:
Добрый день почему как настроить текст До нового года осталось дней , при этом только весь декабрь, не получается что то.

vvip-68, 2021-11-02:
Что конкретно не получается? Что делали, чтобы получилось?
Статью в wiki читали?
https://github.com/vvip-68/GyverPanelWiFi/wiki/Настройка-бегущей-строки

Pravda86, 2021-11-02:
Пишу так : До нового года осталось {R01.01.2022}{S01.12.****}  Наряду с другими текстами , этот тоже выводится, хотя еще декабрь не наступил

vvip-68, 2021-11-02:
Насколько я помню - для макроса S через решетку должна указываться дата окончания периода. С указанием второй даты - что-нибудь меняется?

Pravda86, 2021-11-02:
Да вроде получилось, накрайняк отключу временно. Спасибо.

vvip-68, 2021-11-02:
К слову говоря - у меня и с одной датой в указанном ваме формате все работает - не показывается.
Что пишет в мониторе порта, когда вы в приложении в редакторе текста бегущей строки нажимаете енопку Сохранить для этой строки?

Pravda86, 2021-11-02:
Да все работает, для примера включил 3 строки с отображением в декабре, ничего не работает. Похоже с макросами намудрил

pilnikov, 2022-12-01:
новости можно добавить? 
https://www.youtube.com/watch?v=F3vVPQLzyJk

vvip-68, 2022-12-01:
Да, конечно. Код скетча открытый - добавляйте что хотите.

pilnikov, 2022-12-01:
я не для себя я для людей. у меня уже есть (видео выше)

vvip-68, 2022-12-01:
Ну, люди, если захотят, тоже себе сделают - код открытый же. Мне лично - этого не надо. От слова "совсем".

pilnikov, 2022-12-01:
ясно 
для людей
https://github.com/pilnikov/WeatherStation/blob/master/wc_rt1/lib/MyLib_es/Exts.cpp
с 437 строки до конца

megadenis, 2022-12-08:
Никак не удается настроить автоматический запуск строки, в режиме Просмотра, строка бежит. Интервал установил 90 сек. Строки следующие:
0 строка: Новый год через {R01.01.2023#2}{B#000000}
1 строка: -C Новым {D:2023} годом!!!{B#000000}

Пробовал и простой текст вставить в 0 строку, и нажимал сохранить, но строка не запускается. 
Помогите пожалуйста.
Настройки:
[image]

vvip-68, 2022-12-08:
Ну, начнем с того, что у вас отображение бегущей строки вообще отключено. Совсем. Поставьте галочку вверху, разрешающую использование бегущей строки и будет вам счастье. Если, конечно, в настройках эффектов использование бегущей строки также не отключено.

vvip-68, 2022-12-08:
Ну и порядок как показывать строки обычно забит в нулевую ячейку. Вы туда забили просто строку вместо управляющей последовательности. Оедкий способ настройки. Я, честно говоря, даже не помню как в этом случае показываются строки автоматически. Возможно что и никак, а только кнопкой.

megadenis, 2022-12-08:
Спасибо за быстрый ответ! Заработало. Просто я думал, если в скрипте указано выводить на черном фоне - то должно работать. В моем случае, текст виден хорошо исключительно на черном фоне. 
Ещё пожелание, было бы круто если бы было можно инвертировать джойстик в режиме игр, отзеркалить. Панель я сделал на стеклянном ограждении балкона на улицу, но у меня такие ленты, которые видны и из квартиры тоже, получается инвертирование управление играми. Спасибо за классный проект.

vitalichubur, 2022-12-19:
Здравствуйте @vvip-68 , спасибо большое за великолепное развитие проэкта. Матрица на окно, а теперь уже и панель функционирует у меня несколько лет. С настройками эффектов, времени и бегущей строки разобрался. Не могу победить погоду. Код региона поменял и на Яндекс и на втором "урл" но данные погоды не приходят. Я заметил в скетче во вкладке weather закоментированные строки 6 и 7. Так и должно быть? Заранее спасибо.

vvip-68, 2022-12-19:
В этих строках - пример как выглядит URL, который отправляется на сервер погоды, чтобы получить данные о погоде.
Строку 6 можно просто ввести в строку адреса браузера т посмотреть приходит ли ответ, содержащий погоду. Когда подставите код своего региона - должны в браузере получить ответ в формате JSON конкретно для вашего региона.

Не знаю откуда вы, но в украине вроде бы провайдеры блокируют яндекс и в ответ на запрос ничего значимого не присылается. Тогда остается только получение погоды с openweathermap.  Подставьте запрос в строку браузера из седьмой строки скетча. Если что-то возвращает, содержащее погоду - ищите код своего региона, чтобы вовзращалась погода именно для вашей местности. 

Далее, после того как добьетесь получения внятных результатов в браузере - подставляет соответствующие настройки (код региона) в настройки матрицы в приложении.

Описание как найти коды погоды для своего города есть в wiki

EDUARDT68, 2022-12-19:
Да, в Украине блокируют Yandex и mail.ru. Но есть VPN ;-) чем чя и пользуюсь. Код региона подбирал как описано в вики. 
vitalichubur, если Вы с Украины то могу подсказать не  дорогой VPN.

EDUARDT68, 2022-12-19:
vvip-68, да, в роутере.

vvip-68, 2022-12-19:
Не все роутеры это поддерживают... Огласите модель роутера, чтоб украинский народ знал что брать, если понадобится...

EDUARDT68, 2022-12-19:
Tp-Link Archer C6 можно вписать один VPN
Redmi AX6 Wi-Fi 6 Mesh Router на данный момент прописаано 10 адресов. И это как я понял не предел ( этот я брал для умного дома если что. Работает отлично ).

vvip-68, 2022-12-19:
Не должно время с интернета получаться без доступа к интернету (когда матрица работает в режиме точки доступа). И погода без доступа к интернету не должна получаться. Может с часами - вы в приложении кнопку нажимаете "Получить время с телефона"? 

Когда матрица подключается к роутеру - не то чтобы "стабильное соединение с точкой доступа" должно пропадать - точка доступа вообще должна пропадать. Если только в настройках специально не включено "поднятие точки доступа" ОДНОВРЕМЕННО с подключением к локальной сети.

И вот тогда могут начинаться разные "фокусы". Дело в том, что у микроконтроллера - один передатчик. И он настраивается как для точки доступа так и для подключения к сети на один и тот же частотный канал. Аппаратные особенности. Если ваш роутер меняет частотные каналы автоматически в поисках лучшего сигнала (в настройке не фиксированный канал) - тут варианты. Или будет пропадать связь с роутером при смене роутером частотного канала, или матрица останется на связи с роутером, но пропадет принудительно поднятая точка доступа.

Так то точка доступа нужна исключительно для возможности первоначального подключения к матрице после прошивки с целью настроить в приложении PanelSetup подключение к локальной сети и более ни для чего.

По поводу сбоев в работе IPTV. Тут два момента - а настроен ли роутер для передачи Multicast  UDP трафика (всякие там IGMP Snooping и другое, ответственное за multicast) и - не включен ли у вас режим групповой работы с использованием протокола E131 да еще и матрицей в роле ведущего устройства? Может она пакетами сеть забивает и роутер перегружает?

vitalichubur, 2022-12-19:
Спасибо за разъяснения. Проверю сказанное вами выше. Я упрямый, доковаряю(ахах)

serg-74, 2022-12-19:
> **Tp-Link Archer C6** _можно вписать один_ VPN **Redmi AX6 Wi-Fi 6 Mesh Router** _на данный момент прописано 10 адресов_. И это как я понял не предел ( этот я брал для умного дома если что. Работает отлично ).

Имею TP-Link Archer C6 v2 - не нашел где можно вписать VPN, подскажите, если не сложно....

vitalichubur, 2022-12-20:
> Спасибо за разъяснения. Проверю вышесказанное. Я упрямый, доковыряю(ахах)

Похоже разобрался. Кода я прописывал данные роутера(логин пароль) в программе панель сетап, я указал и айпи роутера чего не следовало наверное делать. Как только убрал айпи из программы всё заработало как надо. Пропали глюки с айпитв и коннект сотки с еспшкой. Спасибо большое за помощь! Vvip-68 огромный респект!

vvip-68, 2022-12-20:
Ну, вообще, как раз задумывалось, что в программе нужный IP и устанавливается.
В вашем случае, не указав там IP, вы получили выделение адреса устройству по DHCP - теперь не знаю как его найти, чтобы подключиться из приложения на телефоне. Разве что смотреть в логах роутера какой адрес выдан устройству и вбивать в приложение такой же.

А то что были глюки, когда прописывали адрес в приложении настройки - скорее всего устройству вы выделили адрес, который уже кем-то занят в сети (например тем же телевизором) - получили конфликт алресов и пропадание пакетов - ни телевизор нормально не работал, ни гирлянда.

vitalichubur, 2022-12-20:
Возможно так и было

vitalichubur, 2022-12-20:
Последний в приложение вбился автоматом

EDUARDT68, 2022-12-20:
> Имею TP-Link Archer C6 v2 - не нашел где можно вписать VPN, подскажите, если не сложно....

Смотрите тут по ссылке. 
https://youtu.be/md-_DTk52DM

vvip-68, 2022-12-20:
2.168.31.1 -  это скорее всего IP самого роутера. Если устройству в сети выдать такой же адрес - само собой, что ничего работать не будет.

vitalichubur, 2022-12-20:
Так и есть. Прошу меня простить за мой "английский" я почти чайник в этих делах.

persivin, 2023-02-08:
Подскажите, где можно поправить прошивку, чтобы сразу записать в Ардуино свои варианты текста?

vvip-68, 2023-02-08:
В файле a_def.hard.h есть определение
```
#define INITIALIZE_TEXTS 0  // Поставьте здесь 1 если нужна начальная инициализация текстов в памяти
```
Если поставлена 1 - при прошивке скетча в микроконтроллер со стиранием памяти (Erase Flash: All Flash Contents) в нее будут загружены примеры строк "по умолчанию", которые находятся в файле eeprom.ino примерно со строки 497

persivin, 2023-02-08:
Спасибо