vvip-68/GyverPanelWiFi #104: панели на 32x32 и 48x48

aniksuper, 2021-01-06:
Привет! Интересует возможность настройки контроллера под квадратные панели шириной больше 16 пикселей. У меня при подключении и настройки всего по факту выходит что оно будто столбиком 16x64 либо 16x144 пикселей. Нужно ли переписывать эффекты или это проблема с настройкой скетча?

vvip-68, 2021-01-06:
Эффекты должны работать. Единственное ограничение на размер 128. Потому что тип переменных индекса столбца/строки используется int8_t с диапазоном допустимых значений - -127..128. Для больших размеров править скетч нужно. И отлаживать. Для панелей 32x32 (да и для всех, меньших 128 по ширине/высоте) проблем быть не должно. Если памяти в микроконтроллере хватает...

closed 2021-01-06 by vvip-68

247760, 2022-12-24:
@vvip-68 прошу прощения за вопрос, немного не догоняю по максимальному количеству светодиодов. Понял что в ширину максимально 128, а в высоту так же до 128? Т.е. можно условно делать 128*128? Хотим в окно сделать максимально возможную плотность с помощью готовых модулей или же склеить и спаять из лент, плотность на метр там же разная от 30 до 144. Сориентируйте пожалуйста по этим условиям.

vvip-68, 2022-12-24:
Можно 128x128, вопрос в том - хватит ли под такие размеры свободной оперативной памяти микроконтроллера. Ведь это только основной массив цветов - 128*128*3 = 49152 байта. А нужен еще массив (немного меньшего размера) под область оверлея (текст и часы поверх эффектов). Некоторые эффекты требуют выделения памяти под свои нужды...  Само ядро расходует сколько-то оперативной памяти... 

К тому же 16384 диодов - а какая скорость обновления такого массива будет - примерно кадр в секунду или еще медленнее? Так что вопрос - "а нужно ли оно вообще?" (кроме "а хватит ли оперативной памяти?") - остается актуальным.

247760, 2022-12-24:
@vvip-68 спасибо за ответ, надо будет подумать.

vvip-68, 2022-12-24:
Добавлю еще, что под такие размеры не тестировалось - может быть косяк, что корректно работает с шириной / высотой 127. Незачем было детально изучать этот вопрос, но если будет на чем и зачем - думаю это поправимо...

247760, 2022-12-24:
@vvip-68 принял, спасибо. В любом случае, по ширине придётся втиснуть максималку в 127-128 из лент если кратно получится, а вот по высоте уже по пропорции, всё же имею прямоугольник окно.