vvip-68/GyverPanelWiFi #106: Не работает бегущая строка

dronov1981, 2021-01-14:
Привет! плата wemos d1 mini прошивка и приложение последние (1,08) 

Не могу запустить бегущую строку, при переходе на настройки текста и нажатию кнопки просмотр ничего не происходит, все остальное работает корректно и стабильно

vvip-68, 2021-01-15:
У кого-нибудь еще такое происходит?
Судя по описанным симптомам - как раз похоже, что приложение не 1.08, а старая версия, которая не совместима с прошивкой 1.08. Попробуйте обновить приложение на телефоне.

closed 2021-01-15 by vvip-68

dronov1981, 2021-01-15:
поставил сейчас 1,09 из репозитория результат тот же может матрица 10х10 маловата?

vvip-68, 2021-01-15:
1. Бегущая строка работает на матрицах с высотой от 8 строк.
2. Не существует приложения версии 1.09. Последняя версия приложения - 1.08, она совместима с прошивкой 1.09

dronov1981, 2021-01-15:
не смогу прислать, приложение 1 стоит на симуляторе, второе на TV приставке, телефона с адройдом нет (

vvip-68, 2021-01-15:
Что, настолько приложение не работает, что даже фотографию (с четко-различимым) содержимым с ЕМ-приставки или скриншот окна симулятора Андроид сделать не получается? Ну нифига ж себе....
Про часы - вообще ничего не понял. Консультации по телефону не оказываю.

vvip-68, 2021-01-15:
Матрица 10x10 для часов, кстати, маловата. Минимальный размер для горизонтальных часов 15x5, для вертикальных - 7x11. Часы на такой матрице работать не должны. Только в виде бегущей строки...