vvip-68/GyverPanelWiFi #107: Бегущая строка

Des-75, 2021-01-15:
Здравствуйте.
Версия скетча "WiFiPanel-v.1.08.2020.1226-beta8", плата WemOS.
Настройки бегущей строки:
0 #12345
1 {B#000000} {C#00FF00} {D:dddd} {C#0000FF} {D:dd} {D:MMM} {C#FF0000} {D:HH:mm} {#3}
2 {B#000000} До Нового Года {R01.01.2021 0:00#Z}{#3}
3 {B#000000} На улице {WT} °С
4 {B#000000} С Днём Энергетика! {S22.12.**** }
5 {B#000000} С Рождеством! {S07.01.**** }
..
Z {B#000000} С Новым Годом!{S01.01.**** 0:00#14.01.**** 23:59}{-}{E2} {#3}

Итак, сейчас вместо макроса в строке 2 отображается макрос из строки Z. Сегодня уже 15.01, но всё равно макрос из строки Z отображается, хотя не должен.

vvip-68, 2021-01-16:
Исправил. Обновите прошивку (и проиложение)

closed 2021-01-16 by vvip-68

Des-75, 2021-01-16:
Версия скетча "WiFiPanel-v.1.09.2021.0116", плата WemOS.
Настройки бегущей строки: та же

Теперь работает в таком порядке: строки 1, 3, 3, 1, 3, 3, 1, 3, 3... 
Т.е. показывает дату, затем сразу же за датой температуру (ок), потом просто температуру (ок), потом почему-то опять температуру (not ok).

vvip-68, 2021-01-16:
Чтобы не показывало температуру (а это обычная строка, участвующая в выборке) 0- поставьте в строке р запрет строки {-} - тогда она будет показываться только как "хвост" в других строках

Des-75, 2021-01-16:
Если бы я хотел чтобы не показывало температуру, я бы так и сделал - поставил запрет строки. А я хочу, чтобы это была как самостоятельная строка, так и подстрока для вызова из других строк.

vvip-68, 2021-01-16:
Тогда вы получаете совершенно справедливый порядок показа 1,3,3,1,3,3 и так далее

Des-75, 2021-01-16:
Хм... объясните, пожалуйста. 
Мне кажется, мы друг друга не понимаем. Я попробую ещё раз объяснить...
Порядок показа 1, 3, 3, 1, 3, 3...
1 - выводится строка дата/время **и температура**;
3 - выводится **только** температура;
3 - опять выводится **только** температура.
затем всё повторяется с начала...

А должно (как я представляю)  - 1, 3, 1, 3....
1 - выводится дата/время **и температура**;
3 - выводится **только** температура.
затем всё повторяется с начала..

Надеюсь, сейчас будет понятно.

vvip-68, 2021-01-16:
С чего бы это?
1. Выводится строка 1 и после ее окончания сразу строка 3 - как "прицеп" (1,3)
2. Строка 2 не показывается за истечением срока давности.
3. Выводится строка 3? поскольку она не выключена
4. Строки 4,5 не выводятся по не попаданию дат в макросе {S}
5. Строка Z не выводится - она отключена {-}
6. Goto п.1

Итого - вывод 1,3,3,1,3,3... и так далее

vvip-68, 2021-01-16:
Короче, возможно что-то и не работает и показывается не так, как ожидается вами.
99.9% пользователей этим не пользуется, я в т.ч.
Мне совершенно недосуг тратить на это время, чтобы чинить то, что не сломалось или используется в 0.01% случаев. У меня есть другие проекты и нет не них времени.
Если вам хочется добиться идеальной и правильной работы - действуйте. Потом расскажете мне что и где нужно исправить, ч поправлю это в коде проекта.

vvip-68, 2021-01-16:
Дам направление, где искать. Вполне возможно, что когда приходит время отображать строку 2 - она заменяется на строку Z как просроченная дата. Строка Я тоже не показывается на истечение периода показа, а вот прицеп 3 или от строки 2 или от строки Z показывается.

Des-75, 2021-01-16:
Ок. В итоге я вас понял. Объясняю в вашей "системе координат". Строки выводятся в следующем порядке: 1,3,3,3,1,3,3,3..., а должны 1,3,3,1,3,3...
К сожалению, моя квалификация как программиста "очень не очень", поэтому, буду считать это фичей. Тем более реализовать отображение строк под мои хотелки возможно и текущими возможностями скетча.

vvip-68, 2021-01-16:
Я внес еще кой-какую правку, связанную с этим. Попробуйте - возможно это решит вашу проблему. Но это не точно.

Des-75, 2021-01-16:
Нет, не оно...
Ладно, хватит заниматься ловлей блох, не так это уж и важно.
Но, всё равно, спасибо за попытку. :)

Mosrom, 2021-01-17:
Во первых большое спасибо за проделанную работу. 
У меня вопрос по бегущей строке в ночных часах. Почему-то отказывается показывать температуру, только значок градуса цельсия. В чём может быть причина? В обычном режиме всё нормально отображается. Версия прошивки 1.09.

vvip-68, 2021-01-17:
Очень странно, что в ночных часах вообще бегущая строка показывается. Не должна.
По крайней мере в моем скетче бегущие строки отключены и не показываются, если включен режим "Ночные часы".

Дело тут скорее всего вот в чем: Ночные часы показываются на минимальной яркости. Особенности светодиодов - в минимальной яркости светиться только красная компонента светодиода. Для зеленой и голубой нужна бОльшая яркость и, если текст не красный - просто ничего не отображается. А для температуры скорее всего выставлен режим "Цветная температура в ночных часах" - сейчас мороз - цифры синие. Яркость минимально - синяя компонента не показывается.

Или увеличьте яркость ночных часов, или выключите галку "Цветная температура в ночных часах".

Mosrom, 2021-01-17:
Бегущая строка - это часы и дата. Может не точно написал. Галку убирал и ничего не менялось. Может быть в режиме ночных часов нет связи с сервером погоды?

vvip-68, 2021-01-17:
Связь с сервером погоды есть всегда, даже когда панель "выключена". Микроконтроллер все равно работает, забирает погоду, синхронизирует время и т.д. Яркость увеличивали?

Mosrom, 2021-01-17:
Галку снял, яркость увеличил. Всё же должно показываться в одном цвете. Но температура нет.
Вот так это выглядит.
[image]

vvip-68, 2021-01-18:
Тогда не знаю. Я не могу починить то, что не могу воспроизвести.
У меня такого не происходит. Если когда-нибудь удастся получить такой эффект - смогу поразбираться, а пока вам, видимо, придется делать это самостоятельно. Если найдете причину - сообщите что нужно исправить. Если придумаете как я могу получить данный эффект - тоже сообщите, будем разбираться.

vvip-68, 2021-01-18:
Попробуйте в основном файле в строке 99 в setup() написать
putNightClockColor(0);

Mosrom, 2021-01-18:
Попробую. По результатам напишу. А в Ваших матрицах на данном режиме отображается значение температуры при снятии галочки со цвета? Значок градуса появляется вне зависимости от того есть сведения о погоде или нет?

vvip-68, 2021-01-18:
Все рисуется целиком в clock.ino в drawTemperature() - сначала знак +/- (если температура не равна нулю), потом одна или две цифры температуры, потом значок градуса, потом буква "C"

vvip-68, 2021-01-18:
Про получено или нет - может быть в этом причина, да.
Но вы же ж вроде писали, что в дневных сасах все отображается - значит температура получена.

Mosrom, 2021-01-18:
Оно не задвоенное, а немного не резкое, так как строка двигается и темно, а также матрица достачно большая (45×30 с шагом 3,3 см). Трудно сфокусироваться. С этим всё нормально. Видно чётко и красиво. В дневных часах да, работает как надо со цветом и характером погоды. Но может быть при переходе в ночные часы происходит разрыв связи и температура не отображается. К выходным испытаю Ваши советы и напишу. Странно, что у других с этим проблем нет. Так к слову, хочу попробовать испытать её на микроконтроллере ESP-32 по Вашей схеме.

simonovich, 2021-01-20:
Попытки вывести строку через апи. Не реагирует.

```
$6 14|Good Night!;
ack0;
$14 0;
$18 W:21|H:12|DM:0|PS:0|PD:20|IT:0|AL:0|RM:1|PW:15000|BR:212|WU:1|WT:15|WR:157|WS:1502026|WC:1|WN:0|WZ:1;
$ =6 14|Test;
$6 14|Test;
ack1;ack2;
$14 1;
$18 W:21|H:12|DM:0|PS:1|PD:20|IT:0|AL:0|RM:1|PW:15000|BR:212|WU:1|WT:15|WR:157|WS:1502026|WC:1|WN:0|WZ:1;
$6 14|text
ack3;
```

PS. Прошивка от 15 янв 2021

vvip-68, 2021-01-21:
Сочувствую. Если у вас какой-нибудь бесплатный MQTT-сервер - попробуйте найти другой, который отвечает. Если собственный и отвечает - смотрите доходит ли до него команда. Настройка соединения с MQTT дело тонкое. Я для проверки использую srv1.clusterfly.ru - но у него так себе надежность. Отвечает через раз, часто теряет соединение, постоянные переподключения...

simonovich, 2021-01-21:
А я не про MQTT, а про udp-пакеты по WiFi.
Хм... Так может дело в том, что команда $6 14 чисто под mqtt заточена?
Но должна же работать и на обычном сокете...

vvip-68, 2021-01-22:
Нет. Для прошивки нет разницы по какому каналу приходит команда - по UDP или по MQTT. Команды поступают в общую очередь и далее обрабатываются одной и той же процедурой. Я написал про MQTT потому что в текущем приложении нет интерфейсных элементов, которые отправляют эту команду приложению.

На каждую принятую команду прошивка, если ее приняла и обработала отправляет команду 'ack' с прикрепленным суффиксом просто инкрементным счетчиком от 0 до 255. Что за команда 'ack#' - не знаю. Не должно такого быть. Суффикс только числовой. Суффикс нужен из за особенностей работы компоненты UdpSocket приложения в Thunkable (а может быть и в AppInventor тоже - работает же одно и то же расширение). Особенность состоит в том, что после отправки команды через сокет компонента тут же считывает из сокета данные, даже если ничего не пришло, никакого ответа не было - возвращается предыдущий ответ, то есть непонятно обработана команда или нет. Таким образом меняющийся суффикс в ответе ack позволяет отличить реальный ответ от фэйкового. После нескольких неответов из прошивки  приложение понимает, что связь разорвана и переводит интерфейс в неактивное состояние.

Mosrom, 2021-01-22:
> Про получено или нет - может быть в этом причина, да.
> Но вы же ж вроде писали, что в дневных часах все отображается - значит температура получена.
> Я добавил еще одну проверку получена ли температура, выложил обновление.
> Но причина точно не в этом, потому что если бы температура не была получена - рисовался бы 0.
> 
> И еще у вас на снимке изображение какое-то "задвоенное".
> У меня вот тоже нет этого - ничего не задваивается...

Да, теперь температура в режиме ночных часов отображается.

vvip-68, 2021-01-22:
@Mosrom, и в чём была причина?
Та проверку, что я добавил - НЕ ОТОБРАЖАТЬ температуру, если погода не была получена.
Другого я ничего не исправлял. То есть исправление могло повлиять так, что вы бы вообще не видели температуру (и значок градуса С)б если температура не получена. Однако у вас температура стала отображаться. Так в чем была причина? Увеличили яркость?

Mosrom, 2021-01-22:
Думаю добавленная функция проверки повлияла. Яркость я не менял и дополнительную строку, что Вы советовали тоже не добавлял. Я только обновил скетч с учетом сделанных Вами изменений. 
И ещё один вопрос возник. В настройках бегущей строки через приложение при добавлении к макросу "звук мелодии" макроса любого из эффектов звук мелодии воспроизводится некорректно, а эффект не воспроизводится вовсе. Строка при этом отображается, но с текущим фоном. Однако если каждый из этих макросов использовать отдельно, то всё работает как надо. Так должно быть?

vvip-68, 2021-01-22:
Нет, так быть не должно. Но и связи между двумя этими макросами (звука и эффекта) я не вижу. Они каждый по себе, совершенно самостоятельные и друг с другом по коду никак не пересекаются. Будет возможность - проверю...

Mosrom, 2021-01-22:
Я тоже так думал, что каждый сам по себе. Сначала казалось, что с картой что-то не так. Но потом нашёл причину этой ошибки. Спасибо, буду ждать результата.

vvip-68, 2021-01-22:
Да, вы правы. Ошибка копи-пасты. Как раз для пары макросов {A} и {E}. Исправил.

Mosrom, 2021-01-22:
Спасибо за оперативность. Сейчас попробую. 
Как-то можно отобразить в бегущей строке время с прошедшего события по аналогии с тем, которое наступит (примеры у Вас к Новому году, Дню победы)? Например, "Тебе сегодня уже 20 - поздравляем!"

Mosrom, 2021-01-22:
> Да, вы правы. Ошибка копи-пасты. Как раз для пары макросов {A} и {E}. Исправил.

Спасибо. Это работает!

Mosrom, 2021-01-31:
> @Mosrom, и в чём была причина?
> Та проверка, что я добавил - НЕ ОТОБРАЖАТЬ температуру, если погода не была получена.
> Другого я ничего не исправлял. То есть исправление могло повлиять так, что вы бы вообще не видели температуру (и значок градуса С), если температура не была получена. Однако у вас температура стала отображаться. Так в чем была причина? Увеличили яркость?

Я тут разбирался с этим. Температура отображается только тогда, когда прогнозируется увеличение температуры. Когда прогноз на уменьшение температуры - цвет температуры голубой. Тут как Вы и предполагали, есть проблема отбражения цвета в ночных часах. Он не меняется вообще. Ну кроме красного конечно. Хотя эта возможность имеется и в скетче и в приложении. Т.е. при попытке изменить цвет ночных часов некоторые цифры пропадают, а оставшиеся отображаюся всё равно красным. Добавление значения в строку 99, а также изменение яркости ни к чему не приводят. Я думаю это ошибка в скетче.

vvip-68, 2021-02-01:
Если температуры отрицательные и включен режим отображения температуры цветом - оно отображается синим. от -3 до +3 - белым, положительные - красным. Никакого "прогнозируется увеличение / уменьшение" в скетче нет. Он не работает с прогнозами вообще. Получается только текущее на данный момент значение температуры.

Если режим отображения температуры цветом выключен - температура отображается цветом ночных часов. Цвет ночных часов - один из 7  - красный, синий, зеленый, желтый, голубой, пурпурный, белый. Устанавливается в приложении выбором. Яркость ночных часов регулируется ползунком яркости в режиме ночных цветов. Если яркость недостаточна (близка к нулю) - светодиоды святятся только красным цветом. Для синей и зеленой компоненты нужен бОльший ток, при недостаточной яркости его не хватает, светодиоды не светятся.

Странно, что данная ошибка проявляется исключительно у вас единственного. В скетче я никаких несоответствий не вижу. Если найдете конкретную причину ошибки и место в скетче - пишите. Ч не могу исправлять ошибки, которые не могу воспроизвести.

Mosrom, 2021-02-01:
1. Насчет прогноза это не суть. Основное это цвет. Значит в указанном Вами диапазоне положительное значение отображается, как я и говорил (скажем +1 показывает) отрицательное нет.
2. Про цвета понятно. Но я пробовал все цвета, не только зеленый и синий. И если делать ярче, то тогда это уже не ночные часы. Почему нельзя жестко прописать всю эту строку показывать каким-то одним цветом, чтобы она не реагировала на изменение цвета в показаниях температуры. Только для режима ночных часов. Пусть только красным.
3. Про исключительно, что у меня только, думаю, что многие этим даже не озадачивались. Использую не переделку, уверяю Вас.
Проблема точно есть и она явная.

vvip-68, 2021-02-01:
>Почему нельзя жестко прописать всю эту строку показывать каким-то одним цветом, чтобы она не реагировала на изменение цвета в показаниях температуры. Только для режима ночных часов. Пусть только красным.

Это как раз делается отключением галочки "Использовать цветную температуру" в ночных цветах. В этом случае все части ночных часов показываются выбранным цветом.
Про "увеличить яркость" и не ночные часы - это уже не ко мне, это к производителям светодиодов. У них выясняйте почему при одном и том же значении компоненты цвета / яркости красный светодиод светит, а синий и зеленый - нет.
Попробуйте использовать ленту на диодах APA102 - там вроде эта проблема решена и все три компонента светодиода светят одинаково при одинаковом значении их компоненты цвета / яркости.