vvip-68/GyverPanelWiFi #109: Ошибка при компиляции

vaicekas, 2021-01-17:
Здраствуйте, пытаюсь обновить прошивку, но получяю ошибку:

a_main:3903:82: error: 'IntToHex' was not declared in this scope
         imageLine += IntToHex(gammaCorrectionBack(getPixColorXY(sendImageCol, i))) + " ";
                                                                                  ^
C:\Users\Zydrius\Documents\Downloads\Compressed\GyverPanelWiFi-master20210117\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.09\runningText.ino: In function 'void rescanTextEvents()':
runningText:1789:76: error: 'LEAP_YEAR' was not declared in this scope
           if (daysInMonth == 28) daysInMonth += byte(LEAP_YEAR(iYear - 1970));
                                                                            ^
exit status 1
'IntToHex' was not declared in this scope

Arduino IDE последней версии, все библиотеки из архива. 
В чём моя ошибка?

Спасибо.

vvip-68, 2021-01-17:
В том, что смотрите на последнее сообщение в стеке ошибок, которая является как правило вторичной (порожденной) ошибкой. Первопричина - в списке ошибок выданных компилятором где-то выше. Скорее всего не поставили точку с запятой, скобку не закрыли или что-то еще. Устраните первопричину - эти ошибки уйдут автоматически

vaicekas, 2021-01-17:
В скече поменял только свои параметры матрицы, к разделительным знакам не притрагивался.
Попробую скомпилировать ничего не меняя, с Вашыми настройками, посмотрью результат.

vaicekas, 2021-01-17:
Нашёл свой косяк ;)
Я заменял utility.ino с предыдущей версии, првленный на литовский язык названия даты и времени.
А теперь сравнил с новым utility.ino, а здесь есть изменения.
Спасибо! ;)

closed 2021-01-17 by vvip-68

Dmitro4ekalov, 2021-01-22:
Здравствуйте! такая ошибка при компиляции:
C:\Users\Максим\Documents\Arduino\GyverPanelWiFi_v1.09\clock.ino: In function 'void checkAutoMode5Time()':
clock:1187:80: error: 'useweather' was not declared in this scope
C:\Users\Максим\Documents\Arduino\GyverPanelWiFi_v1.09\clock.ino: In function 'void checkAutoMode6Time()':
clock:1207:80: error: 'useweather' was not declared in this scope
exit status 1
'useweather' was not declared in this scope
в чем причина?