vvip-68/GyverPanelWiFi #111: Ошибка при компиляции

Dmitro4ekalov, 2021-01-22:
Здравствуйте! такая ошибка при компиляции:
C:\Users\Максим\Documents\Arduino\GyverPanelWiFi_v1.09\clock.ino: In function 'void checkAutoMode5Time()':
clock:1187:80: error: 'useweather' was not declared in this scope
C:\Users\Максим\Documents\Arduino\GyverPanelWiFi_v1.09\clock.ino: In function 'void checkAutoMode6Time()':
clock:1207:80: error: 'useweather' was not declared in this scope
exit status 1
'useweather' was not declared in this scope
Помогите, в чем причина?

vvip-68, 2021-01-22:
Затрудняюсь сказать. Вы что-то исправляли?
В моем экземпляре скетча в 'mn[ строках - переменная useWeather.
А переменная useweather - действительно неизвестна...

Dmitro4ekalov, 2021-01-22:
да, useWeather на useweather я сам исправлял, думал может ошибка в большой букве W.
а вот с оригиналом так же
C:\Users\Максим\Documents\Arduino\GyverPanelWiFi_v1.09\clock.ino: In function 'void checkAutoMode5Time()':
clock:1187:80: error: 'useWeather' was not declared in this scope
C:\Users\Максим\Documents\Arduino\GyverPanelWiFi_v1.09\clock.ino: In function 'void checkAutoMode6Time()':
clock:1207:80: error: 'useWeather' was not declared in this scope
exit status 1
'useWeather' was not declared in this scope

vvip-68, 2021-01-22:
А... У вас наверное в настройках стоит `#define USE_WEATHER 0`
Мой косяк. Сейчас исправлю...

vvip-68, 2021-01-22:
Исправил, забирайте 
clock.ino

closed 2021-01-22 by vvip-68