vvip-68/GyverPanelWiFi #114: ошибки при компиляции ESP32

Dmitro4ekalov, 2021-01-23:
In file included from sketch\a_def_hard.h:432:0,

                 from C:\Users\Maxim\Documents\Arduino\GyverPanelWiFi_v1.09\GyverPanelWiFi_v1.09.ino:19:

C:\Users\Maxim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src/ArduinoOTA.h:84:5: error: 'WiFiUDP' does not name a type

     WiFiUDP _udp_ota;

     ^

                       ^

C:\Users\Maxim\Documents\Arduino\GyverPanelWiFi_v1.09\GyverPanelWiFi_v1.09.ino: In function 'void startWiFi(long unsigned int)':

GyverPanelWiFi_v1.09:325:23: error: no matching function for call to 'WiFiClass::disconnect(bool)'

   WiFi.disconnect(true);

                       ^

In file included from C:\Users\Maxim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src/ArduinoOTA.h:4:0,

                 from sketch\a_def_hard.h:432,

                 from C:\Users\Maxim\Documents\Arduino\GyverPanelWiFi_v1.09\GyverPanelWiFi_v1.09.ino:19:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:130:9: note: candidate: int WiFiClass::disconnect()

     int disconnect(void);

         ^

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:130:9: note:   candidate expects 0 arguments, 1 provided

GyverPanelWiFi_v1.09:329:8: error: 'class WiFiClass' has no member named 'mode'

   WiFi.mode(WIFI_STA);

        ^

GyverPanelWiFi_v1.09:329:13: error: 'WIFI_STA' was not declared in this scope

   WiFi.mode(WIFI_STA);

             ^

GyverPanelWiFi_v1.09:341:40: error: no matching function for call to 'WiFiClass::config(IPAddress, IPAddress, IPAddress, IPAddress, IPAddress)'

                   IPAddress(8, 8, 8, 8));                                 // DNS2 8.8.8.8                  

                                        ^

In file included from C:\Users\Maxim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src/ArduinoOTA.h:4:0,

                 from sketch\a_def_hard.h:432,

                 from C:\Users\Maxim\Documents\Arduino\GyverPanelWiFi_v1.09\GyverPanelWiFi_v1.09.ino:19:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:85:10: note: candidate: void WiFiClass::config(IPAddress)

     void config(IPAddress local_ip);

          ^

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:85:10: note:   candidate expects 1 argument, 5 provided

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:92:10: note: candidate: void WiFiClass::config(IPAddress, IPAddress)

     void config(IPAddress local_ip, IPAddress dns_server);

          ^

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:92:10: note:   candidate expects 2 arguments, 5 provided

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:100:10: note: candidate: void WiFiClass::config(IPAddress, IPAddress, IPAddress)

     void config(IPAddress local_ip, IPAddress dns_server, IPAddress gateway);

          ^

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:100:10: note:   candidate expects 3 arguments, 5 provided

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:109:10: note: candidate: void WiFiClass::config(IPAddress, IPAddress, IPAddress, IPAddress)

     void config(IPAddress local_ip, IPAddress dns_server, IPAddress gateway, IPAddress subnet);

          ^

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:109:10: note:   candidate expects 4 arguments, 5 provided

C:\Users\Maxim\Documents\Arduino\GyverPanelWiFi_v1.09\GyverPanelWiFi_v1.09.ino: In function 'void startSoftAP()':

GyverPanelWiFi_v1.09:403:8: error: 'class WiFiClass' has no member named 'softAPdisconnect'

   WiFi.softAPdisconnect(true);

        ^

GyverPanelWiFi_v1.09:409:23: error: 'class WiFiClass' has no member named 'softAP'

   ap_connected = WiFi.softAP(apName, apPass);

                       ^

GyverPanelWiFi_v1.09:425:27: error: 'class WiFiClass' has no member named 'softAPIP'

       Serial.println(WiFi.softAPIP());

                           ^

GyverPanelWiFi_v1.09:429:10: error: 'class WiFiClass' has no member named 'enableAP'

     WiFi.enableAP(false);

          ^

GyverPanelWiFi_v1.09:430:10: error: 'class WiFiClass' has no member named 'softAPdisconnect'

     WiFi.softAPdisconnect(true);

          ^

GyverPanelWiFi_v1.09:434:25: error: 'class WiFiClass' has no member named 'softAP'

     ap_connected = WiFi.softAP(apName, apPass);

                         ^

C:\Users\Maxim\Documents\Arduino\GyverPanelWiFi_v1.09\GyverPanelWiFi_v1.09.ino: In function 'void connectToNetwork()':

GyverPanelWiFi_v1.09:448:10: error: 'class WiFiClass' has no member named 'mode'

     WiFi.mode(WIFI_AP);

          ^

GyverPanelWiFi_v1.09:448:15: error: 'WIFI_AP' was not declared in this scope

     WiFi.mode(WIFI_AP);

               ^

C:\Users\Maxim\Documents\Arduino\GyverPanelWiFi_v1.09\a_main.ino: In function 'void parsing()':

a_main:1950:22: error: 'class WiFiClass' has no member named 'softAPdisconnect'

                 WiFi.softAPdisconnect(true);

                      ^

exit status 1

no matching function for call to 'WiFiClass::disconnect(bool)'

подскажите, в чем причина?

vvip-68, 2021-01-23:
Нет, не подскажу. Вы первый и единственный у кого такая ошибка. Причины не понимаю. Cудя по логу - версия ядра актуальная, все на что ругается - входит в состав ядра ESP32. Попробуйте переустановить всё - начиная с Arduino IDE, заканчивая всеми используемыми библиотеками и ядром ESP32. Другого совета у меня нет.

vvip-68, 2021-01-23:
У вас какая-то смесь из библиотек, например
`C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:130:9: note: candidate: int WiFiClass::disconnect()`
Что за библиотека WiFi, которая пытается использоваться при компиляции?
Все что нужно для работы скетча - находится внутри ядра ESP32
Удалите эту папку с библиотекой `C:\Program Files (x86)\Arduino\libraries\WiFi` для начала

closed 2021-01-23 by vvip-68

Mosrom, 2021-04-03:
После изменений 6 дней назад файла clock.ino строк 870-876: 
       if (((useWeather > 0)) && weather_ok && (((second() + 10) % 30) >= 28)) {
        uint8_t atH = abs(temperature) / 10;
        uint8_t atL = abs(temperature) % 10;
        display.point(false);
        if (atH == 0)
          display.displayByte(_empty, (temperature >= 0) ? _empty : _dash, display.encodeDigit(atL), _degree);
        else
возникает ошибка компиляции (см. скрин), до этих изменений скетч компилировался без ошибок.          
[image]

vvip-68, 2021-04-03:
Не подскажу. У меня таких проблем не возникает - всё собирается без ошибок.
Я там поубирал некоторые warnings компилятора. Попробуйте собрать обновленную версию.

Mosrom, 2021-04-03:
Спасибо. Помогло.