vvip-68/GyverPanelWiFi #115: Титры и анимация с CD карты

Vovaka1, 2021-01-23:
Здравствуйте! 
После установки интервала вывода титров 15 секунд, перестали чередоваться эффекты с CD карты. 
В общем, после титра эффект начинает идти с начала. Эффекты с CD карты не подчиняются общей установки "Время режима", а идут полностью, от начала до конца. Если время интервала вывода титров больше чем эффект, то эффекты чередуются. Можно изменить алгоритм?

vvip-68, 2021-01-23:
Титры и эффекты с SD карты никак не связаны. Это совершенно отдельные параллельные алгоритмы, пересекающиеся только в моменте, если какой-то эффект имеет отключенный режим "Текст поверх эффекта". Ну или если в бегущей строке есть макрос конкретного эффекта на фоне которого должна бежать строка". В противном случае - текст показывается сам по себе, эффекты поверх которого он показывается меняется независимо. Смотите что из этого мешает в вашем случае и прерывает показ текста.

Эффекты не подчиняются общей установке "Время эффекта" по многочисленным просьбам повторяющих - они просили сделать именно так, чтобы эффект с SD-карты не зависел от установок времени эффекта и проигрывался до конца.

Впрочем, в a_def_soft.h в строке 573 есть определение
#define WAIT_PLAY_FILE_FINISHED 1
которое регулирует этот аспект

Vovaka1, 2021-01-23:
С временем эффекта понятно. Это можно решить создав новые, с нужной продолжительностью. А вот проблема с зацикливанием эффекта не уходит. В бегущей строке погода и время без макроса эффекта. Показ бегущей строки не прерывается, вопросов нет. А вот минутный эффект (с SD карты) воспроизводится постоянно, пока не изменю интервал вывода текста, на больший по времени, чем эффект или пока не нажму на кнопку "С" (следующий).

vvip-68, 2021-01-23:
У меня как раз сейчас крутятся эффекты с SD-карты, интервал для строки 15 сек - поставил настройки как вы сказали. Никаких проблем не наблюдаю - все меняется, все показывается. Поскольку повторить "ошибку" не могу - устранить ее, соответственно. тоже не получится. Предлагаю либо найти и устранить это самостоятельно, потом сообщить мне в чем было дело, либо считать данное поведение "фичей", а не "багом".

closed 2021-01-23 by vvip-68

Vovaka1, 2021-01-23:
ОК.