vvip-68/GyverPanelWiFi #116: GyverPanelWifi

Ady-67, 2021-02-01:
Hi,please give me more instructions regarding the alarm sounds.
I can't make them work, I followed all the installation instructions, everything works perfectly.

vvip-68, 2021-02-01:
Could you ask more specific question? "Help me regarding the alarm sounds" is extrimely common question.

Ady-67, 2021-02-01:
So,the alarm sounds are not played by dfplayer.

vvip-68, 2021-02-01:
What did you do for alarm sounds being played by dfplayer?
Did you put alarms sound on SD card into proper folder?
Did yoг select sound file in application in Alarm Settings page?

vvip-68, 2021-02-01:
Did you connect MP3 DFPlayer board properly to main controller?
Does port monitor logs contains message that MP3 Player is properly initialized and found sound files on SD-card?
Does speaker properly connected to DFPlayer board?
Can you hear sound on speaker when short connect IO1/IO2 pin on DFPlayer board with GND pin?

Ady-67, 2021-02-01:
Yes, I put on the sd card the three files 01.02.03
I downloaded the application but at the application when I set the
alarm the part with the sound activation disappears and I don't know
why?

Ady-67, 2021-02-01:
I connected everything according to the scheme, indeed on the Serial
monitor it does not show me the initialization of the player with 9600 baud.

If you briefly connect the IO1 / IO2 pins with gnd, run the mp3 files
on the sdcard.

vvip-68, 2021-02-01:
> If you briefly connect the IO1 / IO2 pins with gnd, run the mp3 files on the sdcard.

So, DFPlayer is functional

>indeed on the Serial monitor it does not show me the initialization of the player with 9600 baud.

What do you see on Serial monitor at all? Connection speed is set to 115200
Do you see anything in monitor?

> alarm the part with the sound activation disappears and I don't know why?

this part is not visible if sketch has turned off definition USE_MP3 - set to 0

Ady-67, 2021-02-01:
1.Do I have to put the sound files separately or all three in one (sound)?

2. The serial monitor at 115200 shows less and less information about
initializing dfplayer
3. I chose variant 5 from the sketch with 16x16 matrix, mp3 set 1 and
weather set 1

vvip-68, 2021-02-01:
1. Files on SD card MUST be copied into SD-card root as is in three separate folders 01, 02 and 03
2. Give me content of port monitor log immediately after microcontroller is restarted.
3. Variants 1-5 are variant (hardware profile) for MY OWN devices. You MUST specify parameters for YOUR device, which is higly likely differs from MY device.

Ady-67, 2021-02-01:
Image 1 and 2 is Serial monitor without DFPlayer connected and image 3 is
with DFPlayer connected.

vvip-68, 2021-02-01:
There are no images in post.
It is better to copy all ccontent of log as plain text.

Ady-67, 2021-02-01:
Would you receive the pictures?

vvip-68, 2021-02-01:
No. No pictures, no plain text logs. Nothing. No any valuable information.
You own see that above - nothing.

Ady-67, 2021-02-01:
I took the pictures with the phone and they are 6MB and he doesn't want to
send them by email.

I try to make them smaller.

Ady-67, 2021-02-01:
WiFiPanel-v.1.09.2021.0118
18:28:34.552 -> Host: 'WiFiPanel-5' >> 16x16
18:28:34.552 ->
18:28:34.552 -> Загрузка строк выполнена.
18:28:34.552 -> Использованы адреса EEPROM 800 - 837
18:28:34.552 -> Свободно ячеек 3258
18:28:36.803 ->
18:28:36.803 -> Инициализация файловой системы...
18:28:36.803 -> FS: Использовано 16384 из 1024000 байт
18:28:36.803 ->
18:28:36.803 -> Инициализация MP3 плеера.
18:28:42.990 -> Звуков будильника найдено: 0
18:28:42.990 -> Звуков рассвета найдено: 0
18:28:42.990 -> Звуков сообщений найдено: 0
18:28:42.990 -> MP3 плеер недоступен.
18:28:43.177 ->
18:28:43.177 -> Подключение к HD86SAG -> 192.168.0.100
18:28:43.271 -> .............
18:28:47.302 -> WiFi подключен. IP адрес: 192.168.0.100
18:28:47.302 ->
18:28:47.302 -> UDP-сервер на порту 2390
18:28:47.302 -> Режим: The Matrix
18:28:47.302 -> NTP-сервер ru.pool.ntp.org -> 193.192.36.3
18:28:47.302 -> Отправка NTP пакета на сервер ru.pool.ntp.org
18:28:47.302 ->
18:28:47.302 -> Запрос текущей погоды
18:28:47.490 -> Погода получена: OpenWeatherMap
18:28:47.490 -> Город: Simeria
18:28:47.490 -> Сейчас: пасмурно, +3ºC
18:28:47.490 -> Код иконки: '04n'
18:28:47.490 -> Код погоды: 804
18:28:47.490 -> Сейчас: Темное время суток
18:28:47.490 -> Рассвет: 07:49
18:28:47.490 -> Закат: 17:33
18:28:47.490 ->
18:28:47.490 -> UDP << ip='193.192.36.3:123'; ntp sync
18:28:47.537 -> UDP пакeт размером 48
18:28:47.537 -> Разбор пакета NTP
18:28:47.537 -> Секунд с 1970: 1612204126
18:28:47.537 -> Текущее время: 01.02.2021 18:28:46
18:28:47.537 -> --------------------

Ady-67, 2021-02-01:
If I connect dfplayer, nothing starts and in the Serial monitor only the
time and a series of signs appear.

Ady-67, 2021-02-01:
18:37:46.841 -> WiFiPanel-v.1.09.2021.0118
18:37:46.841 -> Host: 'WiFiPanel-5' >> 16x16
18:37:46.841 ->
18:37:46.841 -> Загрузка строк выполнена.
18:37:46.841 -> Использованы адреса EEPROM 800 - 837
18:37:46.888 -> Свободно ячеек 3258
18:37:49.125 ->
18:37:49.125 -> Инициализация файловой системы...
18:37:49.125 -> FS: Использовано 16384 из 1024000 байт
18:37:49.125 ->
18:37:49.125 -> Инициализация MP3 плеера.
18:37:55.326 -> Звуков будильника найдено: 0
18:37:55.326 -> Звуков рассвета найдено: 0
18:37:55.326 -> Звуков сообщений найдено: 0
18:37:55.326 -> MP3 плеер недоступен.
18:37:55.513 ->
18:37:55.513 -> Подключение к HD86SAG -> 192.168.0.100
18:37:55.560 -> ..............
18:37:59.872 -> WiFi подключен. IP адрес: 192.168.0.100
18:37:59.872 ->
18:37:59.872 -> UDP-сервер на порту 2390
18:37:59.872 -> Режим: Светлячки
18:37:59.872 -> NTP-сервер ru.pool.ntp.org -> 80.249.145.122
18:37:59.872 -> Отправка NTP пакета на сервер ru.pool.ntp.org
18:37:59.872 ->
18:37:59.872 -> Запрос текущей погоды
18:38:00.341 -> Погода получена: OpenWeatherMap
18:38:00.341 -> Город: Simeria
18:38:00.341 -> Сейчас: пасмурно, +3ºC
18:38:00.341 -> Код иконки: '04n'
18:38:00.341 -> Код погоды: 804
18:38:00.341 -> Сейчас: Темное время суток
18:38:00.341 -> Рассвет: 07:49
18:38:00.341 -> Закат: 17:33
18:38:00.341 ->
18:38:04.888 -> Таймаут NTP запроса!
18:38:29.878 -> Режим: Радужные переливы
18:38:59.873 -> Режим: Часы
18:38:59.873 -> NTP-сервер ru.pool.ntp.org -> 80.249.145.122
18:38:59.873 -> Отправка NTP пакета на сервер ru.pool.ntp.org
18:38:59.920 -> UDP << ip='80.249.145.122:123'; ntp sync
18:38:59.967 -> UDP пакeт размером 48
18:38:59.967 -> Разбор пакета NTP
18:38:59.967 -> Секунд с 1970: 1612204738
18:38:59.967 -> Текущее время: 01.02.2021 18:38:58
18:38:59.967 -> --------------------
18:39:29.885 -> Режим: Шумящий лес
18:39:59.857 -> Режим: Циклон
18:40:29.882 -> Режим: Погода
18:40:59.896 -> Режим: Вышиванка
18:41:29.922 -> Режим: Лабиринт
18:42:08.431 -> Режим: Светлячки
18:42:38.450 -> Режим: Лампа

vvip-68, 2021-02-01:
> 18:28:36.803 -> Инициализация MP3 плеера.
18:28:42.990 -> Звуков будильника найдено: 0
18:28:42.990 -> Звуков рассвета найдено: 0
18:28:42.990 -> Звуков сообщений найдено: 0
18:28:42.990 -> MP3 плеер недоступен.

This means that DFPlayer has no detect SD-card and did not found any sound files.
By this reason application settings has no section related with settings of sounds

As I remember - this log is for case when DFPlayer is not connected.

> If I connect dfplayer, nothing starts and in the Serial monitor only the
time and a series of signs appear.

It means that you connect DFPlayer with not propper way, some of pins of microcontroller has inappropriate levels on its start, that lead to unable to start MCU in normal mode.

Could you provide pointer to scheme that you have used for your device and real photo of connections between MCU and DFPlayer?

Ady-67, 2021-02-01:
I made this variant.I followed the scheme exactly.

Ady-67, 2021-02-01:
When I connect the TX pin from DFPlayer to pin D4 from ESP it locks.

Ady-67, 2021-02-01:
[image]

vvip-68, 2021-02-01:
So, you still didn`t provide photo of your device and connections. How I can help you under this conditions?

Ady-67, 2021-02-01:
This is.

Ady-67, 2021-02-01:
[image]
![Uploading IMG_20210201_194842.jp
[image]
g…]()

Ady-67, 2021-02-01:
[image]

Ady-67, 2021-02-01:
[image]

Ady-67, 2021-02-01:
[image]

Ady-67, 2021-02-01:
[image]

Ady-67, 2021-02-01:
[image]

Ady-67, 2021-02-01:
I uploaded the pictures 25 minutes ago and they still haven't uploaded.
What city are you from in Russia?

vvip-68, 2021-02-01:
Does it matter? github servers are located somewere in USA, I think...

Ady-67, 2021-02-01:
No, that's not why I asked. I hope you weren't upset.

vvip-68, 2021-02-01:
It will not help to solve issue

Ady-67, 2021-02-01:
Ok.

Ady-67, 2021-02-01:
Talk to you soon and keep up the good content.

Ady-67, 2021-02-02:
[image]

Ady-67, 2021-02-02:
[image]

Ady-67, 2021-02-02:
[image]

vvip-68, 2021-02-02:
Do you have smth like google drive account?
You may upload files onto your drive and put links here.

Ady-67, 2021-02-02:
Okay, later because I'm at work now.

Ady-67, 2021-02-02:
https://drive.google.com/file/d/1pqOLJokTF6UkS5ubiWW5Q0S9YOtJqfzT/view?usp=drivesdk

Ady-67, 2021-02-02:
https://drive.google.com/file/d/1prWqBMa2l_LH6yaPMU8go28YGXBjxZGR/view?usp=drivesdk

Ady-67, 2021-02-02:
https://drive.google.com/file/d/1qFuUm9Zb5B0ZMIDUzVEgCDrh5-6znlOA/view?usp=drivesdk

vvip-68, 2021-02-02:
No access. Content is blocked. For all 3 images

Ady-67, 2021-02-02:
how to unlock them

Ady-67, 2021-02-02:
try now

vvip-68, 2021-02-02:
I didn`t see any suspicious on connections...
A you sure thar resistors on Rx/TX lines are 1k?
Show me file structure on SD-card...
Could you try another card?
I haму no ideas anymore...

vvip-68, 2021-02-02:
Do you have another DFPlayer board for try it?
I assembled about 4-5 devices with sound features. It works
However I had 2-3 DFPlayer board, which I couldn`t make it working.
By unknown reason.

vvip-68, 2021-02-02:
Try temporary excange wires D3<->D4 on MCU...

Ady-67, 2021-02-02:
I guess the Nodemcu board, in the winter I made AlexGyver's garland with
the sdcard module and Jinx effects and out of three Nodemcu modules, only
one initializes the sd card.

Ady-67, 2021-02-02:
https://drive.google.com/file/d/1qTE9tZVzdKxTGvpb4LPdCF_tZ4pprGUU/view?usp=drivesdk

Ady-67, 2021-02-02:
I managed to solve the problem.
1.DFPlayer defective
2.DFPlayer must be powered at 3.3 V otherwise it will not initialize.

Ady-67, 2021-02-02:
Thank you for your cooperation.

vvip-68, 2021-02-03:
Thank you for information. DFPlayer datasheet http://files.amperka.ru/datasheets/DFPlayer_Mini.pdf
specify operation voltage as 3.3V-5V. Probably your power supply unit output gives more than 5V.
Usually chines pover supply units give output 5.4-5.6 Volts
It is highly recommended for ESP8255 MCU use pover supply as 4.8V, otherwise is bad intercommunication between LEDs and MCU - inappropriate logical levels lead to digital artefacts on LED matrix

Where did you get 3.3V? As I understand DFPlayer has built-in 3W amplifier. It means on power supply 3.3 V requite circuit in about 1A. As I understand 3.3V pin of NodeMCU is able to provide not more that 300-500 mA of currency.
Did you add special power supply voltage converter to 3.3V specially for DFPlayer?

vvip-68, 2021-02-03:
Also in case of VCC (Input Power) level 5.4-5.6V from power supply unit output it is possible provide power supply for MCU and DFPlayer through cremnium diode as it shown on following picture:
[image]

Ady-67, 2021-02-03:
I took 3.3 V from the Nodemcu module and it works perfectly.

I will do more tests and I will measure the supply voltage and I will
tell you the results.

Ady-67, 2021-02-03:
I noticed that it does not play sounds clearly.
Could it be because of the Nodemcu module because it has limited current?

Ady-67, 2021-02-03:
I will measure and if the voltage is higher than 5V I will use a diode.

vvip-68, 2021-02-03:
> I noticed that it does not play sounds clearly.
Could it be because of the Nodemcu module because it has limited current?

May be several reasons for unclear sound
1. Bad speaker with work power less then output power of amplifier (speaker overload)
2. Low / limited / not enough  current from 3.3v output pin
3. Position/location of speaker relative underlaying surface

High current from 3.3v pin may lead to overheat of onboard stabilizer and make it broken / malfunctional.

Ady-67, 2021-02-03:
OK thanks

[10 more replies]

closed 2021-02-04 by vvip-68