vvip-68/GyverPanelWiFi #117: GyverPanelWiFi_v1.09 не воспроизводит мр3

120934, 2021-02-15:
Здравствуйте,
заменил версию GyverPanelWiFi_v1.07 на прошивку / GyverPanelWiFi_v1.09 после чего DFPlayer перестал воиспроизводить mp3 файлы. Установки из прошивки 1.07 были 1: 1 переняты в прошивку 1.09 по железу изменений небыло. Если я прошываю с тестовым Sketchem то все работат. На смартфоне прошивка обновлена и возможность {An +} тагже не работает. SD-карта распознается и считывается при запуске, отображается количество mp3 файлов. Помогите мне решить эту ошибку.
С уважением Виктор

vvip-68, 2021-02-16:
Затрудняюсь что-то подсказать. Изменений по коду что относится к поддержке DFPlayer с версии 1.07 до 1.09 не было - там всё осталось как было (за исключением {An+}. На днях выложу версию 1.10 - сегодня буду как раз заниматься тем, что в ней связано с DFPlayer, хотя опять же - никаких изменений и в 1.10 по звуку не было. Если в 1.10 обнаружатся проблемы со звуком - поправлю. Ожидайте выход новой версии прошивки и приложения.

vvip-68, 2021-02-16:
Проверка на 1.10 показала (как и ожидалось) - никаких проблем со звуком нет. Поскольку с версии 1.09 до 1.10 никаких изменений в коде, связанных со звуком не было - думаю и на 1.09 все тоже вполне должно работать. Ищите ошибку у себя в настройках. В приложении на вкладке "Будильник" видны в самом низу элементы управления звуком? Галочка "Использовать звук" стоит? Громкость звука не сведена в минимум?

120934, 2021-02-16:
Я на работе, но завтра перейду на версию 1.10.

В приложении на вкладке "Будильник" видны в самом низу элементы управления звуком? ->   Элементы управления видны

Галочка "Использовать звук" стоит? - > да

Громкость звука не сведена в минимум? - > нет стоит на максимуме

 

Gesendet: Dienstag, 16. Februar 2021 um 04:32 Uhr

closed 2021-02-17 by vvip-68

120934, 2021-02-23:
Sorry habe es nicht geschafft rechtzeitig zu antworten, der Fehler ist geblieben ich bekomme kein Soundausgabe mit der Ver. 1.09 und 1.10. Heute habe ich ein neue NODE MCU und ein neue Playerkarte auf einem Steckbrett aufgebaut den Sketch aktuell geladen und nur das nötige konfiguriert. (WLAN Zugang, Matrixeinstellung).
Die Songs lassen sich durch I/O1 oder I/O2 und GND abspielen. Im Seriellem Monitor wird der Wecker sauber abgearbeitet ( Siehe unten).  Wecker Einstellungen auf dem Smartphone sind sichtbar. Die Lautsterke ist MAX. Es wird nur kein Sound erzeugt. Mit der Ver. 1.07 die gleiche Schaltung funktioniert der Sound. Könne Sie mir weiter helfen?

11:44:09.700 -> UDP << ip='Telefon:2390'; cmd='$18 0;'
11:44:09.747 -> UDP пакeт размером 6
11:44:14.932 -> Карта удалена. -----------------------------------------------> 1
11:44:14.932 -> MP3 плеер недоступен.
11:44:15.120 -> UDP << ip='Telefon:2390'; cmd='$18 0;'
11:44:15.120 -> UDP пакeт размером 6
11:44:18.762 -> Карта готова.--------------------------------------------------->2
11:44:18.762 -> 
11:44:18.762 -> Инициализация MP3 плеера.
11:44:19.182 -> Звуков будильника найдено: 7
11:44:19.182 -> Звуков рассвета найдено: 7
11:44:19.229 -> Звуков сообщений найдено: 7
11:44:20.539 -> UDP << ip='Telefon:2390'; cmd='$18 0;'
11:44:20.539 -> UDP пакeт размером 6
11:44:25.066 -> UDP << ip='Telefon:2390'; cmd='$6 6|10 43 2 1 0 00 11 45 0 00 0 00 0 00 0 00 0 00' ---------------------------->3
11:44:25.113 -> UDP пакeт размером 50
11:44:25.113 -> Следующий рассвет в 11:35, вторник
11:44:25.113 -> UDP Telefon:2390 >> $18 UP:4294946273|AL:0|AW:0.1.0.0.0.0.0|AT:1 0 0|AT:2 11 45|AT:3 0 0|AT:4 0 0|AT:5 0 0|AT:6 0 0|AT:7 0 0|AD:10|AE:43|MX:1|MU:1|MD:1|MV:26|MA:3|MB:3|MP:0~2; --------------------------------------------------------------------->4
11:44:25.253 -> UDP Telefon:2390 >> $18 AL:1;
11:44:25.253 -> Рассвет ВКЛ в 11:44 --------------------------------------------------->5
11:44:28.569 -> UDP << ip='Telefon:2390'; cmd='$20 2 1 26 3 3;'
11:44:28.569 -> UDP пакeт размером 15
11:44:28.616 -> UDP Telefon:2390 >> $18 UP:4294946276|AL:0|AW:0.1.0.0.0.0.0|AT:1 0 0|AT:2 11 45|AT:3 0 0|AT:4 0 0|AT:5 0 0|AT:6 0 0|AT:7 0 0|AD:10|AE:43|MX:1|MU:1|MD:1|MV:26|MA:3|MB:3|MP:0~2;
11:44:28.755 -> UDP Telefon:2390 >> $18 AL:1;
11:44:28.755 -> Рассвет ВКЛ в 11:44--------------------------------------------------->6
11:44:33.941 -> UDP << ip='Telefon:2390'; cmd='$18 0;'
11:44:33.941 -> UDP пакeт размером 6
11:44:33.941 -> Ответ на Telefon:2390 >> AL:1ack99;
11:44:39.358 -> UDP << ip='Telefon:2390'; cmd='$18 0;'
:
11:45:16.814 -> UDP пакeт размером 6
11:45:22.040 -> UDP << ip='Telefon:2390'; cmd='$18 0;'
11:45:22.086 -> UDP пакeт размером 6
11:45:24.794 -> Рассвет Авто-ВЫКЛ в 11:45 --------------------------------------->7
11:45:24.934 -> UDP Telefon:2390 >> $18 AL:1;
11:45:24.934 -> Режим: Лампа
11:45:27.453 -> UDP << ip='Telefon:2390'; cmd='$18 0;'
11:45:27.453 -> UDP пакeт размером 6
11:45:27.453 -> Ответ на Telefon:2390 >> AL:1ack109;
11:45:32.872 -> UDP << ip='Telefon:2390'; cmd='$18 0;'
:
11:45:38.247 -> UDP пакeт размером 6
11:45:39.814 -> Настройки сохранены в EEPROM
11:45:43.654 -> UDP << ip='Telefon:2390'; cmd='$18 0;'

simonovich, 2021-04-23:
В приложении доступны звуки только из папок 01 и 02.
Для чего папки 03, 04, 05?

vvip-68, 2021-04-23:
Планы на будущее, связанные с часами. 03 - отбивка часового сигнала одним из заданных звуков. 04 и 05 - проговаривание каждого часа мужским / женским голосом.

simonovich, 2021-04-23:
Ещё немного дилетанский вопрос. Зачем резисторы на TX / RX плеера?
У меня вроде и без них работает, дыма нет...

vvip-68, 2021-04-24:
Резисторы - из даташита на DF-плеер. Правда, там они указаны для 5-вольтовой логики ардуино. В случае 3.3 вольта ESP8266 они не нужны. Рисовались по схемам из интернета, когда еще вопрос не был достаточно изучен. Так и остались в схеме.