vvip-68/GyverPanelWiFi #119: О DFPlayer mini

KAKTO, 2021-02-22:
Приобрёл у одного продавца DFPlayer (микросхема YX5200), собрал панель с прошивкой v1.08 (обновлял по ходу доработок версии) - как я был счастлив, всё отлично работает.
У того же продавца купил ещё несколько, но обнаружил только через 2 дня возни вплоть до полной разборки панели, что DFPlayer пришёл уже с 16ногой микросхемой GD3200B (и 3 контакта у картоприёмника по задумке производителя не припаиваются). Версии перебрал разные включая v1.08 и соответствующие приложения. 
Пробовал запускать скетчем с Wemos - DFPlayer работает, и напрямую без контроллера как плеер работает.
Проблема в том, что звуки оповещения воспроизводятся при просмотре в приложении, но звуки будильника и рассвета при нажатии "плей" не воспроизводятся и светодиод на DFPlaerе не загорается (вообще с папками 01 и 02 не загорался).
Заметил: после манипуляций на странице настроек будильника (и всегда после нажатия "использовать звук") так и не услышав признаков жизни возвращаюсь к бегущей строке (она-то жива), а там звук тоже перестаёт работать, но светодиод загорается - думаю звук сбрасывается на 0, потому что можно добавить звук на самом DFPlaerе (или просто выкл/вкл питание) и дальше просматривать строку со звуком до следующих ковыряний в настройках будильника.
И ещё, в мониторе порта звуков найдено по 10, хотя в папках 10,10,18. И так вне зависимости от количества файлов в папках выбирает количество файлов первой (я так понял по счёту) папки и выставляет "звуков найдено" для всех 3 папок (даже если на SD всего одна папка!!!)

02:20:16.209 -> WiFiPanel-v.1.08.2021.0103
02:20:16.209 -> Host: 'WiFiPanel-0' >> 16x16
02:20:16.209 -> 
02:20:16.209 -> Загрузка строк выполнена.
02:20:16.209 -> Использованы адреса EEPROM 800 - 851
02:20:16.209 -> Свободно ячеек 3244
02:20:16.537 -> 
02:20:16.537 -> Инициализация MP3 плеера.
02:20:16.959 -> Звуков будильника найдено: 10
02:20:16.959 -> Звуков рассвета найдено: 10
02:20:16.959 -> Звуков сообщений найдено: 10
02:20:17.193 -> Создание точки доступа PanelAP
02:20:19.068 -> Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
02:20:19.115 -> IP адрес: 192.168.4.1
02:20:19.115 -> 
02:20:19.115 -> UDP-сервер на порту 2390
02:20:19.115 -> Режим: Зебра
 
Действительно непоправимая проблема в DFPlaerе - продавцу-то предъявить нечего - вроде всё работает (хотя микросхема совсем не та, что на фото у продавца).
Странно то, что звуки уведомления воспроизводит, а рассвета и будильника даже признаков жизни не подаёт.
Я так и не нашёл зависимость вывода команды на воспроизведение и косяками DFPlaer.
Может быть есть решение проблемы?

vvip-68, 2021-02-23:
Покупал порядка 16 плееров у разных продавцов - все они выглядят одинаково. Одна многоногая микросхема AA19HF8328-94 или AS20HH5883-74. Другая - 8-ногая -8002A CM8K29 1N. Других моделей не не встречал. Ни на одном из них не было микросхем с обозначением XY5200 или GD3200B. Чего там в картоприемнике - вообще не видно что припаяно, что нет.

Эффектов как описано работает/не работает - не наблюдал - вроде всё работает, звук никуда не пропадает, так что - не подскажу. С количеством файлов в папке - тоже все очень странно - ничего такого не наблюдаю. Сколько файлов в папках есть - столько и показывает. 
Подсчет файлов в папках - dfplayer.ino, строки 89-129, вывод в монитор - в 13-17. Ничего подозрительного там не вижу.

Складывается впечатление, что у вас как-то звуки на карте лежат неправильно. Попробуйте удалить всё с карты, скопировать звуки из проекта - папка sounds. Должно в корне карточки быть три папки с именами "01", "02", "03".
В этих папках должны лежать звуковые файлы MP-3 формата, имена начинаться обязательно должны с трехзначного числа - "001", "002" и так далее.

С плеером у меня была единственная проблема - на микроконтроллере сгорел пин D3, - назначенный на передачу команд от микроконтроллера (TX) - идет к пину RX (2) DFPlayer. В результате когда вставляешь / вытаскиваешь SD-карту - плеер присылает посылку данных, что карта вставлена/вытащена, а вот никаких команд получить не может - TX микроконтроллера D3 не работает - команды на RX плеера не приходят - при инициализации команда получения кол-ва файлов в папке не проходит - возвращает в папках 0 - пишет, что DFPlayer недоступен.. 

Как резюме - ничего из описанного в запросе - не наблюдаю, подсказать - тоже не смогу соответственно.

vvip-68, 2021-02-23:
Я не думаю, что от микросхемы что-то зависит. Микросхема вообще, думаю, может быть любой - зависит от производителя, главное, что результирующий продукт соответствует даташиту на него. Управление плеером - через библиотеку, которая все вызовы функций переводит в протокол низкого уровня коммуникации с плеером. Так что, думаю, не в этом всем дело.

Соблюдается ли в вашем устройстве требования даташита - питание DFPlayer - не более 5.00 вольт (в идеале 3.3-4.7 как написано в спецификации)?  Если вы питаете систему напряжением выше 5.00 вольт - все это может и не работать в лучшем случае, а в худшем - плеер просто сгорит...

KAKTO, 2021-02-23:
Получил ещё DFPlaer всё того же производителя, но на этот раз 24 ноги (12) AA20HFJ648 такую же как YX5200 - всё работает отлично.
Правильность сборки и файлов? Всё собрано и работает - заменяю только DFPlaerы из проблемной партии - косячат.
С маркировкой микросхем понятно: как производитель назовёт то, что произвёл - так оно и работает (сегодня в завтрашний день не все могут найти деталь с той же маркировкой)
Надеялся, что команды на воспроизведение звуков всё-таки по разному генерируются и я не могу найти, но действительно подозрительного ничего нет. И Wemos разные ставил (кстати ТОЛЬКО Wemos).
Спасибо за пояснение что и 16 ногая (8) тоже работает, а то я с подозрением смотрел на неё, а она просто косячная (хотя стоит в библиотеку DFPlaer заглянуть) 
Что касается контактов картоприёмника моих косячных 16 ногих (8): если смотреть в слот, то сзади контакты 2,8,9 не припаяны (по китайскому замыслу не предусмотрено), и я не знаю играют ли они какую-то роль.

В том-то и дело страшно заказывать... даже у одного продавца повторный заказ не гарантирует получение того же товара.
Получил партию Wemos, одна была с чистой микросхемой без маркировки (CH340G) и не подключалась. Легко обделался - установил драйвер.
Я к тому, чтоб знать на какие грабли можно наступить и где копать. Не всегда проблемы нужно искать в прошивке.
Закажу другие DFPlaer, вопрос снят, спасибо.

closed 2021-02-25 by vvip-68