vvip-68/GyverPanelWiFi #120: GyverPanelWiFi_v1.10 не воспроизводит мр3

120934, 2021-02-23:
Sorry habe es nicht geschafft rechtzeitig zu antworten, der Fehler ist geblieben ich bekomme kein Soundausgabe mit der Ver. 1.09 und 1.10. Heute habe ich ein neue NODE MCU und ein neue Playerkarte auf einem Steckbrett aufgebaut den Sketch aktuell geladen und nur das nötige konfiguriert. (WLAN Zugang, Matrixeinstellung).
Die Songs lassen sich durch I/O1 oder I/O2 und GND abspielen. Im Seriellem Monitor wird der Wecker sauber abgearbeitet ( Siehe unten). Wecker Einstellungen auf dem Smartphone sind sichtbar. Die Lautsterke ist MAX. Es wird nur kein Sound erzeugt. Mit der Ver. 1.07 die gleiche Schaltung funktioniert der Sound. Könne Sie mir weiter helfen?

[COM.txt](https://github.com/vvip-68/GyverPanelWiFi/files/6028737/COM.txt)

120934, 2021-02-23:
Схема была сделана с новыми деталями. Версия 1.10 не имеет вывода звука.
Еще раз проверил работоспособность на той же схеме, версия 1.07 стабильно работает без ошибок вывода звука.  Не могу найти ошибку.

vvip-68, 2021-02-23:
Я тоже не могу найти ошибку. Работа со звуком не менялась с версии чуть ли не 1.05 или более ранней. Всё остается как было.  И - у меня никаких проблем не наблюдается - всё связанное со звуком работает в 1.10 во всех режимах. Если я не могу воспроизвести "ошибку" - я не могу ее исправить.

vvip-68, 2021-02-23:
Пришлите скриншот приложения с настройками будильника

closed 2021-02-25 by vvip-68