vvip-68/GyverPanelWiFi #121: Проблема с LittleFS ESP32

sepulka123, 2021-02-26:
Добрый день.
Повторил Ваш замечательный проект на ESP32.
Панель 23 х 22  и модуль SD. Ни часов ни плеера ни управления по питанию.
Работает нормально (за исключением нескольких вопросов, но это потом), но при запуске "ESP32 LittleFS Data Upload"
получаю сообщение: LITTLEFS Error: mklittlefs not found! 
Что может быть не так ?

vvip-68, 2021-02-26:
Гуглить надо почему такая ошибка возникает?
Плагин устанавливали из папки tools/LittleFS_Uploader проекта, архив ESP32LittleFS.zip?

sepulka123, 2021-02-26:
Разархивировал ESP32LittleFS.zip , получил
c:\Program Files (x86)\Arduino\tools\ESP32LittleFS\*.*
Перезагрузил Arduino IDE.
В папке c:\Users\Wladimir-i3\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\tools\*.*
Нет файла:    mklittlefs.exe.
  - Создал подпапку data.
        3. В созданной папке data создайте подпапку с именем, соответствующим размерам вашей матрицы. 
   Например для матрицы 16x16 создайте папку "16p16"
А где искать эту папку 
4. В созданную папку 16р16 скопируйте файлы картинок

sepulka123, 2021-02-26:
(Прошу прощения что то "вздрыгнуло" и сообщение ушло недоделанным это продолжение)

Создал подпапку data.
 В созданной папке data создал папку 23p22 скопировал в нее картинку .gif
e:\ESP32\GyverPanelWiFi\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\data\23p22\03a.gif

sepulka123, 2021-02-26:
Может где то в скетче нужно указывать, что используется Система
LittleFS для ESP32 ?

vvip-68, 2021-02-26:
Нет, не нужно. И при чем тут скетч, если у вас из Arduino IDE выгрузка не проходит?
Примеры картинок - в проекте в папке pics/ Но там только для 16x16 картинки.

А ESP8266 ядро у вас установлено? Может там этот файл есть?
У меня пишет - 
mklittlefs : C:\Users\user\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56\mklittlefs.exe

Попробуйте поставьте такой же плагин для ESP8266

vvip-68, 2021-02-26:
> Создал подпапку data.
В созданной папке data создал папку 23p22 скопировал в нее картинку .gif
e:\ESP32\GyverPanelWiFi\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\data\23p22\03a.gif

А при чем тут вообще gif?
Вы теплое с мягким не путайте.
В данном случае - это картинки, сохраненные с матрицы программой WiFiPlayer, которая позволяет их рисовать "пальцем" через смартфон в этом приложении. Там в приложении можно сохранить нарисованную картинку в файловую систему или загрузить ранее нарисованную-сохраненную. Сам файл (содержимое) представляет собой три байта заголовка (размер матрицы) и далее последовательно сохраненные "триплеты" цвета RGB (3 байта) для каждого диода матрицы.

sepulka123, 2021-02-26:
У меня на матрице рисовать можно а сохранять не знаю как.
Не нашёл я такого в приложении.

vvip-68, 2021-02-26:
[image]

sepulka123, 2021-02-26:
У меня это так выглядит.
[image]

sepulka123, 2021-02-26:
Прошу прощения, загружается !
Просто пробовал при выключенной панели. 
А поскольку в приложении на "экране" ничего не менялось, то подумал, что не загружается.

sepulka123, 2021-02-26:
Да, но на "экран" приложения все равно ничего не выводится.

vvip-68, 2021-02-26:
Выбор картинки из списка и нажатие кнопки "Открыть" ТОЛЬКО загружает файл из сохраненного файла на матрицу.
Чтобы получить изображение с матрицы в приложение - его нужно оттуда передать на телефон. Это круглая кнопка со стрелкой вниз - самая левая в ряду кнопок.

closed 2021-02-26 by vvip-68

sepulka123, 2021-11-19:
Добрый день.
Хотел отправить Вам небольшую денежку, но что то не получается.
Спрашивает Ваш адрес. Что туда записать ?
Мой адрес :  sepulka123@gmail.com  
Туда можно отправить Ваш адрес и, может еще, что нужно для оплаты по "ПейПел".

vvip-68, 2021-11-19:
Спасибо за поддержку. Кнопка "Sponsor" вверху странички имеет две ссылки - на отправку доната через PayPal - там нужно только ввести сумму и нажать кнопку "Отправить" - больше ничего не требуется (проверено), никаких адресов.
Вторая ссылка - реквизиты карточки в платежной системе VISA. Не уверен можно ли переводить на карту из за рубежа (не из РФ), но по РФ - точно можно через банк-онлайн, например.

vvip-68, 2021-11-19:
@sepulka123 
Спасибо за поддержку, Владимир.