vvip-68/GyverPanelWiFi #123: Не работает кнопка

ledskl, 2021-03-18:
День добрый, не работает сенсорная кнопка. Менял разные кнопки и платы wemos, nodmcu. Результат только один: очень плавное гашение матрицы и быстрое включение и всё. Скетч не правил, куда копать посоветуете?

vvip-68, 2021-03-18:
Если скетч не правили - возможно причина в этом. По умолчанию в скетче используется обычная физическая кнопка на замыкание. Возможно из за этого. В файле a_def_hard.h в профиле вашего устройства
```
#define BUTTON_TYPE 1         // Тип кнопки: 0 - сенсорная (touch button); 1 - обычная кнопка на замыкание при нажатии (или если кнопки нет вообще)  
```

Если же этот момент настроен - возможно аппаратная проблема кнопки. Сенсорные кнопки очень капризные и требуют тонкой настройки. Возможно для уменьшения чувствительности нужно на кнопку напаять конденсатор 5-15 пф. Возможно на питание кнопки навесить керамический конденсатор 33-100нф.

Симптомы по виду - кнопка у вас считается постоянно нажатой. Исправляйте то, почему она считается нажатой.

ledskl, 2021-03-19:
Разобрался спасибо. На версии скетча 1,08 сенсорная кнопка по умолчанию закомментирована, разрешена физическая, как вы и писали. Все работает без проблем и кнопка и приложение и связь не глючит. На версии 1,10 тоже все поменял, кнопка работает, но приложение на вкладке эффектов глючит, то эффект не выбирается, то галки и ползунки не работают. Приложение обновлялось под 1,10? И после выключения лампы из сети (вилки), включить больше нельзя. Выбирал DEVICE_VARIANT 0, т.к. только кнопка. То же самое на разных платах.

vvip-68, 2021-03-19:
Да, конечно. Приложение обновлялось. И прошивка несоответствующей версии с приложением работать не будет. Или будет "глючить".

closed 2021-03-19 by vvip-68