vvip-68/GyverPanelWiFi #124: Звука будильника нет

RailWar, 2021-03-24:
В Wifi Panel при проверке звуков звук для будильника не проигрался, а звук для рассвета играл. SD карта не вытаскивалась. После выключения лампы стало нормально. Это к сведению что такое бывает

vvip-68, 2021-03-24:
Не смог такого получить при проверке. Всё работает. Опишите последовательность что делать, чтобы получить такой эффект.

RailWar, 2021-03-24:
сам повторить не могу. я к тому что бы пересмотреть моменты когда перечитывается sd в dfplayer

closed 2021-03-24 by RailWar

vvip-68, 2021-03-24:
DFPlayer (сама плата) не имеет возможности считывания. Она такая почти однонаправленная - можно только отправлять команды. При изменении состояния (карту вставили / вытащили) - DFPlayer присылает уведомления о событии с определенным кодом. Можно с карты получить количество файлов в папке (не имена!). Получение кол-ва файлов в папке выполняется в setup() при запуске прошивке и при получении события "карта вставлена". Вообще само устройство DFPlayer - довольно капризное и глючное...