vvip-68/GyverPanelWiFi #125: ошибка в wifi panel

RailWar, 2021-03-24:
В wifi panel 1.10 при нажатии на любую кнопку для строк с 1 до  Z ошибка `Start(1) + length(1)-1 exceeds text length(0)`

vvip-68, 2021-03-24:
Ошибка не наблюдается. Приложите скриншот программы с ошибкой. И последовательность что делаете, чтобы получить ошибку. Версия программы соответствует версии прошивки?

RailWar, 2021-03-24:
[image]
соединяюсь с лампой, захожу в закладку "бегущая строка", нажимаю на любую кнопку кроме "0" - выскакивает ошибка
Версия программы внизу (скриншот). а где увидеть версию прошивки?. собирал и ставил apk в одно время

vvip-68, 2021-03-24:
Последняя версия приложения - 1.10.2021.0225
Возможно там эта ошибка пправлена. Попробуйте ее.
Если нет - будем изучать дальше.

RailWar, 2021-03-24:
обновил. работает. спасибо

closed 2021-03-24 by RailWar