vvip-68/GyverPanelWiFi #126: Пожелания

RailWar, 2021-03-24:
выдавать температуру на дисплей (напр. на несколько сек каждую минуту)
можно ли у yandex узнать температуру на 12 дня?
хотелось бы в wiki инструкцию по OTA
эффект с погодой  - на лампе текст не виден, потому что уходит вправо-назад
сделать звук каждый час. задавать отдельный звук для каждого часа и ночное время. может и писк на полчаса
задать яркость дисплея днем и ночью
есть красивые эффекты в gunner47. Можно их добавить сюда (некоторые)?

vvip-68, 2021-03-24:
1. Она в часах (при отображении часов) и так всё время на экране.
2. Температура вместе с датой (попеременке) показывается через заданные интервалы времени на закладке Настройка часов на матрицах, чей размер не позволяет одновременно показывать время и часы. Насколько я помню, если галка "Показывать дату" нет, а "Показывать температуру" стоит - температура будет показываться через заданные интервалы.
3. Никаких специальных инструкция для прошивке по OTA не требуется - это в общем-то стандартный способ. Когда вы вибираете в ардуино IDE порт куда заливать прошивку - выбирается не COM-порт к которому обычно подключена плата, а из того же списка - сетевой порт микроконтроллера с работающей прошивкой. Далее - все то же самое. Это единственное отличие. Две строчки в вики?
Бывает ситуация, когда сетевые порты не видны. Я не знаю почему и как Arduino IDE находит эти сетевые порты. Иногда помогает несколько раз перезапустить саму Arduino IDE
4. Про эффект с погодой - не понял. Можно фото? Десятки людей тестировали это на своей матрице - никто еще не жаловался.
5. Можно подумать. Пока никому особо не нужно было. В долгий ящик, в общем...
6. Вы ночью картинки разглядываете вместо того, чтобы спать? Ночью обычно эффект ночных часов включают. Для них яркость задается отдельно от яркости всех прочих режимов. Пока никому в голову такой функционал не приходил и не был востребован. Тоже на заметку в долгий ящик.  Потому что кроме уровня яркости требуется еще кучу настроек к этому - например - а со скольки и до скольки это "ночное время", А в разные дни недели оно одинаковое? И так далее.
7. Добавление эффектов в с других ламп - планы есть, времени нету. К тому же "красивые эффекты" - такое себе понятие - чисто индивидуальное.
8. Текст с окантовкой
а. используется стандартная отработанная процедура шрифта, заданного точками (бит есть/нет) в массиве. Окантовка - довольно сложный алгоритм, требующий массы времени на продумывание/реализацию. Времени заниматься этим нет. Да и спроса особого тоже нет - никому это и не нужно было. Вы первый и единственный
б. На малых размерах матриц (а обычно это 16х16 и меньше) - окантовка будет отнимать еще дополнительные пиксели и, как мне кажется, будет выглядеть не очень.

Как итог - если что-то очень хочется, но этого нет и делать не собираются - код проекта открытый. Берем и дорабатываем под свои нужды.

vvip-68, 2021-03-24:
По яндекс и температуру. На текущей реслизации то, что отдает Яндекс - этого нет.
Вроде бы прогноз есть на платных планах обслуживания. Вы готовы платить 5-10 баксов ежемесячно Яндексу, чтобы получать температуру на 12 дня?

RailWar, 2021-03-24:
1. как раз не хотелось выводить часы на матрицу. на мой взгляд портит картинку
2. сам не придумал куда показывать прогноз на 12 дня. но для меня бы инфа была полезной
3. ok
4. белое пятно - это "+". сама цифра ссзади
[image]
5. ок
6. ночью кот будит :), а днем стоит рядом с монитором и боковым зрением вижу. но как-то ярковато
7. поэтому и написал "некоторые". потому что каждому свое ;)
8. да. я сначала сделаю картинку, залью и посмотрю как это будет выглядить. (да, у меня лампа с 16x16) и под плафоном еще мутная пленка от дохлого монитора)

vvip-68, 2021-03-24:
Это странна. У меня большинство изделий 16x16 - на них всё и тестировалось.
Вот как выглядит погода на часах 16x16
[image]

Зачем на этой лампе вообще текст, часы и погода? Всё же размывается достаточно хорошо этими пленками от  монитора. Угадывать в размытом пятне цифры - такое себе...

vvip-68, 2021-03-24:
Хотя, понял - цифры погоды у вас отображаются, но уходят за кривизну лампы, поэтому и не видны. 
Ну тут уж что поделать? Отключите погоду совсем. Текущую температуру пускайте бегущей строкой.
Там, кстати, можно и задавать интервал времени для отображения бегущей строки.

closed 2021-03-24 by RailWar

RailWar, 2021-03-28:
решил сам доработать. может что приживется.
1. 3 раза в минуту на 2 сек температура на дисплее
2. в эффекте "погода" цифры сдвинул левее
3. раз в час звук из папки 4. там лежат 24 файла на каждый час, но в коде жестко условие играть с 7 до 20
[GyverPanelWiFi_v1.10.zip](https://github.com/vvip-68/GyverPanelWiFi/files/6216971/GyverPanelWiFi_v1.10.zip)

vvip-68, 2021-03-28:
Ну, вот - здорово, что самостоятельно нужные доработки делаете.

Эффект "Погода" собираюсь немного переделать - использовать пару ползунков - позиция отображения температуры по X/Y чтобы каждый под себя мог настроить и загнать ее в нужный угол. Пока руки не дошли.

Вот, в том то и проблема доработок, что жестко зашивать в код условия - не по фэн-шую, а каждое незначительное добавление возможности требует добавления пачки побочных настроек. С тем же звуком - как минимум в интерфейсе нужна галка возможности отключить это - не всем и не всегда это нужно, настройки времени с..по.., возможно варианты выбора звука - от простого "бип", до проговаривания каждого часа...

RailWar, 2021-03-28:
да, apk мне не переделать :(

RailWar, 2021-03-28:
хотя...

vvip-68, 2021-03-28:
Вот именно. Там не так уж и сложно.