vvip-68/GyverPanelWiFi #127: pcb

RailWar, 2021-03-24:
для себя разработал плату для esp-12f в sprint layout если кому-то кроме меня интересно могу поделиться - вставите в проект. выглядит так
[image]
готовую уже не сфотогрофировать - запаял и засунул в коробку. разведено все - micro sd, dfplayer, дисплей, тач,  мосфет, порт для usb-uart

vvip-68, 2021-03-24:
Плата - это здорово. Я в этом ничего не понимаю, но если пришлёте всё что нужно - я добавлю это в проект.

RailWar, 2021-03-24:
[lamp8266.zip](https://github.com/vvip-68/GyverPanelWiFi/files/6196693/lamp8266.zip)

vvip-68, 2021-03-24:
Вот спасибо за печатку!
Чтобы не ошибиться - это же для
#define DEVICE_VARIANT 1
?

Дам для варианта 0 и 1 кажется немного отличается распиновка - что к каким пинам подключается.
Этот вариант для тач-кнопки или можно обычную использовать и где-то что-то перемычкой переключается?

vvip-68, 2021-03-24:
Может быть было бы совсем хорошо - если есть какой-то текстик - иснструкция по сборке для "чайников"...

vvip-68, 2021-03-24:
Особенно с инструкцией "как прошивать" по проводам. Для этого же нужен внешний конвертоh USB-TTL? насколько я понимаю или через Arduino (Nano)?

RailWar, 2021-03-24:
#define DEVICE_VARIANT 1 - да
можно и обычную кнопку. там контакт "+" не используется. 
чайники с таким связываться не захотят (наверное). я утюгом делал плату, где покупал на али, где выпаивал из доноров детальки. но напишу-опишу чуть позже.
а прошивать точно так же. просто надо иметь com-uart. напр. на той же cp2102. появится ком-порт

vvip-68, 2021-03-24:
У меня ЧПУ-шка есть. Надо будет попробовать выгравировать плату, как время свободное на это появится...

RailWar, 2021-03-24:
специально делал односторонюю чтобы утюгом . только две перемычки бросить надо

RailWar, 2021-03-24:
вывел туда еще adc. может когда пригодится

closed 2021-03-25 by vvip-68

RailWar, 2021-03-25:
[pcb.docx](https://github.com/vvip-68/GyverPanelWiFi/files/6203823/pcb.docx)

vvip-68, 2021-03-25:
Вот спасибо! Добавил в проект.