vvip-68/GyverPanelWiFi #128: Вышиванка под любой размер\аспект матрицы

kostyamat, 2021-03-26:
https://editor.soulmatelights.com/gallery/831-munch

vvip-68, 2021-03-26:
Спасибо за вариант. Но как-то по мне он не очень. У меня вот матрица 50x20 - так бОльшую часть времени - на две трети черный экран - пока она "расползётся" на всю поверхность. Или бОльшую часть времени - статична - на 2/3 ничего не происходит, только в уголке что-то меняется.

Вообще эффект (по непонятным мне причинам), по крайней мере тот что сейчас в коде - глючен до невозможности. Понять почему - не могу. Периодами он выглядит как через точку в шахматном порядке - четные точки зажигаются, нечетные гаснут. Так и мерцает. А порой при вызове эффекта вообще только черный экран.

Планирую вообще удалить его из прошивки, как только вернусь к ней и найду каким эффектом ее заменить.

closed 2021-03-26 by vvip-68

kostyamat, 2021-03-27:
Я ещё код доработал. В коде переменная rand отвечает за смещение цвета. У нас в прошивке это управляемый параметр, можно подобрать под любую палитру. Вот от этой переменной как раз и зависит шахматность цвета. 
Пс. У нас не глючит. 50х20 проверить не могу, эмулятор не позволяет.

80Stepko08, 2021-03-28:
https://editor.soulmatelights.com/gallery/834-munch-2
kostyamat сделал второй вариант, но почему он вам не показал не знаю

vvip-68, 2021-03-28:
Вот! Этот вариант куда уже как интересней. 
Надо потестить - будет ли он надежно работать или так же как сейчас у меня сбоить будет...
Спасибо за инфо.