vvip-68/GyverPanelWiFi #131: Проблема настройки IDE настройка для доски ESP8266.

Miackl, 2021-04-04:
Доброго времени суток. Плагин для Arduino IDE - загрузчик файлов в файловую систему LittleFS микроконтроллера ESP8266

https://github.com/earlephilhower/arduino-esp8266littlefs-plugin

1. Установите плагин в Arduino IDE:
 a. Перейдите в папку, где установлена Arduino IDE (обычно "C:\Program Files (x86)\Arduino\")
 Ссылка битая, ошибка 404. Как быть?

vvip-68, 2021-04-04:
Да нет, всё открывается. По крайней мере по ссылке, приведенной выше.
Ну а вообще - всё это есть в проекте, в папке tools/LittleFS_Uploader

closed 2021-04-04 by vvip-68

Miackl, 2021-04-05:
Друг, покажи структуру своих библиотек для данного проекта.Все что нужно .

vvip-68, 2021-04-05:
Все что нужно - есть в составе проекта. Библиотеки - в папке Libraries проекта.
Те версии, что в этой папке лежат - актуальные.

Miackl, 2021-04-05:
Arduino: 1.8.13 (Windows 7), Плата:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

C:\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp:17:21: fatal error: vfs_api.h: No such file or directory

 #include "vfs_api.h"

                     ^

compilation terminated.

exit status 1

Ошибка компиляции для платы NodeMCU 1.0 (ESP-12E Module).

Что тогда за ошибка, не хватает чего то?

vvip-68, 2021-04-05:
Какая версия ядра ESP32 у вас?
В версии 1.0.6 этот файл находится в 
C:\Users\<user>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src\

Miackl, 2021-04-05:
Arduino: 1.8.13 (Windows 7), Плата:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Администратор\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Администратор\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 -ide-version=10813 -build-path C:\Users\836D~1\AppData\Local\Temp\arduino_build_877916 -warnings=more -build-cache C:\Users\836D~1\AppData\Local\Temp\arduino_cache_153917 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\GyverPanelWiFi_v1.10.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Администратор\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Администратор\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 -ide-version=10813 -build-path C:\Users\836D~1\AppData\Local\Temp\arduino_build_877916 -warnings=more -build-cache C:\Users\836D~1\AppData\Local\Temp\arduino_cache_153917 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\GyverPanelWiFi_v1.10.ino

Using board 'nodemcuv2' from platform in folder: C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4

Using core 'esp8266' from platform in folder: C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4

Detecting libraries used...

"C:\\Users\\Администратор\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Администратор\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\Администратор\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\Администратор\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\836D~1\\AppData\\Local\\Temp\\arduino_build_877916/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC - [... cut ...]
C:\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp:17:21: fatal error: vfs_api.h: No such file or directory

ResolveLibrary(vfs_api.h) #include "vfs_api.h"

  -> candidates: []                     ^

compilation terminated.

Используем библиотеку ESP8266WiFi версии 1.0 из папки: C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi 

Используем библиотеку ESP8266mDNS версии 1.2 из папки: C:\Program Files (x86)\Arduino\libraries\ESP8266mDNS 

Используем библиотеку ArduinoOTA версии 1.0 из папки: C:\Program Files (x86)\Arduino\libraries\ArduinoOTA 

Используем библиотеку Time версии 1.6 из папки: C:\P

Miackl, 2021-04-05:
По тому адресу у меня не то. C:\Users\Администратор\AppData\Local\Arduino15\packages\esp32\hardware\DFRobot_FireBeetle-ESP32\0.0.9\libraries\FS\src

vvip-68, 2021-04-05:
Откуда ставилась сама Arduino IDE? Если из Windows App Store - снести и поставить с сайта разработчика.
Ядро ESP32 поставить актуальное, через менеджер плат в ArduinoIDE
Ссылка в настройках плат - https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
Рабочая. Файл скачивается, я проверил. Если у вас не скачивается - спрашивайте у интернет-провайдера почему блокируется этот адрес.

Раз по указанному пути у вас не то - значит стоит какое-то другое ядро ESP32 вместо поддерживаемого ESP32 1.0.6
Сносите и ставьте то, которое нужно.

vvip-68, 2021-04-05:
[image]

[image]

Miackl, 2021-04-05:
Смотрю вроде у меня и у вас ссылки одинаковые. Вашу открывает, а мою нет. Непонятно.

Miackl, 2021-04-05:
У Вас опечатка на странице настроек IDE, там написано " raw.githubuserconteenumnt.com" а вы мне дали "https://raw.githubusercontent.com/.

vvip-68, 2021-04-05:
Да, действительно. Спасибо, исправил

Miackl, 2021-04-05:
Arduino: 1.8.13 (Windows 7), Плата:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

C:\Program Files (x86)\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp:17:21: fatal error: vfs_api.h: No such file or directory

 #include "vfs_api.h"

                     ^

compilation terminated.

exit status 1

Ошибка компиляции для платы NodeMCU 1.0 (ESP-12E Module).
По адресу C:\Users\Администратор\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src    у меня 5 файлов: FS.cpp, FS.h, FSImpl.h, vfs_api.cpp, vfs_api.h нужный файл присутствует. Но в ошибке не найден.Что не так?

vvip-68, 2021-04-06:
Ну, как минимум - непонятно - почему у вас "Плата NodeMCU 1.0" (это ESP8266), а библиотеки вы используете от ESP32 - это совсем другой микроконтроллер.

Определитесь какой микроконтроллер у вас на самом деле.
Если ESP32 (а все что выше мы конфигурировали именно для ESP32 - все эти ссылки на подключение платы и т.д) - Выберите в меню "Инструменты" плату ESP32 Dev Module.

Если же ваш микроконтроллер NodeMCU - удалите из папки Arduino/libraries папку LittleFS_esp32 и установите для плат на ESP8266 актуальное ядро - 2.7.4

vvip-68, 2021-04-06:
[image]

Miackl, 2021-04-06:
Да для платы  NodeMCU, и у меня установлено для ESP8266 актуальное ядро - 2.7.4  по адресу C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4  Только при обновлении выдает следующее  Arduino: 1.8.13 (Windows 7), Плата:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

C:\Program Files (x86)\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp:17:21: fatal error: vfs_api.h: No such file or directory

 #include "vfs_api.h"

                     ^

compilation terminated.

exit status 1

Ошибка компиляции для платы NodeMCU 1.0 (ESP-12E Module).

[LittleFS] data    : C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\data

[LittleFS] size    : 2024

[LittleFS] page    : 256

[LittleFS] block   : 8192

[LittleFS] upload  : C:\Users\836D~1\AppData\Local\Temp\arduino_build_893597/GyverPanelWiFi_v1.10.mklittlefs.bin

[LittleFS] address : 0x200000

[LittleFS] reset   : --before default_reset --after hard_reset

[LittleFS] port    : COM28

[LittleFS] speed   : 115200

[LittleFS] python   : C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe

[LittleFS] uploader : C:\Users\Администратор\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\upload.py

esptool.py v2.8
Serial port COM28
Traceback (most recent call last):
  File "C:\Users\�������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/�������������/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2889, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:/Users/�������������/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "C:/Users/�������������/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/pyserial\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "C:/Users/�������������/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/pyserial\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM28': FileNotFoundError(2, '�� ������� ����� ��������� ����.', None, 2)
LittleFS Upload failed!

Miackl, 2021-04-06:
Если  удалить папку LittleFS_esp32, то Arduino: 1.8.13 (Windows 7), Плата:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

In file included from C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\GyverPanelWiFi_v1.10.ino:19:0:

a_def_hard.h:244:24: fatal error: LITTLEFS.h: No such file or directory

   #include <LITTLEFS.h>

                        ^

compilation terminated.

exit status 1

LITTLEFS.h: No such file or directory

vvip-68, 2021-04-06:
Ну тогда я вообще ничего не понимаю и помочь, видимо, не смогу. Вы что-то исправляли в скетче?
То что вы привели выше - это фрагмент:
```
#if defined(ESP32)
  #define   LittleFS LITTLEFS
  #include <LITTLEFS.h>
#else
  #include <LittleFS.h>
#endif
```
То есть - включение   #include <LITTLEFS.h> - происходит ТОЛЬКО для микроконтроллеров ESP32
Если у вас NodeMCU - а это esp8266 - эта строчка вообще не должна включаться компилятором в текст скетча.
Почему она у вас включается (и соответственно и естественно - не находится) - у меня нет ответа.

vvip-68, 2021-04-06:
Ошибке выше для upload - вполне могут быть из за наличия в пути кириллицы - это давно всем известно, что использование русских букв в путях крайне не рекомендуется. Может прокатить, а может и нет. У вас, вероятно - не прокатывает.

Miackl, 2021-04-07:
Привет друг.В скетче ни чего не менял, кроме как в файле  a_def_hard.h поправил под свою схему. Зашел в виндовс под другим пользователем без кириллицы в адресе, теперь выдало другие ошибки.
 Arduino: 1.8.13 (Windows 7), Плата:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

In file included from sketch\a_def_hard.h:218:0,

                 from C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\GyverPanelWiFi_v1.10.ino:19:

C:\Program Files (x86)\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^

In file included from C:\Program Files (x86)\Arduino\libraries\FastLED\src/FastLED.h:65:0,

                 from sketch\a_def_hard.h:218,

                 from C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\GyverPanelWiFi_v1.10.ino:19:

C:\Program Files (x86)\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\GyverPanelWiFi_v1.10.ino: In function 'void setup()':

GyverPanelWiFi_v1.10:138:15: error: 'LittleFS' was not declared in this scope

   spiffs_ok = LittleFS.begin();

               ^

C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\animation.ino: In function 'String openImage(String, String)':

animation:762:10: error: 'LittleFS' was not declared in this scope

     if (!LittleFS.exists(directoryName)) {

          ^

animation:771:12: error: 'LittleFS' was not declared in this scope

     file = LittleFS.open(fileName, "r");

            ^

C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\animation.ino: In function 'String saveImage(String, String)':

animation:866:10: error: 'LittleFS' was not declared in this scope

     if (!LittleFS.exists(directoryName)) {

          ^

animation:876:9: error: 'LittleFS' was not declared in this scope

     if (LittleFS.exists(fileName)) {

         ^

animation:885:12: error: 'LittleFS' was not declared in this scope

     file = LittleFS.open(fileName, "w");

            ^

C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\animation.ino: In function 'String deleteImage(String, String)':

animation:941:10: error: 'LittleFS' was not declared in this scope

     ok = LittleFS.remove(fileName);

          ^

C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\animation.ino: In function 'String getStoredImages(String)':

animation:1001:9: error: 'LittleFS' was not declared in this scope

     if (LittleFS.exists(directoryName)) {

         ^

C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\eeprom.ino: In function 'uint8_t checkEepromBackup()':

eeprom:1709:10: error: 'LittleFS' was not declared in this scope

   file = LittleFS.open(fileName, "r");

          ^

C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\eeprom.ino: In function 'bool saveEepromToFile(String)':

eeprom:1757:9: error: 'LittleFS' was not declared in this scope

     if (LittleFS.exists(fileName)) {

         ^

eeprom:1766:12: error: 'LittleFS' was not declared in this scope

     file = LittleFS.open(fileName, "w");

            ^

C:\Arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.10\eeprom.ino: In function 'bool loadEepromFromFile(String)':

eeprom:1845:12: error: 'LittleFS' was not declared in this scope

     file = LittleFS.open(fileName, "r");

            ^

exit status 1

'LittleFS' was not declared in this scope

vvip-68, 2021-04-07:
Эти все ошибки могут быть только из за установленного ядра для ESP8266 версии меньшe 2.6
Пришлите мне скриншот из менеджера плат с версией библиотеки, типа такого:
[image]

Miackl, 2021-04-07:
[image]
Тут все нормально.

Miackl, 2021-04-07:
Единственное я для ESP32 не устанавливал.

vvip-68, 2021-04-08:
Для ESP32 и не надо, если ваша плата NodeMCU
Ну что ж - если ядро у вас установлено актуальной версии, Arduino IDE - с сайта производителя, а не из Windows App Store - у меня нет ответа как решить вашу проблему.

Miackl, 2021-04-09:
> Моя система не пишет при компиляции такого:
Arduino: 1.8.13 (Windows 7), Плата: "NodeMCU 1.0 (модуль ESP-12E), 80 МГц, Flash, Legacy (новый может возвращать nullptr), Все SSL-шифры (наиболее совместимые) , 4 МБ (FS: 2 МБ OTA: ~ 1019 КБ), 2, версия 2, нижняя память, отключено, нет, все флэш-содержимое, 115200 "
Друг, а вы как копируете сообщение об ошибке?Клавиатурой? Я просто нажимаю кнопку "копировать сообщение об ошибке", потом вставить и первым печатает упомянутые строки ,о настройках IDE. У вас разве не так?

Miackl, 2021-04-09:
[image]
 Мои настройки.

vvip-68, 2021-04-09:
> Друг, а вы как копируете сообщение об ошибке? Клавиатурой? Я просто нажимаю кнопку "копировать сообщение об ошибке", потом вставить и первым печатает упомянутые строки ,о настройках IDE. У вас разве не так?

Мне не приходилось копировать сообщение об ошибке. Я просто просматриваю в этом черном окне вывод и устраняю, если какие-то ошибки есть.

Конкретно в этих настройках - не вижу ничего криминального.
У меня в "Сообщения компилятора" стоит "Ничего", чтобы не засорять вывод лишней информацией. Остальное, насколько я вижу - то же самое.

Miackl, 2021-04-09:
Друг, без платы ESP32 подсоединенной к компу,не удается проверить компиляцию для этой платы.Это так и должно?

Miackl, 2021-04-09:
Выдает ошибку:
exec: "cmd": executable file not found in %PATH%

Ошибка компиляции для платы ESP32 Dev Module.

Miackl, 2021-04-09:
Друг, пароль на private.zip подскажите.

vvip-68, 2021-04-10:
> Друг, без платы ESP32 подсоединенной к компу,не удается проверить компиляцию для этой платы.Это так и должно?

Нет. Наличие физической платы подключенной к компьютеру никак не влияет. Скомпилировать скетч можно для любой платы, просто выбрав ее в менеджере плат в меню "Инструменты"

> Друг, пароль на private.zip подскажите.

**private** - он на то и private, что там данные не предназначенные для посторонних глаз. Вам этот файл для сборки проекта совершенно не нужен.

С вашими ошибками - у меня впечатление, что вам нужно вообще на чистый диск переустановить Windows, потом все нужные программы заново, в том числе и Arduino IDE. У вас система уже похоже в таком раздраенном состоянии, что просто ничего не работает нормально.

Для начала - хотя бы полностью удалите Arduino IDE, поставьте заново скачанный с официального сайта дистрибутив. Если не поможет -- переустанавливайте Windows полностью.

Miackl, 2021-04-10:
Друг, все скомпилировал на другом компе. У меня еще один вопрос. Под конфигурацию оборудования. У вас говорится о 2 -ух вариантах: 1.Все подключено, 0. Только часть подключено. А если подключено почти все, только без одного MOSFETа / реле? Что ставить в DEBUG_SERIAL?И какой вариант править DEVICE_VARIANT == 0 или 1? Спасибо.

vvip-68, 2021-04-10:
Если подключено всё, толь ко без мосфета/реле - это всё равно DEVICE_VARIANT 1 и DEBUG_SERIAL 0

Miackl, 2021-04-14:
Друг, приветствую. Все работает на сто процентов, только если питание подается одновременно и через блок питания на vin и на usb. Без usb не работает. В чем может быть причина? Собирал по схеме.

vvip-68, 2021-04-14:
- Обрыв дорожки от VIN к стабилизатору, например, перегрели пин при пайке.
 - Сожгли дорожку, устроив КЗ
 - Сгорел стабилизатор напряжения на плате...
 - Погуглите еще...

vvip-68, 2021-04-14:
Диод, если используется для поднятия уровня "земли" включен наоборот

Miackl, 2021-04-18:
Приветствую.Короче нашел причину,почему не работала матрица от VIN. Сама плата NodeMCU использует стабилизатор напряжения AMS1117 3.3 вольта. Прозвоном определил,что на него не приходит минус.Далее под микроскопом обнаружил что от пина GND дорожки с двух сторон платы, тупо ни куда не идут. Подпаял проводок от пина GND к 3 ножке  AMS1117 и все заработало. Китайская плата - китайский косяк. Только не стоит одновременно подключать вместе с USB, не известно что там китаец намутил с разводкой на плате.