vvip-68/GyverPanelWiFi #132: погода

EDUARDT68, 2021-04-10:
Привет.
Я не программист и мне трудно что то на лету схватить. По этому прошу помощи. Как настроить погоду? Дело в том что я не знаю код своего региона. А хотелось бы что бы погода была корректной ))))

vvip-68, 2021-04-10:
https://github.com/vvip-68/GyverPanelWiFi/wiki/%D0%9D%D0%B0%D1%81%D1%82%D1%80%D0%BE%D0%B9%D0%BA%D0%B0-%D0%BF%D0%BE%D0%BB%D1%83%D1%87%D0%B5%D0%BD%D0%B8%D1%8F-%D0%B8%D0%BD%D1%84%D0%BE%D1%80%D0%BC%D0%B0%D1%86%D0%B8%D0%B8-%D0%BE-%D0%BF%D0%BE%D0%B3%D0%BE%D0%B4%D0%B5

closed 2021-04-10 by vvip-68

EDUARDT68, 2021-04-10:
Ага. Спасибо. Разбираюсь.
И ещё.  Я для проекта взял вот эту плату: WeMos Mega + WiFi R3 ATmega2560
https://aliexpress.ru/item/4000637643383.html?af=1015802&utm_campaign=1015802&aff_platform=portals-tool&utm_medium=cpa&dp=aa18e1b4c38f56cc90c9caff91e605f7&aff_fcid=fec64c539fc14da19d6acaea75e5d888-1618071663486-03204-_ePNSNV&spm=a2g0s.9042311.0.0.274233edpw5pxH&cv=47843&aff_fsk=_ePNSNV&pvid=cc0c4c29-0884-4650-a457-0ee118b81532&mall_affr=pr3&sk=_ePNSNV&aff_trace_key=fec64c539fc14da19d6acaea75e5d888-1618071663486-03204-_ePNSNV&rmsg=do_not_replacement&scm=1007.23534.124000.0&terminal_id=bdae42b085474cf38c056e272e23ea34&utm_content=47843&utm_source=admitad&sku_id=10000004722725205

Она подойдёт?

EDUARDT68, 2021-04-10:
Погоду настроил.
**Спасибо.**

vvip-68, 2021-04-10:
Думаю, что не подойдет.

EDUARDT68, 2021-04-10:
А можно узнать почему? У Gyver была похожая, но без WiFi.

vvip-68, 2021-04-10:
Потому что это проект рассчитан на три платы - Wemos d1 mini, NodeMCU и ESP32.
Скорее всего, если вы выберите их - в меню "Инструменты" - вас не удастся залить код на ЭТУ плату. Разве что после долгих плясок с бубнами, но сомневаюсь. А если найдете в "Инструменты" именно эту плату - проект просто под нее не соберется. То есть просто не удастся скомпилировать. В общем - этот гибрид и ATmega и ESP8266 - пустая трата денег (в три раза дороже обычной Wemos) и мало чем поддерживается.
Иными словами ЭТОТ проект для ЭТОЙ платы - не предназначен.

vvip-68, 2021-04-10:
Учтите, что вам придется самостоятельно конвертировать код проекта под эту плату, находить к каким пинам подключаться, как все это собирать в кучу. Тут вам никто в этом не поможет.

EDUARDT68, 2021-04-10:
Ясно. Значит будет на NodeMCU или Wemos d1 mini. Они у меня есть. Кстати я
сейчас как таз и играюсь с панелью построенной на NodeMCU. Не всё хочет
нормально работать. Нужно разобраться с настройками. Надеюсь что получится.

EDUARDT68, 2021-05-03:
Привет. Я в какой то из прошивок встречал эффект "ДНК". Не помню чей он был. У Вас не было такого? Если был то можно поделиться? ;-)

vvip-68, 2021-05-03:
Не у меня.

EDUARDT68, 2021-05-03:
Ок. Спасибо за ответ.

EDUARDT68, 2021-05-03:
[Я нашёл. ](https://community.alexgyver.ru/threads/wifi-lampa-budilnik-obsuzhdenie-proshivki-ot-gunner47.2418/post-33652)

EDUARDT68, 2021-05-03:
Там что то около 80 разных.