vvip-68/GyverPanelWiFi #134: Необходима помощь! возможно платная.

wertigo90, 2021-04-26:
Доброго времени суток, возможно ли попросить вас чуть переделать данную прошивку, а именно добавить возможность подключения цифро-аналогового датчика звука как например [тут](https://www.rekshop.ru/product/23917/%D0%94%D0%B0%D1%82%D1%87%D0%B8%D0%BA_%D0%B7%D0%B2%D1%83%D0%BA%D0%B0_%D1%81_%D1%86%D0%B8%D1%84%D1%80%D0%BE%D0%B2%D1%8B%D0%BC_%D0%B8_%D0%B0%D0%BD%D0%B0%D0%BB%D0%BE%D0%B3%D0%BE%D0%B2%D1%8B%D0%BC_%D0%B2%D1%8B%D1%85%D0%BE%D0%B4%D0%B0%D0%BC%D0%B8?page=2&number=20), чтобы при срабатывании датчика матрица заливалась определенным цветом например на пол секунды(или меньше, как будто вспышка), если возможно реализовать то чтобы можно было выбирать цвет, яркость и продолжительность задержки, так же если возможно так же добавить такие возможности в API управления устройством, так же для облегчения  удалить игры, часы, будильники. Если данная процедура слишком сложна или обойдется слишком дорого можно хотя бы просто реакцию на датчик. Соответственно готов заплатить, сколько посчитаете это может стоить?

vvip-68, 2021-04-26:
Нет, со звуком я связываться не буду. мне это не интересно. Попробуйте в "Ищу разработчика" спросить - может кто подправит.

closed 2021-04-26 by vvip-68

kostyamat, 2021-04-26:
Есть готовая прошивка с поддержкой звука, и кучей эффектов под музыку https://community.alexgyver.ru/threads/wifi-lampa-budilnik-proshivka-firelamp_jeeui-gpl.2739/