vvip-68/GyverPanelWiFi #135: HELP! ПЛАТНО!

wertigo90, 2021-04-26:
Нет, со звуком я связываться не буду. мне это не интересно. Попробуйте в "Ищу разработчика" спросить - может кто подправит.

_Originally posted by @vvip-68 in https://github.com/vvip-68/GyverPanelWiFi/issues/134#issuecomment-826943556_

Ок как со звуком не надо, этот датчик в другом моем изделии был реализован как кнопка, по цифровому каналу он был подключен как кнопка, возможно же сделать как на реакцию кнопки, я в принципе уже попробовал  вместо обычной кнопки это сделать что у вас в проекте, там есть только 1 и 0 , и реагирует отлично при настройках сенсорной кнопки. Только не совсем понимаю как сделать чтобы была заливка всей матрицы по событию, и чтобы эффект после этого не сбрасывался. Прошу очень!

vvip-68, 2021-04-26:
Обрабатывать нажатие кнопки (видимо - однократное) - сейчас это примерно в a_main.ino строка 228, далее писать свою функцию эффекта которая будет заливать всю матрицу указанным цветом на основе вызова fillAll(fillColor) с дальнейшим угасанием, если это нужно. В начале обработки нажатия кнопки "взводить" флаг включения "вспышки" и  запоминать момент времени когда начинается вывод заливки. Далее в custom.ino в processEffect() в строке около 365 - если флаг взведен - выводить вашу "вспышку" вместо эффекта. Или то же самое в строках около 220, откуда этот processEffect() вызывается, чтобы вместо текущего эффекта отображать вспышку.  Когда время "вспышки" прошло - снимать флаг - будет восстановлена обработка текущего эффекта. Грубо это так. По нюансам - возможно в custom.ino в строке 71 добавить проверку на этот флаг - если установлен - бегущую строку включать не надо (а то вдруг ее время подойдет или она уже отображается). Так же ниже - проверку флага нужно ли во время вспышки рисовать часы, и т.д.

Далее при необходимости вынесения настроек (цвет, длительность) в приложения - добавить кучу обработчиков команд коммуникации со смартфоном и обработки этих команд в приложении. Это, пожалуй, будет в три-4 раза сложнее чем саму вспышке добавить...

Даже если вопрос закрыт - в нем все равно можно продолжать переписку.

closed 2021-04-26 by vvip-68

wertigo90, 2021-04-26:
Спасибо большое!, в смартфон не нужно, это я делаю в приложение на компе, пытаюсь сам разобраться, главное что уже получилось связать по UDP и настроить отправку, остальное уже не так быстро и надо, а вот это реализовать нужно было. Не совсем понятно как добавить в API, но думаю что в общем то смогу разобраться на основе того что есть. спасибо буду пытаться реализовать, нет серьезно огромное спасибо. у меня не получится к сожалению написать грамотно код, я только начал учиться.

wertigo90, 2021-04-26:
Если вам интересно помочь в данном вопросе, скажем за 3-4т.р. я готов заплатить, вам это будет сподручнее, тем более что вы знаете даже построчно где что находится в коде, а для меня это процесс целого курса обучения. тем более вы сделаете это точно грамотнее меня. хотя бы именно срабатывание на нажатие (click), если это возможно буду рад, оплата переводом на карту. Ну или куда вам удобнее было бы.

wertigo90, 2021-05-17:
Прошу прощения для меня меня это оказалось довольно сложно. Прошу все же помощи за деньги, совсем запутался во флагах. да и в принципе отрисовки. Может быть все же платно сможете уделить время? ни кто не хочет ковыряться и изучать прошивку. я пытаюсь но со скудным знанием языка в принципе трудно, а тут еще и непонимаю какие флаги куда нужно, вроде создал себе флаг, вроде даже добавил таймер, не могу отрисовать. да и вообще для меня процесс внесения изменений кажется мне как щенок слепой тыкается носом везде а без результатно в моем случае(((

vvip-68, 2021-05-17:
Напишите ТЗ сюда - https://vk.com/iivanovii
Я уже не помню что конкретно вы хотите...