vvip-68/GyverPanelWiFi #141: Не подключаеться через WIFI

Nereg, 2021-06-06:
Здравствуйте! Собрал такую лампу и через встроенную сеть `PanelAP` лампа отлично работает. Но когда подключаю к своей сети то не получаеться подключиться. Не работает ни [адрес](https://prnt.sc/14ctxs6) который выдал роутер через DHCP ни тот который [отображаеться ](https://prnt.sc/14cu3j5) в логах (они разные почему то). Защиту беспроводных точек [выключил ](https://prnt.sc/14cu8tn) как и сказано в FAQ. Свою точку ESP не создает. Так же в консоли отображаються команды от моего телефона : 
```
UDP << ip='192.168.0.102:2390'; cmd='$18 0;'
UDP пакeт размером 6
```
Но приложение все равно не подключаеться. Вот настройки сети : 
```
#ifndef NETWORK_SSID
#define NETWORK_SSID "Oleg"                     // Имя WiFi сети
#endif

#ifndef NETWORK_PASS
#define NETWORK_PASS "пароль"                     // Пароль для подключения к WiFi сети
#endif

#ifndef DEFAULT_IP
#define DEFAULT_IP {192, 168, 0, 103}       // Сетевой адрес устройства по умолчанию
#endif
```
Для подключения исползую `192.168.0.103`. Версия прошивки : `v.1.10.2021.0521` Версия приложения : `v1.10.2021.0418`

vvip-68, 2021-06-06:
В логах - 192.168.0.0  - это не валидный адрес. Лампу не будет видно в сети, если у не такой адрес.
Когда лампа создала точку доступа PаnelAP и вы к ней подключились - какой адрес вы назначили лампе в настройках подключения?

Nereg, 2021-06-06:
`192.168.0.103`. Но панель говрит что нету подключения к WiFi хотя пароль правильный

vvip-68, 2021-06-06:
В логах - 192.168.0.0 - это не валидный адрес. Лампу не будет видно в сети, если у не такой адрес.
В f_def_soft.h в строке 8  задайте какой-нибудь другой идентификатор, например 0x55 и перепрошейте лампу.
Когда она создаст точку доступа PanelIP - подключитесь к ней, зайдите на смартфоне в настройки сети, пропишите там имя/пароль сети и адрес, который должна получать лампа. Нажмите несколько раз (для верности) кнопку "Сохранить". Когда заданные параметры правильно сохранятся - лампа нарисуеет на матрице полученный IP адрес. Он должен быть тем же, что вы вбили в настройках. После этого можно подключаться к лампе по локальной сети, задавая этот адрес для подключения.

Nereg, 2021-06-06:
Ок подключил. В DHCP появляеться. В логах адрес такой же. Свою точку не создает. Но приложение все равно не может подключиться. В логах даже отображаются попытки подключиться : 
```
Запрос текущей погоды
UDP << ip='192.168.0.102:2390'; cmd='$18 0;'
UDP пакeт размером 6
Таймаут запроса погоды!
```
но приложение не подключаеться

vvip-68, 2021-06-06:
У меня нет ответа. Судя по описанному - должно работать. Даже пакеты от приложения к лампе приходят.
Не знаю как проверить приходит ли что-то от лампы в приложение. Если не приходит - то соединиться не сможет.
Если по какой-то причины пакеты от лампы к телефону где-то пропадают - соединение не установится.
Нет ли в настройках роутера чего-нибудь, что не дает передаваться пакетам от 102 к 103?...

Nereg, 2021-06-06:
Прошил HTTP серер - работает (и на порте 80 и на порте 2390 и с телефона и с компьютера)

closed 2021-06-22 by vvip-68