vvip-68/GyverPanelWiFi #143: GyverPanelWiFi

Verholaz, 2021-08-11:
Сделал 2 лампы с вашей прошивкой. Всё работает. Огромное спасибо Вам за проделанную работу. Но есть 1 вопрос по последней прошивке. В папке sounds - 04 в readme есть строчка: "Звуки отбивки каждого часа, если включен часовой сигнал в часах (женский голос Оксана)." Вопрос: "Это где?" В приложении нет. Смотрел во вкладках прошивки - тоже не нашёл. Спасибо за ответ. P.S. Приложение из архива с этой прошивкой.

vvip-68, 2021-08-12:
Этого нигде нет. Это нереализованный задел на будущее...

С уважением,
Вадим Пузанов

closed 2021-08-12 by vvip-68

Verholaz, 2021-08-12:
Добрый вечер. Спасибо за ответ. Сделал ещё один проект ( матрице ) с вашей прошивкой GyverPanelWiFi v1.10. Выложил на форуме  Bluetooth матрица под этим ником. Вам желаю успешных проектов и удачи.