vvip-68/GyverPanelWiFi #145: Погода из mqtt

Securond, 2021-10-12:
Хотел бы поблагодарить автора за этот мегапроект. В 2018 году я допиливал проект Алекса для работы с Mqtt, и вывода температуры и времени на матрицу. Так она проработала у меня почти 2 года радуя людей на улице.  А сегодня я наткнулся на этот проект и был безумно рад что многие функции которых мне не хватало, уже реализованы в данном проекте. 
Из того что не хватает - это вывод температуры из своего топика mqtt. Yandex конечно хорошо, но он иногда врёт, а погодная станция которая висит за окном и шлёт погоду в брокер, всегда даёт точные данные. Если не затруднит автора добавить такую возможность буду примногоблагодарен! 
Ну и высшим пилотажем было бы создание интеграции в Home Assistant.

vvip-68, 2021-10-13:
1. Добавление третьей возможности (кроме Яндекса и OpenWeatherMap) теоретически реализуемо. На практике - довольно затруднительно. Допустим, добавить код в скетч - не проблема. Это, насколько я понимаю - добавить параметры подключения к MQTT брокеру и подписка на соответствующий топик. Проблема в том, что вынести настройки в приложение на смартфон достаточно затруднительно - это нужно добавлять соответствующие поля и блоки кода. Приложение для смарта написано в блочном конструкторе AppInventor и достигло таких размеров что а) еле шевелится в редакторе - что-то изменить там - боль. б) AppInventor зачастую просто отказывается собирать приложение из за слишком большого размера - выдает ошибку и не компилирует.

2. Интеграция в Home Assistant - довольно затруднительно реализовать то, в чем ничего не понимаешь. Были энтузиасты, которые что-то делали в этом направлении - например https://github.com/tarasifua/vvip-68-matrix-home-assistant-integration. Тут более ничем помочь не смогу, наверное.

3. Скетч сейчас переписывается под возможность использования модульных сборок матриц. То есть, скажем, берем несколько готовых матриц 8x8 или 16x16 или вообще самодельных, скажем размером 10x10. Потом эти отдельные модули собираются в определенном порядке в большую матрицу и скетч гибко настраивается под указанное количество сегментов  и конечные размеры сборки и способ соединения сегментов задаются из отдельного приложения-конфигуратора на смартфоне. Это позволит также без перепрограммирования (перепрошивки) подключать один и тот же контроллер к разным матрицам (на новый год - к гирлянде на окне со своими размерами, после - к какой-нибудь лампе или панели). Или со временем подключать дополнительные диоды к гирлянде.
Это я к тому, что скетч сейчас все равно в "разобранном" состоянии. Пока по непонятным причинам он периодически падает. Ищу причины, отлаживаю, ковыряюсь. Сколько времени потребуется - не знаю.

closed 2021-10-13 by vvip-68

vvip-68, 2021-10-13:
Если реализуете интеграцию в умный дом с Home Assistant - пришлите пожалуйста инструкцию и все что необходимо - выложу в wiki? чтобы другие тоже смогли воспользоваться. 
PS. Возможно инструкция от @tarasifua немного устарела. Например переменная mqttstatepacket уже убрана из прошивки и там уже нет пакетного режима отправки значений - уже давно всё по отдельности шлется.

tarasifua, 2021-10-13:
@vvip-68, привет! Немного перестал следить за темой, так как пользуюсь гирляндой только на НГ😊
По поводу интеграции. У меня же сделано управление через MQTT, и как раз через тот режим, где параметры шлются по отдельности. Это как раз я Вас и просил так сделать, так как парсить одну строку с кучей параметров сложно. 
Так что если не менялись названия параметров или типы их значений, в принципе всё должно работать

vvip-68, 2021-10-13:
@tarasifua, так я тогда же (ну или немного после) и сделал по твоей просьбе. Плюс вроде еще кто-то пытался интегрировать в умный дом. Параметры не менялись. Вот в связи с переделками выше добавится несколько параметров, а так - всё то же...

tarasifua, 2021-10-13:
@vvip-68, ну это я и делал интеграцию в умный дом Home Assistant. Чуть позже достану ее, прошью новую прошивку и поправлю, если вдруг что-то сломалось

sanyafifa, 2021-10-14:
> что в какой топик отдать надо при старте лампы, чтобы она автоматически прописывалась в умном доме.

LWT сообщение наверное . ))
@vvip-68 Добрый день!
Скоро придут новые лед-матрицы. будем тестить.

vvip-68, 2021-10-14:
@sanyafifa вот конкретна эта сборка из 4 матриц 16x16 по памяти на ESP8266 постоянно остается свободной еще 15-17К. Памяти на самый ресурсожрущий эффект забирает примерно 1.6К. Так что думаю даже сборку 4x4 возможно потянет... Вопрос в быстродействии - скорость передачи на WS8212B - всего около 800Кбит. Хотя вот даже похожие WS8213 - уже 2000К передача. А более дорогие APA102 - кажется еще быстрее... 
На ESP32 остается в пределах 250К свободных по памяти...

> можно подумать за синхронизацию между разными контроллерами кусков матрицы.

Вот в планах на следующую версию - как раз попробовать замутить сеть из девайсов master-slave с синхронизацией управления... Но тут речь пока просто о синхронной работе, а не так как кто-то когда-то мне предлагал замутить типа на окна в доме вешается длинная матрица со своим контроллером на каждом куске и синхронизацией так, например, что длинная бегучая строка сквозь всех их проходит... Это я не знаю как сделать. Что-то сложно. Наверное много думать нужно будет...