vvip-68/GyverPanelWiFi #146: Помогите разобраться с версиями

voshodvoshod, 2021-10-19:
Здравствуйте. Переехал сюда из Ютуба с вопросами  по прошивке.
Помогите пожалуйста понять как различить версии прошивок?
В далёком апреле месяце Вы по моей просьбе исправили глюки   рассвета и будильника в режиме точки доступа.
Это была версия с маркировкой #define FIRMWARE_VER F("WiFiPanel-v.1.10.2021.0418"). Что можно понять как "Версия 1.10 от 18 апреля 2021 г." Правильно?
Позже в ней обнаружился ещё один дефект, который блокировал рассвет и будильник после их отключения с кнопки во время проигрывания. Эту версию с исправлениями проверить не удалось(отпуск, отсутствие девайса и др. дела).
Когда собрался проверять, эту версию не нашёл. Зато нашёл 1.11, которая , как Вы правильно заметили, для лампы избыточна. Сейчас кроме 1.11 предыдущая 1.10.1. Открываем её и видим #define FIRMWARE_VER F("WiFiPanel-v.1.09.2021.0225")Следуя логике это значит "Версия 1.09 от 25 февраля 2021г." Так это 1.09. или 1.10.1? Как мне найти апрельскую версию, в которой исправлено блокирование рассвета и будильника при отключении их во время воспроизведения?
Спасибо.

vvip-68, 2021-10-19:
> Эту версию с исправлениями проверить не удалось(отпуск, отсутствие девайса и др. дела). Когда собрался проверять, эту версию не нашёл. Зато нашёл 1.11,

Так я вроде поправил рассвет уже в 1.11/ ЧТо - все равно не работает?

voshodvoshod, 2021-10-20:
Добрый день. Версию 1.11 не могу настроить. Хочу оставить только 36 рисунков. Т.е. убрать игры, погоду, анимацию, интернет.
В предыдущих версиях всё понятно было. А здесь что-то пропускаю и не работает.

vvip-68, 2021-10-21:
В любом случае вернуться к вопросу смогу после того, как будет доделана 1.12. Сейчас всё разобрано и выкладывать поломанную разобранную версию - вообще не вариант...

voshodvoshod, 2021-10-21:
Я не знаю, можно ли здесь написать, заранее извиняюсь, если чего нарушил.
А конкретно для лампы Вы уже не поддерживаете?
Сейчас загрузил версию 1.02. Практически это мой вариант.Только эффектов меньше и сохранения настроек нет.
Зато громкость и визуализация нормальные. Пока лампа у меня, понаблюдаю.Очень не хватает модуля RTC. Сейчас вот свет моргнул и время слетело.Это мой первый вариант, делал в расчёте на модуль часов с батарейкой. В этом корпусе у меня нет места для лития с обвязкой.Перепечатывать пока нет возможности и пластика.Может найдёте время и желание для этого проекта с  RTC без интернета? 
Спасибо.

vvip-68, 2021-10-21:
Лампа это старый проект, из которого вырос вот этот вот. Он уже не поддерживается. Соответственно ошибки в нем не исправляются и новые функции не добавляются. Добавлять туда RTC на данный момент у меня нет ни времени ни желания. Может быть когда-нибудь, но вероятно не в этом году - точно.

Почему рассвет не включался - я понимаю и исправил. Там память выделялась при смене ID эффекта (эта версия - с динамическим выделением памяти), рассвет - это фактически 3 ID - сам "рассвет как понятие" и - рассвет с беганием полосок квадратами и ли по спирали для лампы - как реализация эффекта. ID при отработке менялся и сама отрисовка все время сбрасывалась на первый кадр. Я это исправил. Но другие аспекты - не менялись от слова "совсем". Там всё как было в предыдущей версии. Я поэтому даже не знаю где и как искать, если вы говорите, что перестало работать. К тому же - когда я все это проверял на своем устройстве после починки (см. выше) - всё работало.

voshodvoshod, 2021-10-21:
Очень грустно. Однако позволю себе рассказать одну историю.Возможно Вы её слышали. Когда Ричард Фейнман был привлечён к атомному проекту он не вникая ткнул в 2 точки на схеме и спросил:" что будет если открыть этот кран не закрыв второй?" После долгих изысканий сотрудники пришли к выводу, что произойдёт страшная авария. А после спросили, как это он догадался.На что Фейнман ответил примерно следующее:" Все сложные системы ненадёжны".
Таким образом, из этого следует то, что с ростом сложности системы разработчики теряют над ней контроль. Причём этот постулат сохраняет свою актуальность на любом уровне развития технологий. Это не в укор Вам, не обижайтесь. Это, можно сказать, закон Природы. По-моему текущий проект приближается к такому состоянию. Ещё раз прошу прощения, если обидел. И надеюсь увидеть прошивку, надёжную, как лом.

voshodvoshod, 2021-11-02:
Добрый день. Погонял несколько дней. Работает!Спасибо!
Заметил две особенности:
Если матрица выключена по расписанию, то в приложении снимается птичка автовоспроизведения деморежима.
Ну это вообще не проблема, если матрица включается по расписанию-деморежим работает.
Второе-если время окончания будильника совпадает с временем включения деморежима по расписанию, то матрица моргает и выключается.При включении кнопкой(1 клик) матрица включается, но зависает на одном эффекте.
Все нужные птички в приложении установлены, а он не включается после таймаута для деморежима(у меня 2 мин.).
Полагаю, что это следствие первой особенности.
Не критично, просто передвинул утренний деморежим на минуту позднее и всё.
Больше пока ничего не заметил.
Ещё раз благодарю!

vvip-68, 2021-11-02:
> Если матрица выключена по расписанию, то в приложении снимается птичка автовоспроизведения деморежима.

А что такое, по вашему "матрица выключается"? Реально - это включается режим (эффект) "черная лампа" - то есть эффект при котором светодиоды не светится. Внутри все продолжает работать - и время синхронизируется и не пора ли включать будильники и т.д.и т.п. Естественно, не нужно чтобы через 2 минуты этот режим не сменялся автоматически на другой - иначе лампа "включится". А как обозначить, что не надо переключаться на другой режим? Правильно - снять галочку "автопереключение". Ну, вот так вот оно и работает.

> Второе-если время окончания будильника совпадает с временем включения деморежима по расписанию, то матрица моргает и выключается.

А тут вопрос приоритетов - что должно делаться вперед, что имеет приоритет? Выключить будильник или включить эффект (и то и другое по сути - эффект, работающий в данный момент времени)? А почему не по другому? Ну вот - в такой последовательности алгоритм и работает. После отключения будильника оно возвращает к эффекту, который был до включения будильника. А это эффект "выключено или черная лампа". - В отслеживании указанного момента времени срабатывает что-то одно. Сначала включается демо-режим по расписанию, на следующем цикле - выключается будильник и восстанавливает эффект, который был до включения.